OpenCPN/OpenCPN #2705: Send WP to GPS using serial:COMx crash O

Hakansv, 2022-05-24:
**Describe the bug**
When send WP to GPS using serial COM port O fails on empty "dstr". See pict 1
For me on Win10 it fails to initialize "dstr" on multiplexer.cpp:1193, See pict 2

Send WP to e.g. a TCP is though working.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a serial connection.
2. Create a route.
3. right-click a WP and select "Send to GPS (Serial COM x)" or "Send to new GPS" and select the assigned serial port.
4. See error

**Expected behavior**
"Waypoint(s) trasmitted"
**Screenshots**
See below

**Desktop (please complete the following information if applicable):**
 - OS: Win10 and [macOS](https://www.cruisersforum.com/forums/f134/opencpn-5-6-2-crash-on-waypoint-send-to-gps-264786.html#post3627888)

[image]

[image]

Hakansv, 2022-05-24:
@nohal 
You was assigned due to some "automatic" rule and not by me. Hopefully the new logic for issues is appreciated? :)
/H

PRBgithub, 2022-05-26:
I have similar problem. NMEA route upload to Furuno works well, but upload to Garmin 76 using Garmin Protocol results in a OPenCPN crash. This used to work well in 2021. Has anything been changed?

PR #2710 (2022-05-30): Take out a doubled definition of the serial Datastream variable: dstr
Fixes #2705
Please check if the solution can be this simple!? 
It works on Win10 for "General" wpl output. (See shot)
Furuno output is still working.
Garmin is "working" but not finalized due to "Time out..." in the log since no Garmin is connected so probably also OK. (Let's hope for testers)

[image]
closed 2022-05-30 by bdbcat