OpenCPN/OpenCPN #2706: libusb error when compiling on odroid C2

bytting, 2022-05-25:
Error message while compiling OpenCPN, File not found: libusb.h

OS: Ubuntu 20.04 LTS for odroid C2

I managed to compile on the odroid C2 card after adding a 2Gb swap file (2Gb memory alone was not enough)

To work around the error mentioned above I had to install the package libusb-1.0.0-dev and modify the cmake command:

$ CXXFLAGS=/usr/include/libusb-1.0 cmake ../

Compiling on the card itself may not be the preferred method but I figured it was worth reporting the issue anyway

leamas, 2022-05-25:
libusb is not the only build dependency. Have you found the  [documentation](https://opencpn-manuals.github.io/main/ocpn-dev-manual/od-compile-linux.html)?

If the cmake script does not locate libusb after installing libusb-dev is could be a bug. However is is not likely:  both CI builds and those done by developers are using the workflow documented without any such problems.  

Compiling it is of course the way if you want to participate in the development (welcome!).   That said, are you aware that OpenCPN is available both from PPA and official repositories. 20.04 is still on 5.6.0 in official backports, to be updated "Real Soon (TM)".

bytting, 2022-05-25:
I was following this document for Debian Sid https://github.com/OpenCPN/OpenCPN/blob/master/INSTALL

libusb-1.0.0-dev is not mentioned there.

I have been testing official repos on a RPi 3 but the o-charts plugin wasn't available for that build.

At some point I realized I had a odroid-c2 lying around with more memory, and I decided to compile both OpenCPN and the o-charts plugin to make sure everything would "just work (TM)"

The plugin loaded fine and I am now waiting for a USB GPS dongle to be delivered.

leamas, 2022-05-25:
> I was following this document for Debian Sid https://github.com/OpenCPN/OpenCPN/blob/master/INSTALL

Sorry, that is obsolete stuff. Unfortunately, not the only example. The actual bug seems to be here.

> I have been testing official repos on a RPi 3 but the o-charts plugin wasn't available for that build.

Did you enable backports? o-charts should be available for the version in backports. The alternative is to use the Ubuntu PPA. It's kind of scary, but seems to work.

bytting, 2022-05-25:
> Did you enable backports? o-charts should be available for the version in backports. The alternative is to use the Ubuntu PPA. It's kind of scary, but seems to work.

Hm, on Raspbian I seem to remember adding the PPA and installing outright, but there was no o-charts plugin. I enabled the CatalogExpert option in the config file and looked at the plugins for other builds, like windows, Ubuntu x86_64 and Android. Some of them seemed to list the o-charts plugin, but not the Raspbian build I was on.

At that point I installed OpenCPN on my Windows computer and there the o-charts plugin was available.

> It's just scary when you are on Debian/Raspbian, not when you are on Ubuntu.

I'm pretty used to setting up PPAs, compiling from source etc. so that's not a problem.

> That is, if you are on a supported platform. What does uname -asay on your Odroid?

I forgot to mention, Odroid c2 is one of those ARM cards, similar to Raspberry Pi, so it's a different CPU architecture than x86_64.
The Ubuntu 20.04 I am using here has been compiled by the Hardkernel guys specifically for their ARM cards.

That's why I figured I had to compile everything myself to avoid conflicts. It does work fine though, I just had to install libusb and add that include path to overcome the error.

leamas, 2022-05-25:
> uname -a says "Linux odroid 3.16.85-65 ... aarch64 GNU/Linux"

OK, that's the reason. We don't build native aarch64 plugins, only flatpak.  The memory consumption might be a problem, it is on an RPi-3 with 1 GB. That said, you could certainly give it a try. This would mean you would have the benefits from updates, more plugins, etc. See the [installation docs](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:getting_started:opencpn_installation:flatpak)

bytting, 2022-05-25:
My guess was that flatpak wouldn't help me on a unsupported architecture like aarch64 so I never tried that. Compiling everything resolves all dependencies right away so I'm fine with that for now. I'll just disable OS updates and dedicate it for this specific purpose. 

Thanks for helping out btw.
Since I'm using a unsupported distro this may not qualify as a bug per se, unless the same thing happens on regular Ubuntu as well.

bytting, 2022-05-25:
The odroid c2 has 2Gb ram, and I have enabled 2 charts from o-charts. Scrolling is slow, as expected, but no memory problems so far

leamas, 2022-05-25:
> Since I'm using a unsupported distro

You are not, actually. It's just that Flatpak is the support.

> Since I'm using a unsupported distro this may not qualify as a bug per se, unless the same thing happens on regular Ubuntu as well.

No, it's not really a bug besides the obsolete build docs. We should fix those, and it makes sense to leave this open until it's done

Glad that everything works for you. Happy sailing!!

bytting, 2022-05-25:
> Glad that everything works for you. Happy sailing!!

ditto

bytting, 2022-05-25:
The cmake files looks for libusb so maybe I made a mistake. I may have forgotten to re-run cmake after installing libusb-1.0.0-dev, ending up adding the command line flag unnecessarily.

Updated documentation would be nice though, maybe installing libusb-1.0.0-dev is enough

PR #2711 (2022-06-05): Update INSTALL documentation (#2706).
- Add appropriate links to the manual.
- Update GTK2 requirement to GTK3.
- Update WxWidgets requirement to 3.0.0.
- Use generic Debian build dependency installation.
- Remove unsupported mingw instructions.

Closes: #2706
commit 59fc025c (2022-06-05): Update INSTALL documentation (#2706).

- Add appriate links to the manual.
- Update GTK2 requirement to GTK3.
- Update WxWidgets requirement to 3.0.0.
- Use generic Debian build dependency installation.
- Remove unsupported mingw instructions.

Closes: #2706
closed 2022-06-05 by nohal