OpenCPN/OpenCPN #2708: Invalid true heading are displayed

mgrouch, 2022-05-26:
OpenCPN 5.6.2

I have GPS (gpsd), SignalK, PyPilot feeding OpenCPN NMEA 0183.

PyPilot gives magnetic heading accurately.
SignalK gives magnetic variation (accurately) via derived data plugin.
GPS also sends RMC sentence where mag variation is blank. (In another set up it sends it with wrong data).
It seems even tough OpenCPN has magnetic variation plugin enabled it
still converts null from RMC into ) mag variation and uses it.

So my true heading is jumping on the screen from correct to incorrect one.

Thanks

Hakansv, 2022-05-26:
> It seems even tough OpenCPN has magnetic variation plugin enabled it
> still converts null from RMC into ) mag variation and uses it.

Yes, that's true. (And not a bug, it's a feature!:))
That's to be able to use old (GPS) devices without a valid or outdated variation table. Then the user can set the device to transmit var = 0.0 and thus force the WMM plugin to serve accurate variation. 
Since you've different sources with diverse data you may try to disable WMM or "tell" SK to omit the variation?

mgrouch, 2022-05-26:
Can OpenCPN recognize and filter out bad HDT? 

If abs(HDT-HDM) < 0.00001 then ignore HDT. 

I think it makes sense.

Hakansv, 2022-05-26:
OCPN do of course postulate data arriving to be correct. If not I suggest to correct the source?

The var = 0.0 adaptation is just a way to be able to use somewhat outdated devices but still doing it's job. Some device can't be set to omit variation but it can be set to 0.0 ( i.e. instead of null)

mgrouch, 2022-05-27:
I disagree. I think you should log the error event somewhere but not display clearly wrong data to the user

Hakansv, 2022-05-27:
> I disagree. I think you should log the error event somewhere but not display clearly wrong data to the user

If there's a unambiguous rule to detect "wrong" data it would be no problem to implement I think. But how should such a rule be formed?

> BTW is same logic to handle zero and null magnetic variation from gps is in dashboard plugin?

"var" is primarily identified, from data or WMM, by OCPN itself and if present also used by Dashboard.

closed 2022-06-01 by mgrouch