OpenCPN/OpenCPN #2712: Ambigous items (e.g. buoys) in object query - mabye sorting would help?

matthijskooijman, 2022-06-02:
**Describe the bug**
I guess this is a bit halfway through a bug report and a feature request.

During normal usage, I've noticed that when I want to find out more details about something on the map, such as a buoy, I double-click it and I get info on multiple nearby buoys, and the buoy I clicked is *not* the first one in the list (which is sort of a bug).

It seems the reason multiple buoys are returned is because the object query does a check against the bounding box of the rendered bouy (or other point object), which includes any text/name of the object. I noticed this happening in particular around cardinal buoys (which often have longer names) and it was more pronounced because I had configured chart text to be 25pt, but it is also reproducible (just less common) with the default settings.

I was originally using o-charts (which seems to have an additional problem where the bounding box seems to be larger than expected), but it is also reproducible using ENC charts.

**To Reproduce**
Steps to reproduce the behavior:
1. Using the chart downloader, download "ChartCatalogs/The Neterlands Inland NEC Charts".
2. Go to 53° 00.1208' N / 004° 46.1101' E
3. Double-click the MK7 Buoy
4. See that the object query window shows info about MK8 at the top (see screenshot)

**Expected behavior**
The object query window should show info about the clicked buoy (MK7) first, at least before other buoys and objects of the same priority.

**Screenshots**
After double-clicking the icon of MK7:
 
[image]

**Additional considerations**
An easy fix for this would be to simply sort the "point" objects returned by their distance to the clicked point on the map, which should make sure clicking on the buoy or other icon itself pretty much always makes that item show up at the top of the list.

*areas*
A somewhat similar issue happens for areas, where info about some small area that you are interested in is obscured by more generic info that applies to a larger area (i.e. the entire sea). This is harder to reproduce with the free ENC charts, but when using o-charts for the Netherlands, you can click pretty much anywhere within the "Waddenzee" (inner sea at the North of the Netherlands), where the "Wadden PSSA" and "Nature 2000-site" RESARE items are returned wherever you click, and have a large amount of text, making it hard to find different things.

For areas, sorting by area size seems an obvious way to ensure a better ordering. Alternatively (and, thinking about it, better) would be to sort by the distance between the clicked point and the area contour. This means that clicking a contour should ensure that the area defined by that contour ends up at the top of the list, making this an easy and reliable way to find out what some contour on the map means. This also makes it easier to mix areas and points (and also lines) in the same sorting.

*priorities*
One complication is priorities: Currently, [if I read the code correctly](https://github.com/OpenCPN/OpenCPN/blob/c13add1eecd8b241629239b15957fe4e2b0844b1/src/s57chart.cpp#L4619), the list of displayed objects is generated by going through object in order of [priority based on the type of object](https://github.com/OpenCPN/OpenCPN/blob/c13add1eecd8b241629239b15957fe4e2b0844b1/include/ocpn_plugin.h#L959-L968), and within the same prioritity, first points, then areas, then lines.

Simply sorting based on distance *within* each priority seems obvious, but will still lead to unexpected behavior when double-clicking a low-priority item. This can be seen now already, when double-clicking a wreck, which have lower priority than (also big) RESARE areas, so the clicked wreck would then still not be displayed at the top (or even visible without scrolling for verbose areas or other items).

I can imagine some kind of balanced sorting, where the priority and distance are combined using some weighting into a importance value, and then items are just sorted by that importance value. The exact formula for determining this importance value would need some consideration (I can imagine that objects that are very close to the clicked point should be the most important, even when they have low priority, and objects further down the priority list will have the priority dominate the importance value).

*global or per-chart sorting*
It would be nice to implement his sorting globally (e.g. after querying the chart for items to show [here](https://github.com/OpenCPN/OpenCPN/blob/c13add1eecd8b241629239b15957fe4e2b0844b1/src/chcanv.cpp#L9416-L9421) and before actually converting them to HTML a bit further below), but at that point the priorities of these objects are no longer known, so the sorting will mix the priorities again. This means that either the charts should return the priority for each item and then let the global code handle sorting, or each chart type/plugin must handle its own sorting (potentially resulting in duplicate code).

Looking a bit closer at the plugin interface, the PI_S57Obj's returned by [GetObjRuleListAtLatLon](https://github.com/OpenCPN/OpenCPN/blob/c13add1eecd8b241629239b15957fe4e2b0844b1/include/ocpn_plugin.h#L852) already contain a [m_DPRI](https://github.com/OpenCPN/OpenCPN/blob/c13add1eecd8b241629239b15957fe4e2b0844b1/include/ocpn_plugin.h#L1068) field containing the priority (though o-charts [does not seem to actually fill this field currently](https://github.com/bdbcat/o-charts_pi/blob/e2acca2e6b3c6e2209695188be8eb68fas065c44c/src/eSENCChart.cpp#L6649-L6675), but I guess that could be considered an o-charts bug).

Doing the sorting globally also could include all displayed objects in the sorting (including S57 overlays, AIS area notices and additional files).

**Desktop (please complete the following information if applicable):**
 - OS: Ubuntu
 - Version 22.04
 - OpenCPN: 5.6.1, and self-compiled git c13add1 (2022-05-30)

campbellhennessy, 2022-06-07:
I just submitted a pull request #2720 to propose a solution to the buoy ordering problem.  Basically the solution recommended by @matthijskooijman but with the implementation in the s57chart object instead of the object query dialog (I feel this ordering makes sense for any possible use of the GetObjRuleListAtLatLon member.  This pull request doesn't address any of the other issues identified by @matthijskooijman  but the approach could be modified to include them.  In particular area sorting could be implemented in a similar way quite quickly.

nohal, 2023-02-04:
There is a lot of good ideas here that could be implemented, I will create a RFE issue for them.

closed 2023-02-04 by nohal