OpenCPN/OpenCPN #2714: Idea to render charts in browser using WebGL 3D.

mgrouch, 2022-06-04:
It's radical but has many benefits. I've found a project which converts S-57 and KAP charts into some kind of JSON.
It's the code which can run in web server side and stream it to client. Vector chart data.

https://github.com/ilebedev/enc_parser

Then inside browser it's possible to write JavaScript code (actually JS code transpiled from Kotlin JS code).
I've done a set up for a skeleton project here: https://github.com/mgrouch/kotlin-sandbox/tree/main/kotlin-webgl
That code could use WebGL and chart data to render ENC (vector) charts natively in 3D with shaders, 3D fly-by modes for marina entrances 
and so on. The result will be very portable and modular. No more issues for building for multiple platforms and distributions.
Development in Kotlin is way easier and safer than in C++.

See also:

https://github.com/ilebedev/enc_parser/issues/2

What do you think?
Thanks

transmitterdan, 2022-06-04:
IMO this is not an enhancement request.  You are describing an entirely new project.

I imagine the performance of such a system would be very very slow.  I doubt it could run on a $150.00 Raspberry Pi at all.

But what does it matter what I think?  Build it and let's see how it runs.

seandepagnier, 2022-06-04:
doesnt avnav come close to doing this?

nohal, 2022-06-04:
Yes, there sure are other existing projects that are a little bit closer to fullfill a feature request basicallly saying "Reimplement OpenCPN from scratch as a web application". They also answer the other question "Is it really not that trivial?"

mgrouch, 2022-06-04:
> doesnt avnav come close to doing this?

not really. It just relies on tile server which sends raster png tiles.

mgrouch, 2022-06-04:
> IMO this is not an enhancement request. You are describing an entirely new project.
> 
> I imagine the performance of such a system would be very very slow. I doubt it could run on a $150.00 Raspberry Pi at all.
> 
> But what does it matter what I think? Build it and let's see how it runs.

I’ve just tried Little Tokyo example from here on my iPhone
https://threejs.org/examples/#webgl_animation_keyframes

It’s very responsive. 60fps. Rotations look real time. It’s similar hardware architecture to pi. 
Interestingly my pi4 is really bad with same. Only 1fps. More than 60 times slower. Something is misconfigured maybe.

rgleason, 2022-06-04:
Tried  Little China on S4 phone, yes very responsive, but how does this compare to charts. Charts have more objects and are far more complex.

seandepagnier, 2022-06-04:
The raspberry pi opengl driver never did live to full potential, and even if it did, the rpi hardware is not that good compared to mali for example.

So suggest pushing all the objects to render through a websocket?    Hmm......   Sounds also slower, but OpenCPN is not very clever in what it sends.   For example re-compiling a vector chart using something like a quad tree data structure for the vbo etc.. and now it probably runs faster than before even pushing everything through a (local) socket.

So it's possible..   I guess, but it would be really hard.  Probably wont happen.    It would sure be nice if opencpn were decoupled from gui/rendering/calculations/plugins.    All of these could use one or more separate threads,   multiple output sources integrate everything but provide it across the network.   Act as a tile server to other applications etc...

To achieve this?   Maybe start over, and pull bits and pieces of opencpn into the new application modifying each bit as you go?   It could take years, and I think no one cares  to do it.

mgrouch, 2022-06-05:
@seandepagnier 

that software in the first post already parses s-57 into json. First stage is in c++ and second one in ocaml. can be used for POC. 
Second software is Kotlin bindings to JS WebGL API. Very fast transpiler. 

There are some concerns about quality of WebGL on raspberry. How if we go this route it doesn’t matter. It will work in any browser on any platform with WebGL. Which is nowadays almost any. 

I’ve also tested same WebGL example on rpi4 with bullseye 64-bit and got 10 times better performance. 8fps. still not as good as on iPhone but much better than in buster. 

For POC we could just render bathymetry data of a chart in 3D shaders that would show if this makes sense or not. I’m more optimistic with what I’ve seen with JS and WebGL in modern browsers.

nohal, 2022-06-05:
It seems that it still was not clear enough, so I will say it as clear as I can "Rewrite OpenCPN from scratch as a Javascript application running in a web browser" is not a feature request for OpenCPN.

closed 2022-06-05 by nohal

mgrouch, 2022-06-11:
I've made better progress on this idea.
The idea is to render S57 ENC vector charts in browser natively using WebGL.
It actually looks quite promising. I took S57 parser and render in Java from OSM.
https://www.javatips.net/api/josm-plugins-master/seachart/src/s57/S57dec.java

I've converted it into Kotlin. (Both parser and renderer which uses AWT canvas).
Rnderer is coded to render all objects specified in chart. (Not by icons, but by actual code writing on canvas)
See here: https://github.com/mgrouch/kotlin-sandbox/tree/main/kotlin-s57
Parser is very very close to be running in browser using Kotlin JS.
I've tested it a little on JVM already.
Renderer should be possible to port from AWT canvas to WebGL in Kotlin).
Might be using https://github.com/revaultch/gwt-awt (converted to Kotlin)

Caching for performance in browser can be handled using IndexedDB of browser.
Might use: https://github.com/JuulLabs/indexeddb
or something else

Conversion of depth contour lines into 3D mesh using triangulation libraries in Kotlin can be done with:
https://github.com/slaviboy/DelaunatorKotlin
https://github.com/ricardomatias/delaunator

3D rendering or 2D with WebGL:

https://github.com/Pozo/threejs-kotlin
https://github.com/markaren/three-kt-wrapper

https://github.com/openrndr/openrndr

I've played with WebGL and 3D rendering of reliefs with it too in Kotlin
https://github.com/mgrouch/kotlin-sandbox/tree/main/kotlin-webgl

Rest of UI can be done using kvision https://github.com/rjaros/kvision

mgrouch, 2022-06-22:
Since then I've made better progress on this idea.
The idea is to render S57 ENC vector charts in browser natively using WebGL.
It actually looks quite promising. I took S57 parser and render in Java from JOSM.
https://www.javatips.net/api/josm-plugins-master/seachart/src/s57/S57dec.java

I've converted it into Kotlin. (Both parser and renderer which uses AWT canvas).
Renderer is coded to render all objects specified in chart. (Not by icons, but by actual code writing on canvas)
See here: https://github.com/mgrouch/kotlin-sandbox/tree/main/kotlin-s57
Parser is very very close to be running in browser using Kotlin JS.
I've tested it a little on JVM already.
Renderer should be possible to port from AWT canvas to WebGL in Kotlin).
Might be using https://github.com/revaultch/gwt-awt (converted to Kotlin)

Caching for performance in browser can be handled using IndexedDB of browser.
Might use: https://github.com/JuulLabs/indexeddb
or something else

Conversion of depth contour lines into 3D mesh using triangulation libraries in Kotlin can be done with:
https://github.com/slaviboy/DelaunatorKotlin
https://github.com/ricardomatias/delaunator

3D rendering or 2D with WebGL:

https://github.com/Pozo/threejs-kotlin
https://github.com/markaren/three-kt-wrapper

https://github.com/openrndr/openrndr

I've played with WebGL and 3D rendering of reliefs with it too in Kotlin
https://github.com/mgrouch/kotlin-sandbox/tree/main/kotlin-webgl

I've also ported NMEA, AIS, Seatalk1 parsing library into Kotlin as well
https://github.com/bareboat-necessities/kotlin-nmea

Rest of UI can be done using kvision https://github.com/rjaros/kvision

What do you think? I'll code more but, more people helping would be better.
Kotlin and IntelliJ really make it way easier than anything I could think of.

Thanks

seandepagnier, 2022-06-22:
opencpn is not really related to what you are talking about in any tangible way.

mgrouch, 2022-06-23:
> opencpn is not really related to what you are talking about in any tangible way.

Got it. Sharing ideas on other open source plotter implementation is not welcome here.