OpenCPN/OpenCPN #2729: Integrated NMEA2000 support for OpenCPN

douwefokkema, 2022-06-25:
Considering the fact that NMEA2000 became the standard for interconnection of nautical devices in the marine industry, OpenCPN should provide integrated support for NMEA2000. This is important for OpenCPN to maintain its position in the chart plotter marked. 

Current situation is that to connect OpenCPN to NMEA2000 (n2k) one should make use of a converter to NMEA083, adding the limitations off both. This connection is done in practice with the Twocan plugin or the Actisense NGW-1. Buy many n2k PGNs can not be mapped on NMEA0183 messages, if they can be mapped this involves decoding, encoding and decoding inside OpenCPN again. Plugins that need to communicate with the n2k net (like controlling autopilots) need to care for their own interface each of them.

Considering the above, n2k should get a role similar to NMEA083 inside OpenCPN. Users should be able to select n2k when adding a connection and select from an extensible list the PGNs he wants to receive and transmit. Plugins should be able to subscribe to (selected?) n2k messages just like we have SetNMEASentence() now. Internal format? Either json or canboat (much code available), whatever you like.

How to implement? I would leave this to the specialists. But my first thoughts would be to use part of twocan with its devices plus Actisense NGT-1 as a front end. Decoding inside O could be done with again twocan software or canboat. But still it will be considerable effort. 

Douwe Fokkema

Then there will be endless discussions that Signal K is better and is more widely used. OK, it is better, OK, it is widely used with hobbyists in very nice home-brew systems. But whatever you say, NMEA2000 is THE standard in the marine industry and better accept that. 

Douwe

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

bdbcat, 2022-06-26:
Let us break this down a bit.
1.  Software is, well, software.  Everything inside OCPN is under our direct control.  So I see no real issues (aside from manpower/hours) in implementing integrated N2k support to a defined spec.
2. Hardware:  I am not current on N2K hardware and physical layer. Is Actisense NGT-1 "_the_" preferred gateway in the real world?  What about the various can-bus hats for rPi?  Who can teach us all about N2k hardware, please?

rgleason, 2022-06-26:
Steve Adler wrote:

 Jose Luis is clearly very knowledgeable on this as he has been able to develop a commercial product. 
[Ocenav communication conversion tables https://ocenav.com/en/home//download/3982/ ]

Other useful tables for describing conversions between NMEA 2000 PGN's and NMEA 183 sentences are available from Actisense https://actisense.com/wp-content/uploads/2020/01/NGW-1-Full-Conversion-List-v2.660.pdf and from Maretron https://www.maretron.com/support/knowledgebase/phpkbv7/assets/USB100UM_1.2.pdf

TwoCan currently performs all of the conversions necessary for OpenCPN to function with a few extras that OpenCPN can't handle such as Man Overboard and some of the Route & Waypoint import/export. I could probably generate my own table of conversions, but I'm sure you'll appreciate that documentation is probably the last thing to be written.

I think there are two major issues arising from the recent threads regarding autopilots.

The first scenario is a user with an NMEA 2000 autopilot who controls the autopilot from its control head (autopilot display, MFD); turning it on, selecting the mode of operation and in this case, following a GPS course. They use a NMEA 183 to NMEA 2000 gateway such as Actisense, Maretron, ShipModul, Yacht Devices, SignalK, TwoCan etc. to perform the conversion from NMEA 183 to NMEA 2000. They then activate a route or waypoint in OpenCON and expect OpenCPN to send the correct data to enable the autopilot controller to follow the course. The data traditionally consumed by NMEA 183 autopilots, and generated by OpenCPN is any of APB, RMB or XTE sentences. The gateways construct PGN 127237 from the data in APB, RMB or XTE sentences which is then used to send steering information to an NMEA 2000 autopilot controller.

 It seems as though one major problem that has been raised in the forum is an OpenCPN bug in the generation of RMB, with incorrect origin & destination id's. There may also be some bugs in the conversion of the NMEA 183 sentences to NMEA 2000 messages but I have no solid data on this. I unaware of any users of TwoCan using it to perform this function, and without end user verification testing I can't confirm whether TwoCan constructs PGN 127237 correctly for all use cases.

The other scenario  is a user with an NMEA 2000 autopilot who wants to control the autopilot fully from OpenCPN. Meaning that turning it on, selecting the mode of operation are all performed from OpenCPN, without the need for an autopilot control head. This is more complex as this is where the autopilot vendors use proprietary NMEA 2000 PGN's to control the autopilot. Most of this has been reverse engineered at least for Raymarine and Navico (B&G/Simrad) autopilots.

Douwe's plugin controls Raymarine  autopilots albeit only on WIndows using an Actisense NGT-1 adapter. We've started some collaboration to extend this support more widely. I've almost completed building support in TwoCan for both Raymarine & Simrad and I think SignalK has done this for Raymarine. Hopefully someone can provide logs so that we can reverse engineer the Garmin autopilots. In my opinion, by the end of the year much of this will be completed and the ability to control an autopilot from OpenCPN will be realised.

Regards...

bdbcat, 2022-06-26:
Autopilot is a specific use case.
Lets try to step to a higher level, and talk about the more general architecture of N2K within OCPN.
My sticking point is the physical layer.  How will Joe user attach a PC (or rPi) to an existing N2K backbone.

keesverruijt, 2022-06-26:
I think @bdbcat is right -- autopilot is a specific use case, and one that is probably better of using a different approach. I will create a different issue for that as I have some ideas.

As I reverse engineered (most of) the N2K PGNs, and wrote a number of hardware interface programs, e.g. the original [canboat](https://github.com/canboat/canboat) project, I am happy to add my 2 cents here.

As to hardware, the Actisense NGT-1 was the only realistic interface ten years ago for a layperson to use, as it made sure your N2K network isn't messed up no matter what the software on the computer side did, and it was the only one out there. At the moment there is some competition, mainly Digital Yacht iKonvert. The other solutions are a direct CAN interface or one connected via USB.  For _reading_, such a CAN interface is fine as the interface can just stay passive. For _writing_ you need to handle the entire J1939 address claim stuff. This is not unheard of, there are projects that we could use to copy this, but it is more work and it means a much more involved piece of code. For Linux, accessing a direct CAN interface can be done using a standard driver in the Linux kernel. I don't think such a thing exists on macOS or MS Windows; on those it would be easier to restrict us to USB connected devices. The problem with the USB devices is that they are all different, so support might need to be added for each individual hardware solution. Some USB devices have drivers on Linux that make them act as directly attached CAN interfaces. This is what I use for my Mastervolt Masterbus interface, which uses CAN as well.

The Digital Yacht iKonvert is interesting, because they have an opensource definition of the protocol and encourage open source developers to work with their hardware. 

I was going to look up how Rosepoint handles this, and found out that they now make an ["NEMO"](https://www.rosepoint.com/nemo-gateway/) interface box that supports NMEA0183 and 2000. 

Since we know that Joe User will be using 80% Windows, 10% macOS and 10% Linux (and 80% of that RPi) I think we could get away with the following: write a plugin that reads/writes messages to a new OpenCPN API layer. The plugin supports NGT-1 and iKonvert. This needs to be a plugin, so it is easier for other devs to experiment and add new interfaces like the Rosepoint Nemo, direct CAN or specific CAN USB devices. If it were base OpenCPN code, it would be a little harder for a developer to get going and set up a dev environment? There could be a plugin per hardware device or a single one that allows you to choose what hardware you have, like the radar plugin.

keesverruijt, 2022-06-26:
I created a new [issue](https://github.com/OpenCPN/OpenCPN/issues/2730) for generalised autopilot support...

keesverruijt, 2022-06-26:
I think before we go any further, I'd like to know how the Signal K support actually works within OpenCPN?

Hakansv, 2022-06-26:
I'm using the GeDab MCS board and a Rpi with SignalK for general NMEA2000 inputs. Haven't yet tried any output. 
The cost is about €80 for the board and about the same for a RPi. To be compared to around €180 for a NGT-1_USB.
(Openplotter does no longer support this board for various good reasons but there's a SK plugin that works well.)

keesverruijt, 2022-06-26:
FWIW I use the open source (CANable Pro)[https://canable.io] adapter as it has isolation. I will not tolerate non-isolated devices on my bus with too much expensive hardware!

As they say in Germany "Wir von WC Ente empfehlen WC Ente" (We from WC-Duck recommend WC-Duck). In other words we _could_ just reuse the entire codebase from canboat which is in C. This would involve starting a pipeline (1) -> Actisense-serial -> analyser -> n2kd -> back to (1) and then OpenCPN would access a network port from n2kd and read JSON formatted PGNs from it. This is much lighter weight than Signal K. Right now I think one small piece is missing in canboat as it doesn't yet format explained N2K PGNs back to raw JSON containing just the bytes for writing. An alternate is to move the PGN translation part into O, and just have the reading/writing to the physical layer done by CANboat.

rgleason, 2022-06-26:
Kees wrote:
`I was going to look up how Rosepoint handles this, and found out that they now make an ["NEMO"](https://www.rosepoint.com/nemo-gateway/) interface box that supports NMEA0183 and 2000.`   nice equipment, about  $699

tehani's  [Ocenav ATM103-A3 or C3](https://ocenav.com/en/home/) (takes awhile to download)  does similar nmea0183, nmea2000, seatalk, has Wifi and controls autopilots and is less expensive.  Jose Luis is working on a very simple and much less expensive version right now.

[iKonvert ](https://digitalyachtamerica.com/product/ikonvert-usb/) about $200 I don't find the opensource Kees mentioned, is it the [signalK iKommunicate](https://digitalyachtamerica.com/product/ikommunicate-nmea-signal-k-gateway/)?

keesverruijt, 2022-06-26:
iKonvert is _not_ iKommunicate. The SDK for iKonvert can be found on GitHub (digitalyacht/iKonvert)[https://github.com/digitalyacht/iKonvert] (obviously :-) )

TwoCanPlugIn, 2022-06-26:
Re: "Hardware: I am not current on N2K hardware and physical layer. Is Actisense NGT-1 "the" preferred gateway in the real world? What about the various can-bus hats for rPi? Who can teach us all about N2k hardware, please?"

There are two types of adapters, dumb & smart. And within each category different capabilities from an electrical & isolation perspective.

Smart adapters such as the NGT-1 can independently connect to the NMEA network by performing the ISO Address Claim, handle address claim conflicts etc. and with respect to NMEA 2000 messages, automatically re-assembles the fragments of fast messages. It can also be configured to filter the reception & transmission of different PGN's. I personally have no hands experience with one, other than writing an Actisense plugin that used some of the TwoCan code base. It was never released and the code is languishing on github. The commands used to control the NGT-1 have been reverse engineered (originally I believe by Kees) and used by many including Douwe in his autopilot plugin. There is also a commercial library available from Actisense.

Dumb adapters on the other hand do not inherently perform any address claim functions or re-assembly/fragmentation of fast message fragments. If you have a look on the [TwoCan OpenCPN Manua] (https://opencpn-manuals.github.io/main/twocan/0.1/index.html )l you will see some examples of supported hardware. The software (in my case TwoCan) does all of the address claim stuff, and reassembles fast message fragments. 

From a software API perspective it is a little fragmented. On Linux (including Raspberry Pi) pretty much all of the CAN bus adapters support SocketCAN, either as native hardware devices (CAN), SerialCAN devices (SLCAN) and even Virtul Devices (VCAN). SocketCAN simply means you can read & write to the CAN network using standard sockets programming. 

For Windows & Mac there is no real equivalent. On Windows, many of the adapters provide a library (Eg. Kvaser and Rusoku), and others (Canable, Peak, Axiomtek) which are essentially Serial CAN (SLCAN) devices have simple commands written to the com port. On Mac, it is similar although there is a third party project (Mac-CAN) that provides drivers for Kvaser, Rusoku and SerialCAN adapters.

TwoCan supports the dumb adapters on all three platforms (Mac, Windows, Linux) for three major reasons. Firstly the NGT-1 is expensive, secondly I don't own one on which to test & develop and thirdly, as it does the address claim dance & handles fast message re-assembly & fragmentation it would have required TwoCan to have separate code paths.

rgleason, 2022-06-28:
Douwe's [AutoTrackRaymarine](https://github.com/rgleason/AutoTrackRaymarine_pi) built Windows, but fails with SerialPort.h and  NGT1Read.h  "HANDLE" not declared.   Is this due to the N2K issue? , but shouldn't it build for use with Artisense?

bdbcat, 2022-06-28:
Rick...
Not related to this issue.  We are speaking here of more general, higher level questions about OCPN and N2K in total.
Please move your issue to the specific plugin issue list.

BerndCirotzki, 2022-06-28:
Hello,

I think I need to join your discussion. And I have the following comments:

Unfortunately, the TwoCan plugin does not translate !!! all PGN required for OpenCPN according to NMEA0183. The entire AIS area is missing and a lot more for dashboardTactics.

It is always necessary to have an interface between OpenCPN and N2k. that is clear.
But that's exactly what's finished!

All that is brought is an Arduino with ESP32 Devboard, or Atmega2560, a CAN bus controller and a NMEA2000 connector. 20€ ! that' all.
Timo Lappalainen has the software for that ready.
The Timos example program "ActisenseListenerSender" is sufficient and all NMEA2000 data are sent to a COM port in Actisence format
transfer.
Everything is described : https://github.com/ttlappalainen
  
Now OpenCPN only has to internally translate all (binary) N2K messages from the COM port to NMEA0183 or process them directly.
The microcontroller handles all communication with the NMEA2000. In opencpn you don't have to worry about that anymore.
OK, that's a lot of work .... and in the first version definitely error-prone.
If OpenCPN has that integrated, all N2k <-> NMEA0183 translators are no longer necessary.
Neither does SignalK, by the way. ... I never understood the meaning. The world outside of open source plotters now speaks NMEA2000.
All necessary PGN for autopilots can be easily integrated. For example, a single plugin for each N2k autopilot type (Raymarine, B&G, ...).
   
greeting

Bernd

sailoog, 2022-06-28:
Just my 2 cent. SocketCAN is already in Linux Kernel and that makes solutions based on chip MCP2515 (and successors) connected by SPI the standard in embedded systems like Raspberry, you can find really cheap hardware for that.

And my brainstorming. Since finding standard cross-platform hardware will be impossible mission, how about ignoring the hardware and having OpenCPN send PGNs over TCP/UDP? small and specific pieces of software could convert this data to the different formats that the different hardware alternatives require. Smartphones and WiFi devices like the SP32 could be easily fed too.

BerndCirotzki, 2022-06-28:
OK,
but unfortunately that is not the issue.
OpenCPN does not send PGNs.
OpenCPN still doesn't understand NMEA2000.... That's the issue!
It doesn't matter whether the data comes via COM or via TCP/IP.
OpenCPN has integrated SignalK. Does anyone use it?
My AIS receiver sends NMEA2000 not SignalK. The Wind Sensor also, and Depth .... and OpenCPN doesn't understand.
GPS transmits via NMEA2000, or NMAE0183. I don't have to spend a lot of time translating that into Signalk to use it then. 
what nonsense.
So OPENCPN .... finally accepted!!!!, the NMEA2000 is the standard! NMEA0183 is old, and SignalK has not prevailed.
There is a big Libray in OPENCPN that receives and sends NMEA0183. .... Why is there No Libray for doing the same with NMEA2000 PGN's
That is what I mean.
Take the NMEA2000 Data from COM, or ... if you like, from TCP or from whereever.
Use the library from Timo. I thing he would help. There is a lot of translation in his code that could be integrated in OPENCPN.
Parsing PGN's and so On.

And then it would be able to send an NMEA2000 Message direct from OPENCPN .. like there could be send NMEA0183 messages.

And in in Plugincode. ... WANT_N2k_ ! to receive these N2k Messages.
And it would be able to send an N2K Message to a Raymarine or B&G Autopilot. .... for to Go to "Standby" or or whatever.

the necessary Hardware I discribed.

leamas, 2022-06-28:
@bdbcat wrote
> My sticking point is the physical layer. How will Joe user attach a PC (or rPi) to an existing N2K backbone.

It is perhaps so that this question hasn't a general answer. Different platforms has different conditions. And some users tends to focus on commercially available solutions which are more expensive while others focuses on the typical DIY hardware. Still, we certainly need a common sw interface to the different hardware solutions.

@sailoog wrote (sort of):

> Using TCP/UDP?

Looks good at a glance, but this means a separate process and a wire protocol between it and main OpenCPN.  Perhaps not that complicated for basic functionality, but in my experience things tends to get complicated when handling errors and other kind of abnormal events.

@keesverruijt wrote:
>  This needs to be a plugin, so it is easier for other devs to experiment and add new interfaces

The problem with this approach is that if might create dependencies between different plugins, like one for communication and another for an autopilot. The plugin system does not really cope with this. Could of course be added, but we have been trying to avoid this complexity, IMHO for good reasons.

BerndCirotzki, 2022-06-28:
I must not be done everthing in time ! Begin it. First Beta Versions that can do.
And the good thing is, you can send whatever you want!  :-)

dmtamsen, 2022-06-28:
today - not expensive nmea2000 “bridges” from shipmodul, activesense, yachtdevices do already let you send/recive nmea2000 cross tcp/udp - ofcause the specific device has to understand THAT specific PGN’s - and on top of that, they translate know pgn’s to nmea 0183 commands.

dmtamsen, 2022-06-28:
> No !! You undersand the Bitsteam from NMEA2000 not any translator

elaborate ?

BerndCirotzki, 2022-06-28:
Yes, but most ist in the Code from Timo integrated. he translated a lot, but this must be done in OPNECPN direct not in the ESP32. Then The Hardware is only the Gateway. Only take a look. Only take what you realy know, and send what you want.
... all known PNG's are discribed in canboat/pgn.h .... I Thing everyone knows it here.

dmtamsen, 2022-06-28:
let me elaborate - nmea2000 is unfortunately and brand specific proprietary protocol - meaning - the brands doesn’t have to follow all the same commands for f.ex. controlling an nmea2000 autopilot…. they use different pgns for heading, tracking, etc… that said - the nmea2000 protocol is way more advanced - and devices on the nmea2000 bus know what each other do..

sailoog, 2022-06-28:
> OK, but unfortunately that is not the issue. OpenCPN does not send PGNs. OpenCPN still doesn't understand NMEA2000.... That's the issue! It doesn't matter whether the data comes via COM or via [...]

Some quick answers. 

If OpenCPN does not understand NMEA 2000 yet (the easy part) is because the hardware part is not solved and that is precisely the issue.

Yes, thousands of Raspberry users use the integrated Signal K connection.

dmtamsen, 2022-06-28:
> Yes, but most ist in the Code from Timo integrated. he translated a lot, but this must be done in OPNECPN direct not in the ESP32. Then The Hardware is only the Gateway. Only take a look. Only take [...]

I understand - but translated “alot” is not good enough :( - Garmin is very bad as documenting their pgn’s - and we THINK that Raymarine - has been transtalated. - well yes - the once we know… but everything?

bcn2, 2022-06-28:
Design decisions to be taken are (not a complete listing for sure):

- use raw N2K input, no 0183 translation? There is no 1:1 correlation between NMEA0183/IEC61126 objects and N2K objects. The Actisense raw format may be another candidate to support (if I interprete the information from SteveAd and Kees correctly)
- which PGNs are supported,  which do OpenCPN need (internal data objects), what to do with objects not required by OCPN, but by some plug-in (now or in the future)?
- function to relay raw N2K nessages?
- vendor PGNs for example case of the Autopilots shall be handled by dedicated plug-ins?

With the work done by Kees, Timo and Steve, almost all standard PGNs shall be reverse engineered and covered. "Just" to merge all that code together...

Dave's pending question: what Gateway will a user plug in and how to deal with this? How is the interface of those device(s) for the different architectures? Does the gateway bring its own installer or do OCPN have to provide some "glue"? For which one? What can we reasonably provide from the OCPN side.

Hubert

dmtamsen, 2022-06-28:
My point is - all standard pgn’s are already translated by many nmea2000 bride providers - like shipmodul, activesense, yachtdevices.
ofcause - those translations should be brought to opencpn….

BerndCirotzki, 2022-06-28:
(which PGNs are supported, which do OpenCPN need (internal data objects), what to do with objects not required by OCPN, but by some plug-in (now or in the future)?)
... This can be specified as a requirement ! 
But I thing this is the way.

BerndCirotzki, 2022-06-28:
YES !!!!

BerndCirotzki, 2022-06-28:
(My point is - all standard pgn’s are already translated by many nmea2000 bride providers - like shipmodul, activesense, yach tdevices.
ofcause - those translations should be brought to opencpn….)

but when you do this youyself, you do't need them any more. Ok, I don't know, if they like that, but is that a reason ?
But the best is, OPENCPN can send to anyone in NMEA2000 whatever it wants. An that makes sense.

leamas, 2022-06-28:
> Dave's pending question: what Gateway will a user plug in and how to deal with this?

Indeed.

My idea is that  we need to create common API for sending/receiving N2K, and glue code which interfaces the different solutions (there is obviously not a single one). This is probably the hardest part to get right. would be about completing something like below. The API  is where the format used is set (raw NMEA2000, NMEA0183, SignalK, some kind of mix?)

   
    -----------------------------------------------------------------------------------
    |                               N2K Device API                                    |
    -----------------------------------------------------------------------------------
    |   ActiSense    |  ttlappalainen stuff   |   SocketCan   |    ...                |
    ------------------------------------------------------------------------------------

The translating stuff is next level above the API. IMHO, we know how to cope with this. The important point is that it must be independent of the hardware layer.

bdbcat, 2022-06-28:
@leamas...
Starts to look like a traditional "network stack".  Clear lines between application->protocol-> link->physical.
For sure it makes sense to isolate the physical interface, so that the various transport mechanisms can be transparently glued into place as needed.
We should try to read some code (Timo's is a good start), and think about APIs for the various layers on the stack.

transmitterdan, 2022-06-29:
Does it make sense to create internal messages within O & plugins that are proprietary instead of using NMEA0183 format? Then we support each boat network protocol with a translator to/from this internal messaging structure. This removes the burden from O and its plugins to decode whacky strings and PGNs.

O could then adopt a pure IP message passing between the core and the translator software. Instead of compiling the translator(s) into O, a translator app could be run in parallel that sits between the lower layers and O or plugins. O just opens a socket to an IP+port and it gets/sends only the data it cares about.

O need never know about NMEA2000 or SignalK or any other message encoding scheme. Let this translation happen outside of O and the comms to O be via Sockets. At that point O cares not a whit about any of this vendor specific stuff. And anyone could write a translator without any GUI code. There could be a Garmin, Raymarine, SignalK and any other vendor specific translator that runs in parallel with O. If it communicates with O via sockets it can reside on the same computer or somewhere else on the IP network.

What I worry about is if we start encoding/decoding NMEA2000 inside O then we will inevitably get calls for doing vendor specific stuff like AP control or calibration sequences. O will never really be stable.

O only needs a handful of data elements to do its job. The real user of the many diverse elements of NMEA2000 is plugins. With an IP based solution, plugins could get their data directly without O involved. This is how the radar plugin works. O and its plugins could connect to the translator app(s) directly thus relieving O of a lot of needless work.

leamas, 2022-06-29:
@transmitterdan  wrote

> Does it make sense to create internal messages within O & plugins that are proprietary instead of using NMEA0183 format?

Two souls, one thought. I have had exactly the same idea.

TwoCanPlugIn, 2022-06-29:
[BerndCirotzki](https://github.com/BerndCirotzki) commented 

Re: "Unfortunately, the TwoCan plugin does not translate !!! all PGN required for OpenCPN according to NMEA0183. The entire AIS area is missing and a lot more for dashboardTactics."

Would you be kind enough to either send an email or raise an issue on the TwoCan repository. Always happy to receive feedback and to add new features.

For AIS, TwoCan provides bi-directional support for the following PGN's
129038 - AIS Class A Position Report
129039 - AIS Class B Position Report
129040 - AIS Class B Extended Position Report
129041 - AIS Aid To Navigation (AToN) Report
129793 - AIS Date and Time Report
129794 - AIS Class A Static data
129798 - AIS SAR Message
129801 - AIS Addressed Safety Related Message
129802 - AIS Broadcast Safety Related Message
129809 - AIS Class B Static Data: Part A
129810 - AIS Class B Static Data Part B

There are a few that I know of that are unsupported; 129795 - AIS Addressed Binary Message and 129797 - AIS Broadcast Binary Message mainly because I have no sample data, and also because I am not sure if or how OpenCPN consumes them.

Apologies to all for the slight hijack of this thread.

BerndCirotzki, 2022-06-29:
Now I installed TowCanplugin 2.1.0.0 ,

... but I only find AIS 129038..41 ? for selection.

bcn2, 2022-06-29:
Some more numbers:

OpenCPN: Distribution by OS (base o-charts, 22.000 identified individual system, rounded figures)
Windows 50%
Linux 25% (thereof 16% RasPi/OpenPlotter)
macOS 10%
Android 10%
Rest Dongles/system not identified

Btw: what would happen with N2K for Android??

bdbcat, 2022-06-29:
@TDan
"Does it make sense to create internal messages within O & plugins that are proprietary instead of using NMEA0183 format?"
Perhaps.  But then we come to the question of what proprietary message format to use, and how to design it.  A format/protocol that is fully "future-proof" will look a lot like sK.  So, for the sake of discussion, why not use sK directly?

Hakansv, 2022-06-29:
> So, for the sake of discussion, why not use sK directly?

That was my first thought reading this thread. Why invent a already running wheel that cost very little CPU power?
What I can see most hardware support is already there, see fig. For my MCS board and for other similar the can0 interface is used.
Even though not tried live by me SK is available also for Windows. A Openplotter Rpi is though a cheep solution for all physical connections that can (IP)-serve whatever system around. 

What O can need is way to understand SK-autopilot messages and serve a NMEA0183-AP. And vice verse to send out an active route messages, via SK or direct, to produce PGN 129283/129284/129285 to a N2k net.

SK data NMEA2000 connection:
[image]

bcn2, 2022-06-29:
And if there is the hard coded POD structure in the release and an optional list of "load on start" objects in an external file for new or additional objects? The content of the latter can be integrated - if the object is required/an add-on/new idea - in the next release. Or one has just "private objects".  Two stages.

leamas, 2022-06-29:
I envision a more elaborated thing, a tagged union like 
```
typedef enum { gps_fix, cog, waypoint, ...}  message_tag;
typedef struct gps_fix { float lat; float lon; ...}
typedef struct cog {float speed, float direction, ...}
typedef struct waypoint  { float lat; float lon; ... }

typedef struct ocpn_message_ {
  message_tag tag;
  union {
     struct gps_fix gps_fix;
     struct cog cog;
     struct waypoint waypoint;
     ...
} ocpn_message;

```
And so on. An object like this will not change memory layout as long as  the size does not change. This can be achieved by adding i really long union member. This would make runtime work even if there is new data types are not known by main OpenCPN or plugins. To really work with new data types recompilation is needed, personally I'm fine with that.

rgleason, 2022-06-29:
Dumb question but could SK provide the POD, kind of built from its own structure?

bdbcat, 2022-06-29:
TDan...
Let me make sure I understand the overall architecture of your proposal. (sure would be easier in person with a whiteboard, but whatever...)
Correct me if wrong, please, in this simple description.
1.  For the simple NMEA and sK data streams that OCPN core understands today, he will package the data into a tagged sruct and "serve" it to any plugin requesting.  This is how a new/improved Dashboard plugin would get basic nav information.
2. OCPN core will also listen at a defined TCP port for plugins (or other apps) that would like to send a compatible struct to OCPN.  This is, for instance, how an N2K plugin would talk to OCPN core.

Am I on the right track?
Dave

p.s.  Please, nobody panic.  None of this should be taken as direction, merely discussion and scoping.

transmitterdan, 2022-06-30:
Dave,

Yes, we need a whiteboard and some aperitifs.  Sorry for delays, I have been busy and only get online about an hour each day.

I was first just thinking about how to get data in various formats (0183, 2000, SigK, etc.) translated into a data structure understandable without further parsing by O and plugins.  O has no need to understand NMEA2000 PGNs IMO.  It just needs to exchange some nav data.

Move the parsing/translation functionality to dedicated apps that focus on a single protocol flavor possibly unique to one vendor (e.g. Garmin).  O would document a common set of interface data structures that all "translators" would need to know about.  Then for example, CoG, whether from SigK/0183/2000/etc. would look the same to O only with a unique source descriptor & time stamp that could be used to establish priority if multiple sources exist.  The exact format of the data structure isn't too important right now but since translators have to convert from/to something it seems reasonable to use an efficient binary structure that could be used directly by C++.

I was thinking of getting O out of the loop for plugins that want nav/boat data.  They could just get it straight from the translator(s).  O need not know about what nav/boat data the plugins want.

In this scenario, O exchanges internal data (canvas handles, etc.) with plugins using an efficient internal format pretty much as it does now.  Maybe no changes in this regard are even needed.

None of this requires throwing away the existing O code base.  Just add another "connection" type that implements talking to translators.  This creates a flexible adoption path since I think it will take years for the majority of uses to move away from NMEA0183 to NMEA2000.

It wouldn't be too hard to write an NMEA0183 translator as an example.  I was thinking it could be written in Python which would make it platform independent (except I am unsure how Android supports Python).

Obviously, translators don't have to run on the same hardware as O.  Translators could reside on any computer on the IP network.  For example, a RPi could host translators along with CAN hardware and Android O could connect to the Pi over IP.  This could be the preferred way to give Android O access to NMEA2000 network data.

One small challenge is to document the limited number of data structures O is interested in.  There aren't very many of these.  The bigger challenge is to document all the data structures plugins are interested in.  There are lots of these.  Evolution of translators<->plugins interface does not affect O whatsoever unless O becomes interested in some new data type.

If we wanted to have a Zoom or Teams meeting I can facilitate as I have good access to these.

Regards,
Dan

bdbcat, 2022-06-30:
TDan...
Taking the specific case of N2K.
Assume:
1.  There is a plugin that "owns" the N2K physical interface, whatever that is.
2. OCPN core is listening at some socket for the proprietary OCPN data structure to come in, from any source.
3. Other plugins (Dashboard?) may also be listening for the OCPN data struct at a defined socket.

Why is there an efficiency in having a separate "Translator" process running?  Would not the N2K plugin be best positioned to simply parse N2K, and post the resulting OCPN data structure back to the core socket and other plugins?  That is, would not the translator logic live in the N2K plugin naturally?

br
Dave

transmitterdan, 2022-06-30:
Dave,

I was thinking of a way to do this without linking the O executable to the translator.  Unlike plugins that must be linked to O there isn't any need for that in this case.  The translator standalone app owns the N2K interface.  It publishes the data via a socket to which anyone can connect (O, plugins, some other unrelated app). The "efficiency" is in decoupling the translator feature from the O code.  Translators are standalone projects.

bdbcat, 2022-06-30:
TDan...
OK, I see.
Hmmm... We went to a lot of effort to move the plugin management UI to user-space, closely integrated into OCPN core.  How will Joe-user get and install the correct translator app (for upwards of 20 platforms now supported x Y protocols)?  I think this might be a step backwards in usability, even if architecturally satisfying.
Other opinions?
Dave

douwefokkema, 2022-06-30:
From a simple user perspective, I would like to add a N2k connection just like any other connection. And select the interface from a (dropdown) list of available interfaces. No puzzling around activating plugins. Considering this, I would put the interface code not in a plugin but in the main O code. This should not be a problem for developers adding an additional interface, coding for O resulting in a PR is easier than making a new plugin.

As a plugin writer I will need access to a choice of navigational data, same way as we have now, like SetPositionFixEx(), SetActiveLegInfo(), and messages generated by events in O. On top of that, I would need to receive the proprietary PGNs that my device (autopilot) transmits. These PGNs do not need to be understood or interpreted by O, that is the job of the plugin. I also need to compose and send proprietary PGNs, with a payload exactly as I define.

TwoCanPlugIn, 2022-06-30:
Many interesting ideas. Keep in mind that decoupling OCPN from NMEA 2000 adapters and translators may impact usability; where and how do users install & obtain them, how does one diagnose problems  and how to debug. 

There are enough posts on the forum related to networking that we don't want to create a support burden.

For data exchange you could go from human readable text (Signal K JSON with metadata) through to binary protocols such as ASN1. The former would be easier for novice plugin developers to use, the latter would require us old bastards to search our archives of old code!

The question is hiw to make this dynamic at runtime and not have to recompile every time there is some new data. Canboat could be modified to build a dictionary from its pgn xml file. The pgn.json has a problem in that a pgn can only have one definition which breaks when manufacturers use the same proprietary pgn (eg. 130820 IIRC is used by many vendors).

Douwe's comments make sense for OCPN to inherently support N2K adapters (easy for Linux), you'll have select which ones for Mac/Windows.

But how to architect the connection to the "translator" without adding excessive latency.

I disagree that autopilot plugins need to compose NMEA 2000 pgn's, Does that mean Douwe will also have to understand Garmin, Simrad proprietary PGN's or is his plugin limited to Raymarine users?

leamas, 2022-06-30:
I wonder if we are making this too complicated. The first issue is the use of a socket interface and a separate process for decoding. Yes, there are advantages, but it's a more complicated solution and as Dave pointed out it might affect usability.

OTOH, we IMHO do need some serious design here.  A core issue is the the need to create separate modules for N2K devices like autopilots etc. Let's call such a module a "driver".  My ideas circles around some kind of "semi-plugin" approach where drivers are truly separate entities which are registered like plugins and made available in the "N2K Device API". This would strike a balance between the need for a clean interface when coding a driver while still keeping it within the same process.

In order to keep the discipline here many things we are used to should not be used in a driver. This includes global variables and direct calls to the GUI. In short, stick to a sound API.  By adhering to such principles I think it would still be reasonably easy to create and maintain drivers even if running in the same process. It would require devs to re-compile OpenCPN, but I think we could  live with that (if it's too hard, let's make it simpler!)

leamas, 2022-06-30:
The other issue is the perceived need of making translations dynamic in runtime: @TwoCanPlugIn wrote: 

> The question is hiw to make this dynamic at runtime and not have to recompile every time there is some new data

As I see it, there are just two kind of messages.

The first is those core OpenCPN "understands" so to speak. These could be decoded completely to  binary representation, something like [above](https://github.com/OpenCPN/OpenCPN/issues/2729#issuecomment-1170159760). No need for anything dynamic here, these messages are set in stone when compiling.

The second are those core OpenCPN does not understand while a plugin still might use in some way. For these, the core should basically just act as a router in the network stack. This means handling of defragmenting and addressing.  However, the message payload is just a blob in the internal ocpn_message structure, a blob which hopefully can be interpreted by the receiving plugin.

Since the overall N2K message structure is known, the core should still be able to for example let plugins subscribe to specific PGNs.

Using this (too?) simple approach, I actually don't see any need for some kind of dynamic adding of message decoding.  What am I missing?

leamas, 2022-06-30:
> The solution (and best way forward IMO) is a clean API between O and the rest of the ecosystem

Again, we are thinking in similar directions. However, focusing on the plugin API: if you open a separate issue I would be happy to share my thoughts with you and others on this really important issue.

Going back to the N2K stuff, my idea is that we have a clean API between the drivers and the rest of opencpn, but still makes as a traditional C++ API. Despite the current plugin API status, clean C++ APIs are not unheard of. A traditional API is IMHO just simpler to design and implement.

leamas, 2022-06-30:
> N2K is really hard because few OpenCPN computer platforms support the necessary hardware directly. Some physical adapter is almost always needed. That makes supporting N2K complex from a "clueless user" perspective. Nothing we are talking about will change that.

Indeed. But as you say, nothing we do can change that. Well... we could document how to use some adapters. Given enough man-hours we could also autodetect them and do the Right Thing. But we probably don't have the man-hours for that...

mgrouch, 2022-06-30:
Examples for bus would be D-Bus, and for serialization protobuf.

TwoCanPlugIn, 2022-07-01:
Re: “ Given enough man-hours we could also autodetect them and do the Right Thing. But we probably don't have the man-hours for that”, 

For Linux, perhaps unnecessary, SocketCAN is pretty simple, especially for Pi, something in boot.txt for the hardware adapter (the h/w vendors document this) and either a udev rule or IIRC something in init.d.

For Windows & Mac, TwoCan autodetects the supported adapters, so new code is only required if new adapters are added. Again no big deal.

I think we are overcomplicating the issue wrt to hardware interfaces, there is probably an order of magnitude more users asking how to connect USB GPS devices, or how to configure their TCP/IP network settings for NMEA over WiFI/Ethernet, than I receive regarding connecting CAN Bus adapters. IIRC Only two questions ! 

(BTW why default to 2947 when the IETF standard port for NMEA is 10110 ?)

And also as someone else has commented in the OCPN has a reasonably finite & documented set of data that it needs: position, heading, speed (SOG, STW), COG, Wind, Depth, AIS & DSC. It is the other plugins that have an interest in other blobs of data.

If plugins or whatever it is that we want to call them are made available as separate apps, what about usability (distribution/installation, support/debugging) ? Are we perhaps making a noose for ourselves. How do we interconnect OCPN and these plugins (IPC, RPC, REST, sockets, message queues, for Windows perhaps COM, etc.)

At the end of the day it is software that many of us can contribute towards, but the plugin manager, CI build process and the expanding number of supported platforms is creating a rather large hurdle for many plugin developers (or maintainers, kudos to Rick). Do we want to make that hurdle larger ?

I am off sailing (finally), so may be unable to offer any more comments For a few weeks.

ttlappalainen, 2022-07-02:
I have been working over 7 years with N2k and few past years as professional.

OpenCPN is good product so I hope that possible tighter N2k connection would be done properly so that it could be serious choice for any boater and not just some group of hackers. 

There are two possible proper way:
1. Use some certified gateway e.g., Actisense NGT-1, Yacht Devices YDNU-02RF, Maretron USB-100.
  This method is free of any charges for OpenCPN. Other devices on N2k bus sees OpenCPN as selected interface manufacturer device. Actisense protocol is know and simple. I have no idea of Maretron protocol.

2. Make proper registered OpenCPN gateway.
The hw costs would be <200 $ - I know at least two companies, who could provide that as ready device. Registration costs for OpenCPN would be about membership 800 $/y + certification 1500 $.

Starting now with 1. does not close that it could be later moved to 2.

Connecting to N2 with sw like OpenCPN can be compared to normal traffic. Single driver can drive on his own yard with unqualified / uninspected car ( actually even this would break law at least in Finland ). But if you go to public traffic, you have to have proper car and follow rules. You have to use turn signal properly and drive right side. So is with N2k. If OpenCPN starts to provice support to connect to N2k, it should follow rules. Single users can do what ever they like and build own gw with my library (there is Actisense example), but as public it should provide proper way. So at this point I would drop off support fo all unregistered CAN gateways.

When OpenCPN has been connected somehow to N2k is totally different thing what happens after that. N2k does no affect how OpenCPN handles data internally. It can provide API for plugins to handle other data. Connection to N2k brings some reposibilities. If OpenCPN or its plugin sends some data to bus, it must also handle some requsts, what other devices may send. In simple listening mode there is not much requirements except on lower level, which certified gateways does.

Android can be also connected to N2k. There you need some gateway, which connects to N2k and provides UDP for listen only or TCP connection for bidirectional. There are devices on market for that too.

There is also coming One Net - networked version of N2k, but that does not make much difference. It is just an other type of connection and has similar rules in some points as N2k.

leamas, 2022-07-03:
Hi Timo, thanks for chiming in!

First: I'm just about to dip my toes into this while you obviously have some experience so to speak. That's the starting point.

Here are some different views about using professional, certified hardware vs using more hacker-oriented  stuff. And in any case they have all different interfaces. The first question would then be if the layered approach described [above](https://github.com/OpenCPN/OpenCPN/issues/2729#issuecomment-1168879299) is feasible. Could we use such an approach to handle the multitude of hardware options available? 

Another question: If we should go for the second alternative and make a registered N2k gateway, would we then have to sign any NDA w r t the protocol details?

Thinking about it I see advantages having a TCP/UDP option available from the very beginning for testing reasons since it will be hard have all hardware available everywhere. Have you any thoughts about this?

ttlappalainen, 2022-07-03:
I see that OpenCPN has grown fron hacker sw to any boater sw!

With good layered solution it should be possible to offer support for different hw. 

You do not need to write any NDA as soon you do not buy any document. I have not written any NDA and digged my knowledge from leaked sources.

It does not matter is data coming through serial (USB) or TCP/UDP as soon as it has been designed so. Just think N2k like old NMEA0183 - it can also get data through serial or TCP/UDP. Only difference is protocol, which is not directly human readable in N2k - except if you can see data bits. I think most people thinks N2k too complex. Only difference is that there are more rules.

andyrbarrow, 2022-07-04:
Just lurking here in this very interesting discussion. 

Question: Wouldn't any solution for OpenCPN ideally be able to migrate to NMEA OneNet when it becomes more widely available?

ttlappalainen, 2022-07-04:
It is good to prepare for One Net. One problem is that it is again closed and takes time to get it open. Also I think that it will take some years to get popular. People are still using NMEA0183 and N2k has been available two decates.

So there should be layer to handle hw and other for internal usage. It is challenging, since who knows what requirements there will be in future. E.g., with N2k we know that if you are sending data to the bus, there are some requirements to respond requests. But what happens with future connections? 

About autopilots I know one company trying to make some kind of general controller. There is no standard commands to control autopilots on N2k. E.g., Garmin autopilot controlling is looks very complex. The one way (I think works with most) is to send route and track control information to the bus and set pilot to follow route.

leamas, 2022-07-04:
> "Be ready for a migration to "n2k over ethernet", whenever that starts to gain traction."

Perhaps even start here, since it seems  to be the only way to create something which can be tested without extra hardware on most if not all platforms.

[148 more replies]

closed 2024-04-25 by bdbcat