OpenCPN/OpenCPN #2730: Generalised autopilot support

keesverruijt, 2022-06-26:
**Is your feature request related to a problem? Please describe.**
Currently there is a [PyPilot](https://github.com/pypilot/pypilot_pi) plugin that allows direct control of the pypilot open source AP (by Sean), and an [AutoTrackRaymarine](https://github.com/douwefokkema/AutoTrackRaymarine_pi) plugin (by Douwe). 

**Describe the solution you'd like**
A generic method that allows any autopilot to be controlled, and this to be easy to adapt to other devices.

**Describe alternatives you've considered**
The current 2 plugins actually repeat each other in different ways.

**Additional context**
If we had a single unified controller, we would only need to develop the GUI in wxWidgets once, and any improvements are shared by all.

**Suggested solution**
I haven't looked at either source codebases in detail, but what I do know is that the pypilot plug-in talks using a network connection to the actual Python code in PyPilot. The AutoTrackRaymarine code base is all in C++ and has copied the reading/writing to/from the Actisense NGT-1 from the canboat code. 

My thinking is that we take the _pypilot_ plugin architecture, and write a "remote" piece of code, not (necessarily) in C++, to talk to the NMEA network. A TCP/UDP network protocol is the simplest method of achieving this.

We need to be careful, because otherwise we are re-inventing Signal K here.

TwoCanPlugIn, 2022-06-28:
Re: "If we had a single unified controller, we would only need to develop the GUI in wxWidgets once, and any improvements are shared by all."

The UI itself is quite simple, Autopilot Modes (off/standby/Heading/Wind/GPS", Heading Alteration (+/-1, +/- 10), Some status & heading/wind angle fields. Initially no need to implement the custom steering patterns that the manufacturers include or the initial autopilot commissioning.

The difficult part could be the back-end, which performs the transmission, reception, decoding & encoding of NMEA messages sent & received by the autopilot.

The current version of the TwoCan plugin allows any OpenCPN plugin to write messages to the NMEA 2000 network. I may contemplate adding the reverse, sending requested messages to other plugins. Refer to the [V2.1 announcement](https://www.cruisersforum.com/forums/f134/twocan-a-new-plugin-209706-16.html) for details on how a plugin can use TwoCan to send a NMEA 2000 message.

It also planned for TwoCan to use the OpenCPN Messaging Framework and JSON formatted text to support autopilots. TwoCan will send messages with autopilot mode and heading/wind values  (and possibly other alarm & status conditions) to other plugins and will receive messages from other plugins with changes to autopilot mode and heading course alterations. 

For example OpenCPN Message Id: "TWOCAN_AUTOPILOT_RESPONSE" or "TWOCAN_AUTOPILOT_REQUEST" and a message body containing something similar to 
{
   "autopilot":{
      "mode": GPS,
      "heading":123
   }
}

There are at least two pieces missing, 

Firstly definitions of the proprietary PGN's used by the autopilots.  Much of this has been done for Raymarine (with the exception of PGN 126720, although these are probably encoding the old Seatalk datagrams) , Simrad is quite well understood but I am not aware of any reverse engineering of Garmin's nor do I personally have any Garmin network captures.

And secondly, which is more of an implementation issue, how the autopilot UI can be dynamically adapted to use different algorithms for following a course. (Eg. compare OpenCPN's built-in APB/RMB/XTE generation with Douwe's course correction computation or how the back-end can be dynamically adapted to support new/different autopilots with different NMEA 2000 PGN's.

rgleason, 2022-06-28:
List of AutoPilot Interfaces and other AP related PI

1. [Pypilot_pi](https://github.com/rgleason/pypilot_pi/tree/master) - Nmea 0183  - Sean
2. [Autopilot_Route_pi  ](https://github.com/rgleason/autopilot_route_pi/commits/master) - Nmea0183 - Sean
3. [AutoTrackRaymarine](https://github.com/douwefokkema/AutoTrackRaymarine_pi) - Raymarine N2k with Artisense - Douwe
4. AutoPilotRM - Nmea0183 and Seatalk with converter - Bernd 
5. [TouCan](https://opencpn-manuals.github.io/main/twocan/0.1/index.html) - SteveAd
6. [Shipdriver(https://opencpn-manuals.github.io/main/shipdriver/index.html) - May benefit

Hakansv, 2022-07-04:
To the above list we may not forget what's actually is working fine without a plugin on all NMEA0183 APs:
OCPN console in routeman.ccp
if (pActiveRoute) {
    //      Update bearing, range, and crosstrack error
and...
Routeman::UpdateAutopilot() {
  // Send all known Autopilot messages upstream
and....
g_pMUX->SendNMEAMessage(snt.Sentence);