OpenCPN/OpenCPN #2736: Control checksum

jbonde, 2022-07-10:
Seems like the "Control Checksum" option is not working properly.

If a RMC string is received by UDP it works fine by positioning cursor on the chart if the string is concluded with a checksum. For testing purposes it is required (simpler) sometimes to create a string without the checksum calculation and for this the "Control Checksum" is checked off. However it does not make a difference as only strings with checksums are accepted. 

You can try these examples that are both displayed in the NMEA debug windows a green (=accepted)

Doesn't work (checksum not correct)
<GREEN>17:36:28 (UDP:0.0.0.0:2000) $GPRMC,173627.00,A,5642.89930,N,01130.41500,E,2.1,1.7,100722,,,A*6C<0x0D><0x0A>

Works (checksum correct)
<GREEN>17:38:19 (UDP:0.0.0.0:2000) $GPRMC,153757.00,A,5541.56505,N,01231.01930,E,0.027,,100722,,,A*7D<0x0D><0x0A>

Tested at W11 and Linux at a RPi. Same problem on both.

transmitterdan, 2022-07-10:
It looks like a bug to me.  I will look deeper into it.

Hakansv, 2022-07-10:
TD.. pls check my PR https://github.com/OpenCPN/OpenCPN/pull/2737. 
If you have a better fix pls do.
/H

transmitterdan, 2022-07-10:
@Hakansv .. The bug seems to be in the NMEA0183 library RMC function.  It ignores the global **m_bchecksumCheck** flag and rejects any RMC sentence with invalid checksum regardless of user selection.  See code: https://github.com/OpenCPN/OpenCPN/blob/3f6385db4e17ac6cb9d9f89e5952cc06e7a3dda8/libs/nmea0183/src/rmc.cpp#L111

closed 2022-07-11 by bdbcat

Hakansv, 2022-07-11:
@transmitterdan ..
If I understand it correct the "if ( check == NTrue )" is only to check for a existing "bad" type of checksum. The actual checksum comparative calculation is done earlier in "datastream"? 

What I changed in #2737 was to actually save a changed option "[ ] Control Checksum". The program flag was always true unless you didn't hacked the config file.

Hakansv, 2022-07-11:
@jbonde ..
Thanks for the report. As you can see it will be solved. The first attempt is already in the code, master branch, for next O release.
A meanwhile possible solution is to change in the config file: ( DataConnections= )
See my example below for a connection to port COM2. The first field after the baud rate has to read: 0 when option Control Checksum shall be unchecked.  (Pls save a config file copy before you try the "hack")
`0;2;0.0.0.0;0;0;COM2;4800;0;1;0;;0;ECRMB,ECRMC,ECAPB;1;0;1;0;1;;1`

Hakansv, 2022-07-11:
@transmitterdan ..
Two parts here obviously.
What I did was to make the flag "m_bchecksumCheck" to be changed and saved after the options update.
The part you describes is a more sophisticated one. And your idea to actually correct a bad checksum if m_bchecksumCheck is not set seems appropriate.
So, if you can take this ball it's really appreciated.
Thanks a lot.

rgleason, 2022-07-11:
Isn't it better to have the problem corrected at the source rather than help the message along?
You are liable to be making many corrections eventually.

bdbcat, 2022-07-12:
TDan...
"your idea to actually correct a bad checksum...":
This one bothers me a bit.  Lets see what the ramifications are of doing this "right".

transmitterdan, 2022-07-12:
If we are going to accept any random checksum it seems harmless to repair a bad one.  I don't see the downside.

bdbcat, 2022-07-12:
"Later on in other code the RMC sentence checksum is again checked and if the checksum is bad it is not used to control movement of own ship. This code does not obey the option flag from the config file. It always checks the sum. This is the real bug."
And where is this, in the code?

transmitterdan, 2022-07-12:
@bdbcat - see my previous message in this issue: https://github.com/OpenCPN/OpenCPN/issues/2736#issuecomment-1179794555

bdbcat, 2022-07-12:
Tdan...
Hmmm, I see.  Long-standing bug, this is. To me, the "correct" fix is either:
1.  Implement a method on NMEA0183 class, instructing the library to ignore checksum, or
2. Remove the checksum test from the Parse() method of all sentences, assuming that the message processing (Parse()) will not be called if the checksum is bad and the "ignore checksum" option has been specified.  That is, the message was filtered upstream.

transmitterdan, 2022-07-13:
I think we should fixup the checksum so all downstream consumers, including plugins and outbound pathways, get a corrected message. I think that is the intent of the feature. Even though the checkbox is named "ignore" it really means accept bad checksums.

I'm trying to think of a case where that is not what the user expects to happen if they select the ignore checksum feature. The OP expects that to happen I think.

bdbcat, 2022-07-13:
"Even though the checkbox is named "ignore" it really means accept bad checksums."  I am not persuaded by this argument. "Ignore" is supposed to mean "don't bother to compute the checksum, since the device on this channel is known to produce incorrect checksums". 

However, I could become convinced for the sake of convenience to plugins that request and process NMEA0183 data.  Since we send the raw sentence, with no descriptive metadata, then it might reasonably be expected by the plugin that the checksum is correct.  Or that they will not get a message with a bad checksum. The alternative is, of course, to rebuild the plugin to "do the right thing" when it sees a checksum error, whatever that may be.  Not attractive...

But the entire discussion seems a little "backwards" to me.  . I think we should do better when the entire comms model is refactored for O58.  See n2k discussion.
Meanwhile, I will accept a PR now, correcting bad NMEA0183 checksums on-the-fly, but only in the case where the user prefers to "ignore checksums".   Still gives me the willies....

transmitterdan, 2022-07-13:
> But the entire discussion seems a little "backwards" to me. . I think we should do better when the entire comms model is refactored for O58. See n2k discussion.

In this we are in 100% agreement.

I believe there are three possible classes of downstream consumers (core, plugins and multiplexer output).  I think we agree that the core and plugins should get a valid sentence. Should we also send a valid sentence to a multiplexed output service, or is that going a bit too far?

Hakansv, 2022-07-13:
Opinion:
We would as now by default reject a message without or with a false checksum. We would do that at first reception and also before any downstream transmission.

If the user select to ignore the checksum that message would also be forwarded downstream to all possible receivers. We would thrust the user doing so to be conscious about any consequences. 
A log message  about a refused checksum validation could be created.
Example:
datastream.cpp:215(?) Checksum validation refused
And move this down:
datastream.cpp:216(?) ConnectionParams CTOR

Hakansv, 2022-07-13:
So what's left? Pass the connection's ChecksumCheck status to sentence.cpp::ComputeChecksum()?

transmitterdan, 2022-07-14:
We have to do more than that to get downstream consumers to accept the message with bad (or no) checksum.

Since we are thinking of a revamp of the entire input->mux->output system I think a simple fix for now will suffice.  If checksum is to be ignored on input streams, then we will replace the incoming checksum with the correct checksum thus "ignoring" all incoming checksums on that stream and all downstream consumers (core, plugins, mux output) will get a correct checksum.  Input/output filtering will be the same as it is now.

The user must be sure that they truly want the checksum to be ignored.

transmitterdan, 2022-07-14:
> I think I agree with Dave to leave downstream consumers to make it's own validation of the checksum if needed and MUX the verbatim message.

I can be easily persuaded to agree. The implementation is trivially simple to not disturb the outbound multiplexer copy of the input stream whilst O and plugins still get valid checksums. I can submit a PR for testing this weekend.

Is it ok to merge your options PR for completeness?

transmitterdan, 2022-07-17:
Anyone like to test on Windows a proposed solution to this issue can download installer here: https://dl.cloudsmith.io/public/dan-dickey/opencpn-unstable-testing/raw/files/opencpn-msvc_5.6.1_setup.exe

If you want to see the code or build on other platforms, see branch: https://github.com/transmitterdan/OpenCPN/tree/NMEAchecksum

rgleason, 2022-07-17:
So you are going to have a checkbox "Correct incorrect checksums"?

transmitterdan, 2022-07-17:
No, the existing checksum box will suffice. If it is unchecked then bad checksums will be accepted by On and all plugins the same as good checksums. I think this is the expected behavior and just fixing a bug. 

Best regards,
Dan Dickey

Hakansv, 2022-07-17:
Dan..
Nice.. I'll test as soon as possible but will be off for a couple of days due to "father supporting daughter" 
Håkan

rgleason, 2022-07-17:
You are not telling the user that they are enabling correction of checksums.

transmitterdan, 2022-07-17:
> You are not telling the user that they are enabling correction of checksums.

By unchecking the box, they are saying they do not care about the checksum. They expect O to accept the message regardless of what value is in the checksum.

It does not send the corrected checksum outside of O or plugins.  The multiplexer output side works as it always did.

I also made a little change to the NMEA log window.  If we accept bad checksum messages, they appear in the log window in red.  Before they never appeared in the log window at all.

rgleason, 2022-07-17:
I like that!

Hakansv, 2022-07-18:
TDan...
I've built your branch on Win and it works good.
One question though:
When "Use Checksum" is not selected a message with wrong checksum is colored Red in the NMEA Debug window although it's used. Would it be more wrongly conceived to use another color?
Not less since the now "fixed" message is colored "Normal" when sent out to another stream by the MUX.
I used UDP 10110 for incoming and serial COM for output.
Thanks
Håkan

rgleason, 2022-07-18:
Make it Dark Pink?

Hakansv, 2022-07-19:
@transmitterdan ..
Yes I think the red lines may confuse.
What about this log message instead?

```
@@ -224,17 +224,18 @@ DataStream::DataStream(wxEvtHandler *input_consumer,
       m_params(*params) {
   m_BaudRate = wxString::Format(wxT("%i"), params->Baudrate),
   SetSecThreadInActive();
 
-  wxLogMessage(_T("ConnectionParams CTOR"));
-
   // Open();
 
   SetInputFilter(params->InputSentenceList);
   SetInputFilterType(params->InputSentenceListType);
   SetOutputFilter(params->OutputSentenceList);
   SetOutputFilterType(params->OutputSentenceListType);
   SetChecksumCheck(params->ChecksumCheck);
+  if (!m_bchecksumCheck)
+    wxLogMessage(_T("No Checksum verification"));
+  wxLogMessage(_T("ConnectionParams CTOR"));
 }
 
 void DataStream::Open(void) {
```

rgleason, 2022-07-20:
If you are doing it for bad ones, why not for signalK ? (:-))

Hakansv, 2022-07-20:
The signalk json text protocol don't use checksums

commit f8376f64 (2022-07-11): Merge pull request #2737 from Hakansv/chksum

Save a changed connection checksum option. Fix: #2736