OpenCPN/OpenCPN #2739: Common NMEA2000 support library

bdbcat, 2022-07-16:
In another thread, we discuss architectures for integrated NMEA2000 support in OpenCPN.
One topic that arises is the potential need for a common NMEA2000 PGN decoder library, usable by core OpenCPN, as well as by plugins.
This issue will investigate that topic in more detail

bdbcat, 2022-07-16:
I have done some poking about on github, looking for code models that might suit our need here.  There are a few authors of such  projects.  We are fortunate that two of them (Timo and Kees) are well represented on the OCPN team.

The one code base that interests me most is this piece of Timo's github set:
https://github.com/ttlappalainen/NMEA2000/blob/master/Examples/DataDisplay2/DataDisplay2.ino
I like this because:
1.  The essential elements of the function have minimal (or no) hardware dependencies.
2.  The code is simple readable inline c++.  Does not rely on esoteric compiler or system library features.  Will code this be extremely portable.  Anyone can read it and understand immediately the runtime flow.
3.  It is extensible, simply just by inspection.
4. With small revision, it could be built with no state retention at all.  This would be a real simplification for the average use-case..

With a little thought, I can say that this code, or something very similar, will be just enough for integrated OCPN core n2k message decoding.  Will it work for plugins?  Input required....

Now, surely there will be calls for more complete, or feature rich libraries.  For instance, synthesis of PGNs for transmit from OCPN is not contemplated here.  And of course, the supported PGN list may need to be expanded somewhat.  Even some aspects of network management may be wanted, although we are attempting, by design, to isolate all hardware dependencies to our driver stacks.
To this I say, making things more complex _because we can_ does not always lead to clean, maintainable code.

Thoughts, all?  (esp Timo)

ttlappalainen, 2022-07-16:
First about my NMEA2000 library:
I wrote it after working about 2 years with N2k and extented it later by request to be as certification ready. It is currently used in several commercial products as "N2k stack". Products I know has hw based to Teensy 3.2, 3.6, ESP32, RPi. And I heard from former Maretron developer that it is much simpler to use and understand for developer that their stack. Since library has been originally wrote for small MCU:s, it does not use e.g., std, which would have helped in some cases.

The library has some parts:
1. Main core taking care of building tN2kMsg and most N2k related tasks and offering way to register specific functional handlers.
2. Message parsing functions
3. Device list tracker
4. Actisense reader, which converts Actisense format message from stream to tN2kMsg.

For message parsing library uses functional way for two reasons: 
1. It is simpler to understand for beginners. 
5. Compiler will automatically pass all unused functions and so save memory, which is important for small MCU:s.

I some commercial solutions I have been working and in my NMEA Simulator there is database, which describes PGN:s a bit like Kees CanBoat. Difference is that from database developer can simply list only necessary PGNs to save memory while CanBoat has one huge structure containing all PGN description. This method is good, when all data must be available with common handlers.

An other way is to use classes as PGN handlers. It can have good performance in some cases compared to functional way since it does not need to parse all values, if you need only one. Code is larger.

About library and OpenCPN:
If OpenCPN would work as TwoCanPlugin with frame based gw, library could be used nearly as is. But as I have prefered to use certified gw in front some library code should be skipped. Anyway I do not see that library can be used as is since there is some new requirements and some new possibilities.

OpenCPN and N2k:
On each N2k bus there should be only one gw. Supporting multiple gw:s as spare works on listening. Sending anything to bus complicates things because of N2k NAME. Other devices fixes their input to NAME, which means that in case gw stops working, they show NA. If user wants to use spare gw, he need to reconfigure other devices to listen new gw.

It is possible that user has multiple separated N2k bus. Still not problem for listening. For sending bus tag will be needed.

Listening via certified gw is very simple, since it does not bring much requirements. Only thing is that other devices should get information about PGNs OpenCPN will listen. Not sure yet how that can be handled with e.g, NGT-1, but it does not cause any effect, if information is not complete. For future core and each plugin should register PGNs they will listen.

Sending brings more requirements. System should respond to ISO request and group function 126208. Also since bus sees OpenCPN only one device, there can be only one sender for each PGN.

N2k Protocol level?
Yet it is unclear for me how do we call this. Anyway I think that for this level any driver should provide specified N2k message like I have tN2kMsg (just extend it). Protocol level can then:
1. Offer message handler registering
2. Offer  ISO request and group function 126208 registering for senders.
3. Build device list by tracking. With this there is a problem if connection listen only like UDP broadcast.

Each driver should then provide 1. and 2. for N2k Protocol.

Parsing is independent library, which reads/set tN2kMsg:
- Functional way as published with my library. This is simple to maintain and has ready most common PGNs. PGNs with longer parameter list should be done by using structs like in some cases on my library. (Even struct has been difficult for beginners).
Example reading lat/lon PGN 129025: ParseN2kPositionRapid(Msg,Latitude,Longitude);

- Database way. If there would be one common database, there is no problem to have all PGNs like on CanBoat. This works best, if data will be shown "blind". E.g., with dashboard kind system one can simply have list of received data to be selectable for user.
Example reading lat/lon PGN 129025: tMsgHandler MsgHandler(Msg);  Latitude=MsgHandler.GetN2kField(1);   Longitude=MsgHandler.GetN2kField(2);
In this case tMsgHandler can also offer information like GetFieldCount, GetFieldDescription, GetFieldRange, GetFieldDataType

- Class way. Each PGN handler can have own class. Still easy to maintain and easy to understand for class oriented programmers.
Example reading lat/lon PGN 129025:  tPGN129025 PGNMsg(Msg); Latitude=PGNMsg.Latitude(); Longitude=PGNMsg.Longitude();

In one solution I have mixed class/database. Each class has full description of PGN like on database so it can have same informational functions on base class. Specific handler can be then get by casting. This brings advantages of both class an database way.

I think here is already too much thoughts for beginning.

bdbcat, 2022-07-16:
Timo...
Thanks for the input.
For the common parsing library, I am inclined to use what you call "functional way", for the reasons you note.  So I will tentatively start by pulling bits and pieces from your library into a new project, and see how it goes.
Thank you for publishing this work.  I look forward to your comments as we proceed.

ttlappalainen, 2022-07-16:
I have been playing a bit with NGT-1. It can be simply configured as Receive all in the beginning. For transmit user has to configure required PGN or they can be added by some NGT-1 specific commands.

douwefokkema, 2022-07-16:
@timo, for transmitting any PGN with the NGT-1 use the following initialization, works fine, I copied it straight from Kees Canboat.

 /* The following startup command reverse engineered from Actisense NMEAreader.
  * It instructs the NGT1 to clear its PGN message TX list, thus it starts
  * sending all PGNs.
  */
  static unsigned char NGT_STARTUP_SEQ[] =
  { 0x11   /* msg byte 1, meaning ? */
    , 0x02   /* msg byte 2, meaning ? */
    , 0x00   /* msg byte 3, meaning ? */
  };

ttlappalainen, 2022-07-17:
This enables on receiving all messages not transmitting.
0x11 = request/command operation mode
0x0002 = Transfer: receive all, 0x0001 = Transfer: normal

NGT-1 can be commanded to output selected messages. For OpenCPN this means:
1. Can be done with Actisense NMEA Reader provided with NGT-1. Write simple instructions at first how to do it. Boater use needs it is not in my thinking any boater, since controlling e.g., autopilot with uncertified way looks risky for them. So these type of boater can read instructions.
2. Each plugin should register to N2k protocol, what messages they listen or send. For messages to be sent out, NGT-1 driver can then send necessary enable commands to NGT-1. Driver can on start request enabled PGNs and if registered PGN is not on list, it can enable it.

Since enabling is one time task, I prefer at the beginning case 1. But keep in mind that NGT-1 driver could have it in future.

ttlappalainen, 2022-07-17:
About N2k code and OpenCPN:
If you copy code used in my library, can that part be kept with MIT licensing? Problem arises, if some adds new PGN handlers to OpenCPN code and same is needed on my library. Code would ne nearly similar and some could sue by braking license.

I wrote library before I had knowledge of all N2k types. Thats why there are N2kTypes.h and NMEA2000StdTypes.h. Also in functions there is calls like:
ETAtime=N2kMsg.Get4ByteUDouble(0.0001, Index);

This could be in future:
inline double N2kGetTime(tN2kMsg N2kMsg, intger &index) { N2kMsg.Get4ByteUDouble(0.0001, Index); }
...
ETAtime=N2kGetTime(N2kMsg,index);

In N2k each type defines exact reading. For e.g., temperature reading there are TempHigh and TempLow, which has different multiplier.

Also I have used originally index as byte based, while it could be bit based. Now there must be:
   uint8_t  vb=N2kMsg.GetByte(Index); MessageID=(vb & 0x3f); Repeat=(tN2kAISRepeat)(vb>>6 & 0x03);
while it could be as bit based:
  MessageID=N2kMsg.GetBits(6, Index);
  Repeat=N2kMsg.GetBits(2, Index);

As in my library I prefer to keep all real values as double and in SI despite original definition. E.g., Depth Range is on N2k as integer (as actually everything). So often users adding functions to library thinks that also variable can be integer. But this is physically real data and must be double. Data like SID, ID etc. can be integers. Lists should be typed as I have.

And still MIT after changes, since I may release v2 of library with some modifications, which also contains above ideas.

leamas, 2022-07-17:
@ttlappalainen  wrote:

> If you copy code used in my library, can that part be kept with MIT licensing?

Since you have claimed copyright for your code the license cannot be changed by anyone besides you, this is legal basics. So no need to ask.

OTOH, using MIT-licensed code in OpenCPN is  no problem, the licenses are compatible.

bdbcat, 2022-07-17:
What he said...

bdbcat, 2022-07-18:
Timo...
To be clear, I will not be using your library verbatim.  I will be extracting methods and types to build the common parser library for OCPN.
The API to the library which I plan to publish will not expose the internal details of your implementation. The API will be very simple, with functional interfaces. This plan allows changes and improvements to the internal structures/methods without disturbing the use of the library API in OCPN.