OpenCPN/OpenCPN #2740: [Feature Request] Night mode ideas

mgrouch, 2022-07-16:
I looked at other charts and map programs and I've seen one feature I think would be useful to have in OpenCPN too.

In those programs you zoom to see world map it (knowing time) shows parts of earth shadowed from sun in darker color.
When you zoom on regions which already experienced sunset you get night mode of chart and 
daylight charts for other regions.

Thanks,
--MG

mgrouch, 2022-07-16:
More thoughts there could be 2 modes, night/twilight,day based on local time of your location and another based on 
time (sun position) in the location you are zooming into

mgrouch, 2022-07-16:
related to https://github.com/OpenCPN/OpenCPN/issues/2427

campbellhennessy, 2022-10-06:
A small comment if anyone goes to implement this.  I have worked with lots of different chart display systems and there is always an issue (decision) on how to handle night mode between vector and raster charts.  Vector charts can generally be easily made into a night mode by changing certain colors completely (changing white areas to black for example).  Raster charts are more difficult.  I have seen some applications that make no attempt to put them into night mode, the best adjustment I have seen on raster charts is to shift the full display to minimize blue light and then to dim a little the overall brightness.