OpenCPN/OpenCPN #2743: Merge of #2689 ([WIP] Model refactoring, ....) o-charts plugin is not working

Hakansv, 2022-07-26:
@leamas 
Report for Git master branch after merge of #2689
Win10, built by VS2017
Builds fine, also the package.
- O-charts chart bars are not visible until the plugin is deactivated and activated.
- After that o-charts rendering is not correct. See pict. Raster charts are working as usual
- An attempt to reinstall o-charts plugin caused a O crash. See attached log file. (No error while debug though?)
- Uninstall and install of o-charts plugin worked fine though. No difference for the rendering bug.
   MESSAGE fpr.cpp:99 oexserverd results:
   MESSAGE fpr.cpp:102 0

[opencpn - reinstal_plugin.log](https://github.com/OpenCPN/OpenCPN/files/9187224/opencpn.-.reinstal_plugin.log)

[image]

leamas, 2022-07-26:
It's certainly about the message handlers, I have done what I could to debug but to be honest things like this was to be expected, this is a big change.

Thanks for report. Will try to look into it, but will not have time in until the day after tomorrow or so. If the problem is blocking we need to revert.

Hakansv, 2022-07-26:
It's just a report for the big change. I can imaging all is not easy to test. Not less on Win for a native Linuxer.
No hurry, enjoy cruising if it's like that. The "sandbox" is not blocking. It's for tests? I just try to report while fresh.

bdbcat, 2022-07-26:
Corrected in github master now.
Thanks for the report.

Hakansv, 2022-07-26:
@bdbcat 
Splendid! Now o-charts are working as usual.

I also tested to reinstall o-charts_pi and it failed again. Another plugin successfully reinstalled though.
Now this reinstall function I've not tested for long so if this failure is connected to the actual merge I don't know.
Log messages for the reinstall start to O crash attached.

Thanks
Håkan

[O_log_reinstall_o-charts.log](https://github.com/OpenCPN/OpenCPN/files/9193315/O_log_reinstall_o-charts.log)

transmitterdan, 2022-07-28:
Has anyone been able to build opencpn-cmd.exe on Windows?  I cannot get past this link error:

Error	LNK2019	unresolved external symbol _WinMain@16 referenced in function "int __cdecl __scrt_common_main_seh(void)" (?__scrt_common_main_seh@@YAHXZ)	opencpn-cmd	D:\transmitterdan\OpenCPN\build\MSVCRT.lib(exe_winmain.obj)	1

leamas, 2022-07-28:
@transmitterdan : For such really complicated problems you might need to contact the world-famous Windows guru Alec  Leamas "blushes"

Seriously: I have built and tested it. Also, more important, Appveyor does build it for  example in [last build](https://ci.appveyor.com/project/bdbcat/opencpn/builds/44286640). I basically mimicked the Appveyor command line build, worked for me.

Somehow, that unresolved symbol is related to the last line `wxIMPLEMENT_APP_CONSOLE(CliApp);`in _console.cpp_. I remember running into similar problems while working with it on Windows. Unfortunately, that macro is not that documented :(

leamas, 2022-07-28:
@Hakansv 

> Log messages for the reinstall start to O crash attached.

hm... Any chance you could provide a stack trace from the  crash?

leamas, 2022-07-28:
hm... the stacktrace is really helpful, thanks!

This is about a nasty fix caused by the use of raw pointers. 
  - The bug happens in OnPluginDeactivate()
  - This function is bound to the on_event_deactivate event variable.
  - The code triggering the event is in plugin_loader.cpp around line 550. 
 
Here is a dirty, dirty fix to handle a situation where the event should provide the current PluginContainer, but this is about to be deleted and have ceased to exist when OnPluginDeactivate() does it's job. The fix creates a verbatim copy of the PluginContainer using malloc and memcpy. This works in other places which just needs to remove UI elements containing the plugin, but fails miserably if the undocumented  read-only semantics are broken, for example when doing a delete like here.

I think I understand the problem, but don't really know how to solve it. Not really time now for a proper fix using ref-counted pointers, but perhaps it is the only solution. Need to file a new bug about dirty fix.

@bdbcat : thoughts?

leamas, 2022-07-28:
Filed a new bug #2744 for this. This is a separate, partly known issue and I have promised to file a bug on it. Let's have discussion about Håkan's last crash there, leaving place here for other possible problems emerging from #2689

Hakansv, 2022-07-28:
No other problems found by me. The main issue about rendering o-charts is solved.

closed 2022-07-28 by Hakansv