OpenCPN/OpenCPN #2745: Thoughts for prioritization of data sources for OCPN core.

Hakansv, 2022-08-01:
**Is your feature request related to a problem? Please describe.**
More types of data connections like NMEA2000, N2k, are most probably coming up on next OCPN release.
This implies more rules for prioritization of similar data from various sources.
Also is the present prioritization logic not fully adequate. We now set a prioritization level for each a data connection and that will for example sort out a second RMC from another connection but not a GGA from the second if GGA is not included in the first. That will often result in a "jumping" ship position.

I'll here try to describe possible rules for data prioritization. As always it's my thoughts for a start of discussion.

**Describe the solution you'd like**
I've a dream.......Let me start there:

I'd like an automated data and source detection where the user can request a list of available connected sources for each data type and edit priority levels on a form.
For OCPN core we need to priority between sources for the following data types:
Position, COG, SOG, heading and variation.
Every other data prioritization will be handled by plugins like for example Dashboard, Watchdog and Tactics.

Such a automated search function would walk through all available connections for each Data types in the table below.
For each data type a list of available connections will be presented on a form where the user can set desired priority.
A possible result for Positions is the Providers table herein.

**Additional context**
Now we can't expect every user to initially being interested in this maybe complex structure so we need to make an assumption in some way. If nothing is earlier set let's make the first detected connection for each data type first priority and the second next and so on.
This also implies we have to run the automated detection function not only by a user request but also on a OCPN start up. If a previous prioritization is present we use that otherwise make or own.
When a new connection is added I suggest the "Apply" will lead the user to the source prioritization form. It would of course be great if both connections and Data type prioritization and a update button could fit on the same dialog form but I suppose small screens will have trouble with that.

For N2K it's possible we would look in more PGNs. Pls advise.

The rule to switch to next priority on lost data can be the same as now. (gps_watchdog_timeout_ticks)

**Tables for Data Types and Data Providers example**

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/HAS-WIN/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/HAS-WIN/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Data   types | NMEA0183 | N2K PGN | SignalK (own prio postulated) | Option
-- | -- | -- | -- | --
Position | GGA | 129025 | navigation.position |  
  | GLL | 129029 |   |  
  | GNS |   |   |  
  | RMA |   |   |  
  | RMC |   |   |  
  | AIVDO |   |   |  
  |   |   |   |  
COG | RMC | 129026 | navigation.courseOverGroundTrue |  
  | VTG |   |   |  
  | AIVDO |   |   |  
  |   |   |   |  
SOG | RMC | 129026 | navigation.speedOverGround |  
  | VTG |   |   |  
  | AIVDO |   |   |  
  |   |   |   |  
  |   |   |   |  
Heading true | HDG | 127250 | navigation.headingTrue |  
(or mag + variation) | HDT |   | navigation.headingMagnetic |  
  | HDM |   |   |  
  | AIVDO |   |   |  
  |   |   |   |  
  |   |   |   |  
Variation | HDG | 127250 | navigation.magneticVariation | VMM
  | RMA |   |   |  
  | RMC |   |   |  

</body>

</html>

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/HAS-WIN/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/HAS-WIN/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

  Example of   detected position providers| |   |   |   |   |  
-- | -- | -- | -- | -- | -- | --
User PRIO | NMEA0183 | Source | Data port | Address | Name |   |  
6 |   | RMC | Serial | COM3 | User comment: Garmin 120
3 |   | RMC | UDP | localhost | User comment |   |  
0 |   | AIVDO | TCP | 192.168.10.21 | User comment |   |  
4 |   | GGA | GPSD | 192.168.10.22 | User comment |   |  
  |   |   |   |   |   |   |  
1 | N2K | 129025 | N2K | COM4 | PGN: Furuno SC50 |  
5 |   | 129025 | N2K | 192.168.10.20 | PGN: Garmin XYZ |  
  |   |   |   |   |   |   |  
2 | SignalK | navigation.position | SignalK | 192.168.10.13 |   |   |  
  |   |   |   |   |   |   |  
Prio note: |   |   |   |   |   |   |  
0 | Don't use |   |   |   |   |   |  
1 | First choice |   |   |   |   |   |  
2 | second |   |   |   |   |   |  
3 | third |   |   |   |   |   |  
4 | and so on…. |   |   |   |   |   |  
  |   |   |   |   |   |   |  
Position   note: |   |   |   |   |   |  
Prioritized   provider for position will also be used as source for "satellites in   use" (like GGA/129029/navigation.gnss.satellites)  if present.

</body>

</html>

leamas, 2022-08-01:
great work, about to read. But isn't this a typical discussion item rather than an issue?

Hakansv, 2022-08-01:
Since we've issues with the present prioritization it needs to be corrected. So I thought it would fit here?

leamas, 2022-08-01:
> Since we've issues with the present prioritization it needs to be corrected. So I thought it would fit here?

Fair enough, let's skip this discussion about the discussion...

hm... Yes, new code is under way, and while it cannot do anything at all for now, it has hooks for this. Taking the position data as an example. the idea is that it works like this.

1. When a position arrives it is converted to a GnnsFix decoded message. This is basically the same message, independent of the source which could be  0183, n2k  or SignalK.
2. The message contains, besides the position data, the source and a priority which by default is 0.
3. If there are multiple messages with position data from different sources all of which with priority 0 a new message is triggered which eventually starts a dialog where user gives the sources priorities for this data.
4. Obviously, the priorites selected by user needs to be persistent to next run...

This is a scheme which is reasonably simple to implement which I think should cover this (?) One advantage is that we don't really need to think about running dialogs when for example connecting or disconnecting devices. We just run dialogs when there is data which actually needs to be prioritized.  The overall view you describe would then be about editing and refining all priorities set so far.

Thoughts? Should something like this work? If not, it's definitely the right time to talk about it.

Hakansv, 2022-08-02:
>  1. When a position arrives it is converted to a GnnsFix decoded message. This is basically the same message, independent of the source which could be 0183, n2k or SignalK.

Like today's PostProcessNMEA() where the prioritization could be done?    

> 2. The message contains, besides the position data, the source and a priority which by default is 0.

Yes, sounds fair. Although in my legend above a priority "0" means "don't use". First received using, the same legend,would get priority "1" and if already used "2" and so on.    

> 3. If there are multiple messages with position data from different sources all of which with priority 0 a new message is triggered which eventually starts a dialog where user gives the sources priorities for this data.

Something like that could do. If this is the first time we walk through the connections and no priority are set before I think we should collect all connections before triggering a user message asking to confirm the priorities.
If a new connection appears, not previously prioritized, we may set it to next free number and ask for confirmation.    

> 4. Obviously, the priorites selected by user needs to be persistent to next run...

Yes. agree.

TwoCanPlugIn, 2022-08-03:
For NMEA2000 derived position data why not just automagically select a preferred source? Providing a UI and user input will lead to configuration errors and support questions.
Suggest you prefer the GNSS source with the lesst HDOP and in the case of two sources with the same HDOP, the source with either the lowest source address or the lowest network name. Note that PGN's 129025, 129026 & 129029 (and others such as 129539 & 129540 ) are generated by the same device, so once the preferred source has been chosen, they can be selected on the basis of source address. Either a watchdog or a time stamp can be used to trigger the failover. The same approach based on source address or network name can be used for other duplicated sources such as heading, variation, attitude, rudder angle.

Hakansv, 2022-08-06:
@TwoCanPlugIn 
Yes I agree the code must in some way make the first priority assumption. My suggestion above was to make first found prio 1. It can of course be more sophisticated like searching for best position quality if we find it worth the efforts. 
If several sources, like my extended discussion example, the user input would though be necessary? Not less if the antennas are placed far apart from each other. 
If we need user input for every new connection can be discussed. It may be enough to have a visible button to open the user dialog.

Hakansv, 2022-08-06:
@All..
You may have noticed I suggest a new designation for priority order? 
Priority "1" is the first choice, "2" the second and so on... 
That's the opposite to present code but my imagination is that many, as me, understand priority "1" as the first priority?
Also I used priority "0" as "don't use" and should consequently not be used when a program function sets a priority.

Thoughts?

Håkan

leamas, 2022-08-06:
Perhaps we could see this from a little higher level. As I see it, we have sod far three kind of priorities:

-. An  ordered set of manually and/or automatically defined priorities.
- A manually or automatically set priority which means "don't use at all"
- An automatically set priority which means that this item has not had it's priority set.

Whether this is implemented as 1, 2, 0 or -1 is IMHO an implementation detail. However I see other questions:
1. What should we do when the only available data is marked as "don't use"? Detecting this obviously includes waiting for some time, but how long?
2. How should we handle data which is yet not prioritized? Force a dialog? Use/don't use until dialog is completed?  Set an automatic priority and just use that? Set an automatic priority, use that and also invoke a user dialog, with the automatic priority as a default value? ...

bdbcat, 2022-08-12:
Trying a different angle:
We have two types of priority decisions to make.  Thinking just about basic nav data (position, GOG/SOG/HDT/AR):
1.  Per channel priority (what we have today).  This covers the case of having two GPS receivers on two different connections, as well as the case of one traditional GPS receiver on one channel, and an AIS transponder sending AIVDO on another channel.
2. Intra-channel prioritization:  For example, the network N0183 case, where RMC and AIVDO messages are arriving on the same channel.  We do not do this well, or at all, today.  Only recourse today is to filter away RMC on this channel.  Not great.

The priority logic must consider these two situations in implementation.

Hakansv, 2022-08-12:
Fully agree.
To elaborate 2. Intra-channel prioritization a position can as you say for example being present in several NMEA0183 messages but also a N2K channel can have a position from different devices using the same or different PGNs. 
A possible scenario:  Table "Example of detected position providers" in the first post here.

leamas, 2022-08-13:
Note that "my" approach covers this  more or less out of the box. For now, discussing data in BasicNavData, but this is just an example. Lets assume this is about three entities:
  - CommBridge, which converts incoming raw messages to BasicNavData. This message is marked with the source i. e., the n0183 physical interface or an n2k NAME..
  - The Updater, which listens to BasicNavData  messages and updates the global data.
  - The Prioritizer which also listen to BasicNavData. It detects of there are BasicNavData messages from different sources and makes the decision which one to use. It then invokes:
  - AppMsgBus, which blocks BasicNavData from  specified sources, as set by the Prioritizer.

This simple scheme handles both per-channel and intra-channel without any specific adaptations. The trick is basically to mark each BasicNavData message with the correct source. It is also important to postpone the update of global data until BasicNavData reaches AppMsgBus where it is subject to the common priority mechanism

Hakansv, 2022-08-13:
> if we need to choose between multiple PGNs from the same node (i. e., same NAME)
Maybe not necessary? Position info from the same node would be equal even if in multiple PGNs? (The antenna location is the same)

Hakansv, 2023-01-30:
I'd say this discussion ended up in a reliable and logical solution fully in line with the approach to make prioritization to favor a stable position for the ship.
Stable in order to use only one desired source even if the same data is available from another.
Reliable position in order to switch to next in priority if the the first fail. And back again when eventually restored.
Thanks.

closed 2023-01-30 by Hakansv

rgleason, 2023-01-31:
Isn't there some need for User Documentation?  or at least an explanation of all the work done and the purpose? 
I think this is much needed in the wiki, maybe not as detailed and technical, but at least something that leads to understanding by the Users of what to expect!!  

I have started a page in the manual, here for anyone who knows more, to add details about the implementation.
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:advanced_features:nema2k

I don't think this should be closed yet.  Shouldn't parts of it become documentation for both the User Wiki and Developer Wiki?

leamas, 2023-01-31:
If we should keep every issue which needs documentation open, well, just imagine.

As for the developer manual, please consider filing a bug against it, possibly with a reference to the issue here.

As for the User Manual, you might want to implement something similar to filing bugs. Perhaps a simple "TODO" page, perhaps a dedicated githug repo, perhaps something else...

rgleason, 2023-01-31:
You have your methods.. I have mine.

Hakansv, 2023-01-31:
Rick..
Yes, this prioritization function needs to be described in the manual.
To be mentioned in the GNSS set up section with a link to the connections section.
In the connections section a more complete description of the functions, logic and how to use.

rgleason, 2023-02-01:
Thanks so much, Hakan, how do you want to handle this?
Can you edit the current manual? 
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic
GNSS Setup  https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:get_started:gps
Connections 
Connections- Multiple GNSS Input Filtering https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:toolbar:options:connections#multiple_gnss_input_filtering

I think you need to add the Priority Table and we need an explanation.

Hakansv, 2023-02-01:
Rick..
Let's see if can manage to make some text about that. I've in fact used the prioritization and the new possible OCPN connections during the test sequences. I'm though doubtful to make it direct in the manual. That's based on my sometimes "own" use of the English language. One method can be I write something to a Word document and you correct the text and load it to the manual?

rgleason, 2023-02-01:
Hakan that would be a fine way to get this done. Thanks so much.