OpenCPN/OpenCPN #2750: OpenCPN locks when fed SignalK from NMEA 2000

mgrouch, 2022-08-20:
Multiple reports

https://forum.openmarine.net/showthread.php?tid=4188&pid=23471#pid23471

https://github.com/bareboat-necessities/lysmarine_gen/discussions/211#discussioncomment-3409458

https://www.cruisersforum.com/forums/f134/opencpn-5-6-2-signalk-connection-causes-freeze-265999.html

bdbcat, 2022-08-20:
Hoping to see a logfile of the sK data stream.

wsturner88, 2022-08-20:
Sorry, I am a bit new to this...   What is the best way to get you a log file?

transmitterdan, 2022-08-20:
Sk server can stream its data to a file and then later we can play it back into O for diagnostic purposes. If we had a saved stream that causes O to misbehave we could find the cause.

wsturner88, 2022-08-20:
I am a bit new to SignalK how to you send the data to the file? Is it under server log?

mgrouch, 2022-08-20:
To save into sk.log file from command line:
````

wsdump 'ws://localhost:3000/signalk/v1/stream?serverevents=all&subscribe=all&sendMeta=all'  > sk.log

````

wsturner88, 2022-08-20:
[sk.log](https://github.com/OpenCPN/OpenCPN/files/9387617/sk.log)
Let me know if this works

transmitterdan, 2022-08-20:
Did O lockup while you were making this file?

wsturner88, 2022-08-20:
No...  SignalK never locks up...  only OpenCPN when receiving signalK data.    If I go straight to NMEA source OpenCPN is fine

transmitterdan, 2022-08-20:
We need the log data from a session where OpenCPN locked up. We need the data that causes OpenCPN to malfunction.

wsturner88, 2022-08-20:
To detail the issue a bit more:
I am using MiniPlex to convert my NMEA 2000 backbone to the network on my boat.   

setup signal K to to received this data via UDP 10110
OpenCPN get it's data from SigalK
OpenCPN Locks up.

If I setup OpenCPN to get data directly from the MiniPlex - No issues.

If I understand the log correctly, this data is what is being handed off to OpenCPN from SignalK?

Does that help?

transmitterdan, 2022-08-20:
Ok, we will try to duplicate the hang up.

transmitterdan, 2022-08-21:
I have duplicated the hang.  Investigating.

transmitterdan, 2022-08-22:
Update on findings:

1) Having trouble using the Signal K server playing the log file because the server plays the file as fast as possible.  The file covers about 50 minutes of real time, but the server plays the entire file in 13 seconds on my RPi 3.  This overwhelms OpenCPN because it is dealing with thousands of Signal K messages in that short time.  The event queue becomes full of events and O appears to freeze because the event handler is doing nothing but queuing-dequeuing Signal K messages.  Maybe that is also happening to the user, but I cannot be sure.

2) The log file submitted above has a bunch of esc codes apparently meant to color the output on a terminal. I have removed all of that and made it into a suitable log file here:
[sk.log](https://github.com/OpenCPN/OpenCPN/files/9391474/sk.log)

3) The plugin code in OpenCPN sends every Signal K message to all plugins even though most of the messages are of zero interest to any plugin.  Maybe this is part of the problem because the flood of messages into O also creates a flood of messages to plugins.  Hard to tell unless there is a way to slow down the Signal K server when it is playing a log file.

Does anyone know how to throttle the Signal K server when it is using a FileStream/SignalK data source?

transmitterdan, 2022-08-22:
I found a Signal K plugin that can play a file with throttle controls. I am now playing the log file using a build from current master and it seems to work as expected.  Will play it overnight and see how it looks in the morning.  So far, it appears that the only way to make OpenCPN "hang" is to flood it with thousands of messages per second.  It is not surprising that will cause problems I think.

transmitterdan, 2022-08-22:
[image]

Playing the log file this is what I see on the chart window.

mgrouch, 2022-08-22:
raw option will produce more usable results

````
wsdump --raw 'ws://localhost:3000/signalk/v1/stream?serverevents=all&subscribe=all&sendMeta=all'
````

transmitterdan, 2022-08-24:
I am fairly certain the cause of this issue is a high number of SignalK messages (deltas) per second dispatched.  In the SignalK server web page it shows the message rate like this:

[image]

The relatively new webSocket thread system does not have any way to throttle incoming message rate.  It posts every incoming socket event into the message queue.  In the case I have been able to duplicate the "freeze", the message queue grows without bounds and OpenCPN spends all its CPU time posting SignalK messages to the exclusion of all else.

It may be a simple workaround to look at the SignalK server dashboard and see how many deltas/second are being dispatched.  This is displayed on the SignalK server home page.  If it exceeds 100-200, then examine the SignalK data source and determine why so many message are being created.  Try to find a way to limit the message rate to a reasonable value.  Or get a faster computer that is able process all these messages.

hans-martijn, 2022-08-24:
[image]

This is my average load from signalk, but my OpenCPN crashes after a minute or usually less. It completely freezes, and I'm sure it's not the amount of messages it gets.

I use a raspberry pi 4 4GB with the latest 64bit raspberrypios, flatpak OpenCPN 5.6.2. I have to admit, I cannot see the Options window either if I don't maximise it.

wsturner88, 2022-08-24:
I have 

> raw option will produce more usable results
> 
> ```
> wsdump --raw 'ws://localhost:3000/signalk/v1/stream?serverevents=all&subscribe=all&sendMeta=all'
> ```

I have switched over to the Openplotter build which has far less plugin's running in SignalK.  Same result, OpenCPN locks up shortly after (10sec) after start.   
74 Delta's per second
34 SignalK paths.

I tried the wsdump on this version and "(bash: wsdump: command not found).     Sorry I am not fantastic at unix, but is there a way I can execute this.. I think the data off this version will be much more clean.

wsturner88, 2022-08-24:
[image]

transmitterdan, 2022-08-24:
@wsturner88 if you shut down the SignalK server and wait for several minutes (maybe 15 minutes) does OpenCPN return to normal responsiveness?

wsturner88, 2022-08-25:
I did some experimenting this afternoon.   My system has the power to the NMEA backbone split so I can run the wind sensor with out the instruments outside.   This was done so I can adventually get wind remote.  Here is what I found:
1. Reboot - Turn on Signal K only with the following - Wind/Speed/Depth/yawl/pitch/ais - OpenCPN - No issues no lockups - 30 signal Paths at 30 delta's per second.
2. Reboot - Turn on SignalK and start OpenCPN - With all the above data I also added Rudder/GPS - 34 Paths / 54 Delta's - LOCKUP (This also enabled all of my NMEA 2000 instruments which are Triton2)

Is there any chance that rudder angle data could be the issue?

wsturner88, 2022-08-25:
> @wsturner88 if you shut down the SignalK server and wait for several minutes (maybe 15 minutes) does OpenCPN return to normal responsiveness?

No Tried for 30 min and it remained locked.

tkurki, 2022-08-28:
I am a little late to the party, but the sk server input side is not unlike a shell pipe: something generates data and then transforms turn it into SK deltas. Naive throttling playback can be achieved by adding a generic throttle element to the pipeline. 

So if you modify https://github.com/SignalK/signalk-server/blob/master/settings/volare-file-settings-filtered.json like so 
```
      "pipeElements": [
        {
          "type": "providers/filestream",
          "options": {
            "filename": "samples/plaka.log"
          },
          "optionMappings": [
            {
              "fromAppProperty": "argv.nmeafilename",
              "toOption": "filename"
            }
          ]
        },
        {
          "type": "providers/throttle",
          "options": {
            "rate": 10000
          }
        },
        {
          "type": "providers/liner"
        },
        {
          "type": "providers/from_json"
        }
```
then you can run the file with `bin/signalk-server --nmeafilename=./sk.log -s your-settings-file.json`. Throttling is in bytes/s.

There's code that does playback by timestamps ("real time", but it is not readily accessible for delta playback).

tkurki, 2022-08-28:
Everybody: feel free to ping me for SK related assistance.

tkurki, 2022-08-28:
@transmitterdan are you sure you are barking up the right tree with the message rate? We've had incidents where the ws client is not reading the websocket stream fast enough (or not at all) and the outgoing buffers causing trouble. Now the server monitors the buffer and if it grows above a set limit the connection is severed to protect the server. I'd imagine that happening if O is not ingesting data fast enough.

https://github.com/SignalK/signalk-server/wiki/FAQ:-Frequently-Asked-Questions#websocket-connections-are-being-dropped-and-syslog-shows-send-buffer-overflow-terminating-connection

transmitterdan, 2022-08-28:
In my testing that's exactly what happens. The server tries to reset the connection when O gets overloaded and stops responding.

transmitterdan, 2022-08-28:
I have seen a possible issue related to opengl.  It is intermittent.  Anyone who has seen the OpenCPN stall when connecting to SignalK please report whether or not the opengl option is checked (Options-Display-Advanced at the bottom of the dialog).

hans-martijn, 2022-08-28:
I have it checked, hanging is almost instant. As far as I can remember I turned it on for a plug-in, but I don't remember which one. O-charts, dashboard, engine dashboard or AIS-overlay.

transmitterdan, 2022-08-29:
@hans-martijn can you attach a copy of the log file ~/.opencpn/opencpn.log?

hans-martijn, 2022-08-29:
[opencpn.log](https://github.com/OpenCPN/OpenCPN/files/9443150/opencpn.log)

I have a flatpak installation, so it's ~/.var/app/org.opencpn.OpenCPN/config/opencpn/opencpn.log, I hope it's the same.

transmitterdan, 2022-08-29:
There is no indication of a problem in the log file.  It appears that O just stops logging which is what an infinite loop hang would do.

I cannot reproduce this so far except by flooding hundreds of SignalK messages per second into OpenCPN.  I have only one SignalK log file and I am not sure it is actually known to be recorded from a session the locked up OpenCPN.  If anyone can make more SignalK log files in native SignalK format that actually lock of OpenCPN I can try again to reproduce the problem.  Without a reliable way to reproduce it I cannot make further progress.

hans-martijn, 2022-08-29:
Hi Dan,

I tried without OpenGL, and I had no problems at all (except it being slow). Unfortunately, I'm home now, but I do have an old log. 

Hans

[sk-delta-log.2022-08-20T07:33:43.994Z.log](https://github.com/OpenCPN/OpenCPN/files/9443854/sk-delta-log.2022-08-20T07.33.43.994Z.log)

transmitterdan, 2022-08-29:
After setting the MMSI to 244187570 which I think is the vessel MMSI the log file plays ok on my RPi 4.  OpenCPN does not hang.

I am curious about the OpenGL setting.  According to the source code for Arm processors this is turned off automatically by the startup code.  Where did you get the flatpak version you are running?

wsturner88, 2022-08-29:
I found the on [Cruiserforum](https://www.cruisersforum.com/forums/f134/opencpn-5-6-2-signalk-connection-causes-freeze-265999-3.html)

Trying to find the Beta that this was released in to see if it fixes the issue.   If the Course over ground is the issue, it would make sense in my case.   When I deactivate the GPS NMEA feed the issue goes away.

on...
Its late here, but [food](https://www.cruisersforum.com/forums/tags/food.html) for thought today...

The RMC messages as processed by sK will contain some "NULL" values, especially for CourseOverGroundTrue.

{"updates":[{"source":{"sentence":"RMC","talker":"GP","type":" NMEA0183","label":"ais"},"timestamp":"2022-07-01T02:24:32.000Z","values":[{"path":"navigation.position","value":{"longitude" :153.18893266666666,"latitude":-27.4544215}},{"path":"navigation.courseOverGroundTrue","value":null},{"path":"navigation.speedOverGround","value":0.0 020577782990816138},{"path":"navigation.magneticVa riation","value":null},{"path":"navigation.magneti cVariationAgeOfService","value":1656642272},{"path ":"navigation.datetime","value":"2022-07-01T02:24:32.000Z"}],"$source":"ais.GP"}],"context":"vessels.urn:mrn:imo:mmsi:503447900" }

We have seen OCPN 562 react poorly to this, in this code:

Code:
void SignalKEventHandler::updateNavigationCourseOverGround(
    wxJSONValue &value, const wxString &sfixtime) const {
  double cog_rad = value.AsDouble();
  double cog_deg = GEODESIC_RAD2DEG(cog_rad);
  // wxLogMessage(wxString::Format(_T(" ***** COG: %f, %f"), cog_rad, cog_deg));
  m_frame->setCourseOverGround(cog_deg);
  m_frame->PostProcessNMEA(false, false, true, sfixtime);
}
If "value" is JSON_NULL, then reading value "AsDouble() will produce garbage. Later stages may fault with this undefined input.

This bug is fixed in OCPN github master, by still exists in 562.

I'll be back tomorrow with more info...
Dave

hans-martijn, 2022-08-29:
I turned on OpenGL myself, since that was required according to the Dashboard Tactics installation. Whet I turn it off I have no problems either, but OpenCPN slows down.
The installation is basically done through the basic manual flatpack version, an installation on the latest 64bits RaspberryOS.
I like Dashboard Tactics better than the normal Dashboard, since it's smaller than the normal Dashboard, has more options, and I have Engine Dashboard too, so I need the space.

transmitterdan, 2022-08-29:
So it appears turning off OpenGL will allow it to run on RPi. This is good news and offers a path to investigate. Thanks for all the replies.

P.S. the null values item are probably unrelated to this issue. The fix for that is in the latest git master.

mgrouch, 2022-08-29:
I also have OpenCPN locking up shortly after start. I disabled all plugins.
I do not have NMEA 2000. 

arm64 bullseye on raspberry pi4.

transmitterdan, 2022-08-29:
> I also have OpenCPN locking up shortly after start. I disabled all plugins.
> 
> I do not have NMEA 2000. 
> 
> 
> 
> arm64 bullseye on raspberry pi4.
> 
> 

This is surely unrelated to this issue. Please start another issue. And be aware I don't think we support that hardware/software/build config except via flatpak.

mgrouch, 2022-08-29:
I have some nulls from SignalK

```

   "updates" : [
      {
         "source" : {
            "label" : "gpsd",
            "type" : "NMEA0183",
            "talker" : "GN",
            "sentence" : "VTG"
         },
         "$source" : "gpsd.GN",
         "values" : [
            {
               "value" : null,
               "path" : "navigation.courseOverGroundMagnetic"
            },
            {
               "value" : null,
               "path" : "navigation.courseOverGroundTrue"
            },
            {
               "value" : 0.02666666667,
               "path" : "navigation.speedOverGround"
            }
         ],
         "timestamp" : "2022-08-29T17:20:53.054Z"
      }
   ],

  "updates" : [
      {
         "source" : {
            "label" : "gpsd",
            "type" : "NMEA0183",
            "talker" : "GN",
            "sentence" : "RMC"
         },
         "$source" : "gpsd.GN",
         "values" : [
            {
               "value" : {
                  "latitude" : 40.92966617,
                  "longitude" : -74.12679283
               },
               "path" : "navigation.position"
            },
            {
               "value" : null,
               "path" : "navigation.courseOverGroundTrue"
            },
            {
               "value" : 0.02880889619,
               "path" : "navigation.speedOverGround"
            },
            {
               "value" : "2022-08-29T17:20:54.000Z",
               "path" : "navigation.datetime"
            }
         ],
         "timestamp" : "2022-08-29T17:20:54.000Z"
      }
   ],

```

mgrouch, 2022-08-29:
Here is the last message from debug log before OpenCpn hangs:

```

17:21:22.205   DEBUG logger.cpp:130 {"updates":[{"source":{"sentence":"HDM","talker":"AP","type":"NMEA0183","label":"pypilot_nmea"},"timestamp":"2022-08-29T17:21:22.200Z","values":[{"path":"navigation.headingMagnetic","value":3.7205707272033886}],"$source":"pypilot_nmea.AP"}],"context":"vessels.urn:mrn:signalk:uuid:3e220085-f3aa-4c4f-add1-bb8e1eedbfb1"}

17:21:22.205   DEBUG chart1.cpp:8678 setHeadingMagnetic: 213.173000
17:21:22.206   DEBUG pluginmanager.cpp:3007 OCPN_CORE_SIGNALK
17:21:22.206   DEBUG pluginmanager.cpp:3008 {
   "updates" : [
      {
         "source" : {
            "label" : "pypilot_nmea",
            "type" : "NMEA0183",
            "talker" : "AP",
            "sentence" : "ROT"
         },
         "$source" : "pypilot_nmea.AP",
         "values" : [
            {
               "value" : -5.817764175e-07,
               "path" : "navigation.rateOfTurn"
            }
         ],
         "timestamp" : "2022-08-29T17:21:22.204Z"
      }
   ],
   "context" : "vessels.urn:mrn:signalk:uuid:3e220085-f3aa-4c4f-add1-bb8e1eedbfb1"
}

17:21:22.206   DEBUG logger.cpp:130 {"updates":[{"source":{"sentence":"ROT","talker":"AP","type":"NMEA0183","label":"pypilot_nmea"},"timestamp":"2022-08-29T17:21:22.204Z","values":[{"path":"navigation.rateOfTurn","value":-5.817764174642799e-7}],"$source":"pypilot_nmea.AP"}],"context":"vessels.urn:mrn:signalk:uuid:3e220085-f3aa-4c4f-add1-bb8e1eedbfb1"}

```

Looks like rate of turn. Might be it can't process

"value" : -5.817764175e-07,

Thanks

mgrouch, 2022-08-29:
Tried again. One CPU is 100% by openCPN (some busy loop)
Last message in the log is rateOfTurn. But many of the were processed fine.

mgrouch, 2022-08-29:
If I shut down SignalK and leave OpenCPN with just input from GPSd then it doesn't hang.

Thanks

mgrouch, 2022-08-29:
I've blocked SignalK from sending navigation.courseOverGroundTrue
and made OpenCPN to calculate COG from position changes every 3 sec.

I do not see lock ups.

It seems OpenCPN doesn't handle 'null' from SignalK messages.

Thanks

wsturner88, 2022-08-29:
Thank you for all the work you have been putting into this..    
How do we get this corrected in openCPN?   Or is this a SignalK fix?

transmitterdan, 2022-08-29:
> Thank you for all the work you have been putting into this..    
> 
> How do we get this corrected in openCPN?   Or is this a SignalK fix?
> 
> 
> 
> 

I think it is already fixed in the git repo. I will check later today to be sure.

wsturner88, 2022-08-29:
I am a bit naive when it comes to some of the installs.. I am running through OpenPlotter Release..   Is there a walk-though to update from the repo?

mgrouch, 2022-08-29:
The fix requires a new OpenCPN release.
What I have for now is an ugly workaround.

Thanks

leamas, 2022-08-30:
@transmitterdan 
 > I think it is already fixed in the git repo. I will check later today to be sure.

Hm... If so, in which commit(s)?

transmitterdan, 2022-08-30:
Would be nice to apply these fixes downstream on armhf debian native as well as flatpak.  I think most of the openplotter complainants are running armhf on RPi.

transmitterdan, 2022-08-30:
It is easy enough for an OpenPlotter user to replace the main O executable. And the OpenPlotter team accepts downstream patches fairly easily.

leamas, 2022-08-30:
IMHO we need this from Hubert, but then again ^^

mgrouch, 2022-08-30:
> IMHO we need this from Hubert, but then again ^^

@leamas 
I hope Debian patched package would be available soon. Do you have direct commit rights into OpenCPN Debian?

thanks

transmitterdan, 2022-08-30:
> I hope Debian patched package would be available soon. Do you have direct commit rights into OpenCPN Debian?
> 
> 
> 
> thanks
> 

I can post a link to a .deb package which can be installed with dpkg.  Give me an hour to upload it.

leamas, 2022-08-30:
>  Do you have direct commit rights into OpenCPN Debian?

Yes.

transmitterdan, 2022-08-30:
For those running armhf Raspbian Buster there is a test package here:

https://dl.cloudsmith.io/public/dan-dickey/opencpn-raspbian-unstable/deb/raspbian/pool/buster/main/o/op/opencpn_5.6.2/opencpn_5.6.2-1_armhf.deb

sudo dpkg -i opencpn_5.6.2-1_armhf.deb

If anyone can test and confirm the SignalK hang is not present in this version, it would be helpful.

mgrouch, 2022-08-30:
I’d need Bullseye arm64 version. deb package. 

thanks

mgrouch, 2022-08-30:
Does code patch take care of all cases where OpenCPN reads a value from SignalK json? I think null can potentially be in any of them.

thanks

transmitterdan, 2022-08-30:
I think it handles all the cases OpenCPN core cares about. Plug-ins are a different case. Try it and see if it works or not.

wsturner88, 2022-08-30:
@mgrouch if you can release a BBN version, I would be more then happy to test on the boat..

[18 more replies]

closed 2023-02-27 by nohal