OpenCPN/OpenCPN #2752: CI: MacOS builds broken

leamas, 2022-08-21:
**Describe the bug**
The MacOS CI builds are broken since some time, failing to rsync the macports sources

**To Reproduce**
1. Make a commit to OpenCPN repo.
2. Push
3. Watch logs with failing MacOS build

**Expected behavior**
Well...

**Additional context**
Typical error log here: https://app.circleci.com/pipelines/github/leamas/OpenCPN/2215/workflows/663bc861-a648-439a-9c0c-8613af7752ce/jobs/3907

Currently most OpenCPN work is on the comms branch, so the errors shows up here. Error has showed up for about a week,  last successful build about three weeks ago here: https://app.circleci.com/pipelines/github/leamas/OpenCPN/2185/workflows/3df9046a-f1cb-419c-bb7e-608e0955ee81

leamas, 2022-08-21:
@nohal: ^

nohal, 2023-02-02:
I think we are good for now.

closed 2023-02-02 by nohal