OpenCPN/OpenCPN #2753: Infinite recursion happens when closing O with tracking turned on

transmitterdan, 2022-08-24:
**Describe the bug**
Current OpenCPN master branch crashes during close when tracking is turned on.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on tracking
2. Close O
3. Crash

**Expected behavior**
No crash

**Desktop (please complete the following information if applicable):**
 - OS: Windows 11
 - 
 
MyFrame::OnCloseWindow calls MyFrame::TrackOff calls RouteMan::DeleteTrack calls MyApp::TrackOff calls MyFrame::TrackOff

This blows the stack up until O crashes.

transmitterdan, 2022-08-24:
I think this patch may fix the issue but I could not find when the potential recursion came into existence.  I don't think it has been there a long time yet git history does not reveal any recent activity that created it.

```
diff --git a/src/routeman.cpp b/src/routeman.cpp
index f17c7fc67..788523839 100644
--- a/src/routeman.cpp
+++ b/src/routeman.cpp
@@ -1060,7 +1060,7 @@ void Routeman::DeleteTrack(Track *pTrack) {
       pTrackPropDialog->Hide();
     }

-    if (pTrack == g_pActiveTrack) {
+    if (pTrack == g_pActiveTrack && g_pActiveTrack->IsRunning()) {
       pTrack = m_pparent_app->TrackOff();
     }
     //    Remove the track from associated lists
```

closed 2022-08-31 by transmitterdan

leamas, 2022-08-31:
@transmitterdan : 

Issue closed since patch above handles it or for other reasons?

transmitterdan, 2022-08-31:
The root problem is a commit from about 4 months ago into master. Will make a new issue once I figure out what exactly happened. But the issue happened after Release_5.6.2 and does not impact 5.6.2-2.

leamas, 2022-08-31:
OK, thanks