OpenCPN/OpenCPN #2782: TP Plugins are not going to have build-flatpak-arm64 metada

rgleason, 2022-10-06:
build-flatpak-arm64 builds are not working at the moment.
Jon and I have tried to find out what has changed but that information does not appear to be available.
Therefore we will be dropping build-flatpak-arm64 metadata from plugins being built.
We will then reincorporate when it is available and when convenient, on a subsequent update.

leamas, 2022-10-06:
" are not working " is kind of a description which doesn't make it possible to handle. Please provide details.

leamas, 2022-10-06:
One a sidenote, the shipdriver plugin seems to build and load just fine for flatpak-aarch64. Metadata for that file is available [here](https://dl.cloudsmith.io/public/alec-leamas/opencpn-plugins-unstable/raw/names/ShipDriver-3.2-flatpak-A64-20.08-metadata/versions/3.2.0+14869.31bac30/ShipDriver-3.2-flatpak-A64-20.08.xml)

rgleason, 2022-10-06:
Alec, see the error report here. It just fails. There is no comprehensible "error" that I can find. There are some deprecated calls.
https://app.circleci.com/pipelines/github/rgleason/weatherfax_pi/320/workflows/de977eb4-3b67-4e8c-bdaf-4176d31b6e7d/jobs/4198

leamas, 2022-10-06:
First of all: this is not an OpenCPN issue. The core problem is a testplugin (TP) build issue, this should be filed against testplugin. If there is a need to inform a wider audience about the TP state, this rightly belongs to a Show and Tell Discussion item.

Ergo: this issue should be closed. 

Looking at an upcoming testplugin issue I suggest that you and/or @jongough  looks into the shipdriver build script which does work. 

All discussions about wxWidgets version, 5.7.1  and what not are basically moot. There are no ABI changes at present point, they will appear in the 5.8.0 beta cycle, perhaps in November or so.

bdbcat, 2022-10-06:
I investigated.
All flatpak platforms except amr64 work correctly, and all use the same script.  So I think the script is OK.
I forked the repo, and followed the CI build on my account. And arm64 failed.  So I re-ran, with ssh shell available after the fail.
And it is strange.

The command to add flathub repo seems to work fine.
$ flatpak remote-add --user flathub http://dl.flathub.org/repo/flathub.flatpakrepo

Next, the command to fetch and install the freedesktop SDK fails, stangely.
$ flatpak install --user  flathub org.freedesktop.Sdk//20.08  
Looking for matches…
error: Nothing matches org.freedesktop.Sdk in remote flathub

Hmmm...
So, I try:
$ flatpak remote-list
Name    Options
flathub user

OK.
Then I try:
$ flatpak remote-ls

No response.
It seems to me that "flathub" is somehow not accessible to the flathub system.
I have not compared the scripts to shipdriver build....

leamas, 2022-10-06:
The problem is the outdated flatpak version: https://github.com/Rasbats/shipdriver_pi/issues/457. Related fix https://github.com/Rasbats/shipdriver_pi/commit/ae037342e87f3

bdbcat, 2022-10-06:
OK, I see that.  Will verify.
Aside:  How does this happen?  What broke flatpak in the standard repo, if you know?
We should generally prefer not to depend on private PPA builds (except OCPN core, of course).

jongough, 2022-10-06:
This appears to be a problem with flathub/flatpak, focal is hardly 'very old Focal flatpak' as described in your issue. We are still supporting Debian 9 (Stretch), Debian 10 (Buster), Ubuntu 16.04 LTS (Xenial), Ubuntu 18.04 LTS (Bionic), and Ubuntu LTS 20.04 (Focal)/ Ubuntu 22.04 LTS (Jammy) has only just been release and Ubuntu Focal still has >2 years support. Does this mean that flatpak will only support the latest releases? If so, how do we handle those who are not on the bleeding edge?

I used Win XP as an example of supporting environments well beyond the date at which they ceased to exist because users still use them. Has this policy now be dropped and we will only support the latest versions?

bdbcat, 2022-10-06:
"Has this policy now be dropped and we will only support the latest versions?"
Well, no.
Evidently, no-one using the outdated flatpak on some platforms will be able to build any app, including OCPN.  So, obviously, if we want to support flatpak on arm64, we will do what is necessary to make it work. 
Best not to try to extend specific solutions into into blanket policy.  It is never that simple...

jongough, 2022-10-06:
There is a new version of testplugin_pi 1.0.215.0 that will build all the current environments. It uses the "ppa:alexlarsson/flatpak" respository the same as shipdriver_pi.

leamas, 2022-10-07:
> ? What broke flatpak in the standard repo, if you know?

I *think* they updated the flathub repo "database format" so to speak so that ancient versions couldn't handle it. Seems like the only possible explanation, but I haven't checked further.

> We should generally prefer not to depend on private PPA builds

Agreed. However, this is sort of semi-official, the repo owner Alexander Larsson is the main Flatpak developer. Since Flatpak moves faster than Debian/Ubuntu he (i. e., the Flatpak project)  provides these packages which are up to date. It's not the first time we are using them.

Ideally, the Flatpak packages would be backported. However, backporting non-leaf packages is utterly complicated due to the dependencies, so I think the PPA is what's possible..

rgleason, 2022-10-07:
It would be nice to have some semophore communication about this type of thing when it occurs, but then there is so much communication it my pass us by anyway.

```
the repo owner Alexander Larsson is the main Flatpak developer. Since Flatpak moves faster than Debian/Ubuntu he (i. e., the Flatpak project) provides these packages which are up to date. It's not the first time we are using them.

Ideally, the Flatpak packages would be backported. However, backporting non-leaf packages is utterly complicated due to the dependencies, so I think the PPA is what's possible..
```

closed 2023-01-13 by rgleason