OpenCPN/OpenCPN #2783: Display wave height/period/white cap probability/boat letf-right heeling period/amplitude on dashboard

mgrouch, 2022-10-06:
Display wave height/period/white cap probability/boat left-right heeling period/amplitude/G-force on dashboard

Assuming pypilot can implement it from IMU readings.

You might have to agree on custom NMEA sentence for it.

Link: https://github.com/pypilot/pypilot/issues/168

rgleason, 2022-10-29:
What methods are used by  NDBC  to measure significant wave height at buoys?   Note that Hs symbol is now WVHT.
[Graphic representation (Hs is now WVHT)](https://www.weather.gov/dlh/WaveHeightExplanation).
[How are spectral wave data derived from buoy motion measurements?](https://www.ndbc.noaa.gov/wave.shtml)  "..accelerometers or inclinometers on board the buoys measure the heave acceleration or the vertical displacement of the buoy hull .. A Fast Fourier Transform (FFT) is applied to the data by the processor on board the buoy to transform the data from the temporal domain into the frequency domain. ...[and processed on board the bouy]."   For more detail see the three articles on FFT.

mgrouch, 2022-10-29:
I’m planning to play with this code:

https://github.com/ryanbarna/waveHeight