OpenCPN/OpenCPN #2799: The build should be possible without Internet connection

asy972, 2022-10-29:
I attempt to build current (20221029) master and I took a problem:

```
-- Found LibELF...
--     ELF Lib: /usr/include
--     ELF Include: /usr/lib64/libelf.so
-- Building for target (Platform; version; arch): alt-x86_64; n/a; x86_64
[ 11%] Creating directories for 'googletest-populate'
[ 22%] Performing download step (git clone) for 'googletest-populate'
Cloning into 'googletest-src'...
fatal: unable to access 'https://github.com/google/googletest.git/': Couldn't connect to server
```
Build host haven't access to Internet and this a normal practice. At the same time the system has installed test packages:

libgmock 1.12.1
libgtest 1.12.1
libgmock-devel 1.12.1
libgtest-devel 1.12.1

bdbcat, 2022-10-30:
Build without "test" subproject, like:
cmake -DOCPN_BUILD_TEST:BOOL=OFF ..

leamas, 2022-10-30:
Admittedly, we should check if there are system libraries available before starting to download and build sources.  Taking the ball...

asy972, 2022-10-31:
> Build without "test" subproject, like: cmake -DOCPN_BUILD_TEST:BOOL=OFF ..

Thanks, it's good as workaround. But build tests is a good idea by itself. It would be nice to leave the opportunity to use tests without Internet.

commit 2f8f4a5f (2022-12-05): Build: Use existing gtest libraries if existing (#2799).

If gtest is available, avoid downloading sources and use the
existing libraries instead.

This does not handle the gtest sources installed in
Debian/Ubuntu.

Closes: #2799
closed 2022-12-05 by bdbcat