OpenCPN/OpenCPN #2800: Kalman Filter issue

oscar-polheim, 2022-10-31:
I've compared Kalman Filter used in OpenCPN with the [one](https://github.com/lacker/ikalman) I use in my projects with GPS.

From what I can see, OpenCPN's KF seems not so good even compared to raw GPS.

Legend:
[image]

Sample is from a boat GPS at 10Hz:
[image]

Is this something known or expected? Or should I dig deeper and try to understand if I did something wrong using OpenCPN's KF.

rgleason, 2022-10-31:
Can you upload a VDR file of this, it might be helpful.

transmitterdan, 2022-10-31:
To my knowledge OpenCPN has no Kalman filter of GPS coordinate data.  It merely records the values sent to it using a heuristic that prevents recording the same or very small coordinate changes multiple times in a track.  So, you should look to your GPS device for coordinate filter settings.

OpenCPN optionally filters course over ground and speed over ground but that does nothing to the recorded track.

The lack of a coordinate filter is a design decision.  We did not think it a good idea to alter the coordinates received in any way as this creates ambiguity as to what coordinates are recorded in the track.  So, the decision was made to record the coordinates exactly as they arrive from the GPS device.  I think this is the appropriate design decision.

oscar-polheim, 2022-10-31:
@transmitterdan my mistake. Should've posted this in radar_pi plugin, which is using Kalman filter.

https://github.com/opencpn-radar-pi/radar_pi/issues/214

closed 2022-10-31 by oscar-polheim