OpenCPN/OpenCPN #2801: Do not allow to build route via land. Make it more fool proof

mgrouch, 2022-11-01:
Do not allow to build route via land. Make it more fool proof.

Currently anyone can create a route with waypoints for autopilot which will not make any sense.
It could cross land, etc. Building a route should be made more fool proof.

Thanks

transmitterdan, 2022-11-01:
I don't see how this is feasible.  We cannot know where all possible "land" or "shallow water" might be.  So, it would be foolish to attempt to promise a user that their chosen route does not cross land.  It is the responsibility of the navigator and not software to ensure a route is "safe".

leamas, 2022-11-01:
Actually, I once had a student which regularly used a route passing land -- the charts is not fully correct, the narrow channel he uses is misplaced.

But the route is fine :)

rgleason, 2022-11-01:
Try using weather_routing with land detection. It fails in complicated "close in" situations (islands and complex land/water interfaces)  and requires parameter adjustments, but does work reliably otherwise.

Opencpn could have a Land detection checkmark and use high res gshhs but that is not its primary role.

Please note that high res gshhs is not perfect and there are areas where it is not accurate.