OpenCPN/OpenCPN #2808: Toolbar positioning on Wayland broken

nohal, 2022-11-04:
**Describe the bug**

Toolbars are misplaced on Wayland + Gnome (Fedora 37)

Also there are other related issues:

 - Main toolbar can't be dragged anywhere
 - The hamburger options window sizing and positioning is broken

**Screenshots**
[image]

**Desktop**
 - OS: Fedora
 - Version 37
 - wxWidgets master codebase

nohal, 2022-11-14:
Problem not present with the wx 3.0.5 as delivered from the official repositories.

leamas, 2022-11-14:
hm... are you sure it is 3.0.5? The version in F37 is 3.2.1 (?)

nohal, 2022-11-14:
Both are available - `wxGTK3` is at 3.0.5, `wxGTK` is at 3.2.1
Using the former we do not have this problem, using the later or current wx codebase, the problem is present.

leamas, 2022-11-14:
This is a perhaps a known issue (?), I have some ancient bugs filed. If so,  the work around is  `export GDK_BACKEND=x11`. 

It it is lke I suspect  it might make sense to check for `XDG_SESSION_TYPE=wayland` in the environment and set GDK_BACKEND at startup. I have filed PRs for this long time ago, but they were not merged. wayland will hit all Linux users sooner or later, so this is not just about Fedora. Don't remember if it could be done inside the opencpn program or if it must be  set before it's started in a wrapper script.

nohal, 2022-11-14:
Just tested, you are right as usually. With `GDK_BACKEND=x11` the problem is gone.

leamas, 2022-11-14:
The remaining issue is then to ensure we have this in the environment. Again, this is Wayland problem rather than Fedora. OTOH, it could be argued it is a packaging issue. 

In any case, it should at least be documented somehow...

nohal, 2022-11-14:
The variable needs to be set in a wrapper script, even our `main()` is too late.

leamas, 2022-11-14:
I guessed so. 

For whatever reason, this does not happen in the Debian world even when using Wayland. I now remember this was the reason my patches was not merged.  As I see it it means that if someone undertakes to package opencpn for Fedora, she will need to cope with this, moving the binary and replace it with a wrapper script.

OTOH, our official support on Fedora is Flatpak which does not exhibit this bug. So, perhaps this just should be closed as NOTABUG, dunno.

nohal, 2022-11-15:
I would probably at least print some warning about possible problems in the log, otherwise this will keep coming back over and over again. Will try to have a look if it is actually somehow detectable from the size/position values we have available...

leamas, 2022-11-15:
I'm not sure this is an actual problem, I seriously doubt we have or will have any Fedora users on native packages. The reason  is that while it's possible to compile opencpn (and thus make an rpm), there are no native Fedora  plugins available.

OTOH, we know users are using Flatpak which works out of the box on all relevant distros. Using Flatpak also means that the ordinary set of plugins is available. This is actually the first bug reported for a very long time for non-Debian based Linux distros like Fedora i. e., distros targeted by Flatpak.

leamas, 2022-11-16:
This is not really about wx32, it's far older: https://opencpn.org/flyspray/index.php?do=details&task_id=2198 and https://github.com/OpenCPN/OpenCPN/issues/1166.

leamas, 2022-11-16:
The question is if we have any users on affected platforms. I'd basically say no, but might of course be wrong.

If we want to support native Fedora (and probably some other Linux platforms) we could probably rebase the patch in #1166 and use that.  It should not affect the Debian-based builds.

In the end, it's kind of a policy discussion. So far we have put our efforts to support non-Debian Linuxes in the Flatpak basket and let native builds like Fedora bit-rot.

nohal, 2022-11-21:
Seems even Ubuntu is seeing the same problem now https://www.cruisersforum.com/forums/f134/opencpn-version-5-7-1-alpha-test-2-a-270357-3.html#post3709184

nohal, 2022-12-01:
New wxGTK update just landed on Fedora with the following changelog entry

```
Changelogs for wxGTK-3.2.1-3.fc37.i686, wxGTK-3.2.1-3.fc37.x86_64
* Tue Nov 22 12:00:00 AM 2022 Scott Talbert <swt@techie.net> - 3.2.1-3
- Rebuild (again) with wxGLCanvas GLX support (resolves many OpenGL issues)
```
resulting in the following messages being delivered when OpenCPN is started:
[image]

So unless we want to do something about it, this issue can be closed as the user gets the info needed.

bdbcat, 2022-12-01:
For info, the default Ubuntu Jammy installation presents with Wayland.  We can reasonably assume that this will be true for all subsequent ubuntu releases, as well.
So let's talk about a reasonable solution for production.
Where/how can we introduce a script to set GDK_BACKEND before starting O58?

bdbcat, 2022-12-01:
I experimented, implemented a trivial patch that looks like this:

```
MyApp::MyApp() {
#ifdef __linux__
// Handle e. g., wayland default display -- see #1166.

  if (wxGetEnv( "WAYLAND_DISPLAY", NULL))
    setenv("GDK_BACKEND", "x11", 1);

#endif   // __linux__
}
```

In github master now.
Works on ubuntu Jammy.
Elsewhere?

nohal, 2022-12-01:
Good on Fedora as well, I think we are done here for now.

closed 2022-12-01 by nohal