OpenCPN/OpenCPN #2809: Unwanted artifacts on ENC charts

nohal, 2022-11-04:
**Describe the bug**
There are unwanted line artifacts rendered in ENC cells as soon as the objects they contain are removed from the display but the cell is still part of the quilt set.

**To Reproduce**
Zoom in/out until artifacts appear in some of the cells

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
 - OS: Fedora
 - Version: 37
 - Wayland
 - OpenGL

**HW info**

```
17:24:07.094 MESSAGE chcanv.cpp:2385 Metrics:  m_display_size_mm: 508     g_Platform->getDisplaySize():  1920
:1080   
17:24:07.094 MESSAGE chcanv.cpp:2390 wxDisplaySize(): 1920 1080
17:24:07.094 MESSAGE TCDataSource.cpp:50 Loading Tide/Current data source: /usr/local/share/opencpn/tcdata/ha
rmonics-dwf-20210110-free.tcd
17:24:07.223 MESSAGE TCDataSource.cpp:50 Loading Tide/Current data source: /usr/local/share/opencpn/tcdata/HA
RMONICS_NO_US.IDX
17:24:07.306 MESSAGE glChartCanvas.cpp:1051 OpenGL-> Renderer String: RENOIR (renoir, LLVM 15.0.0, DRM 3.47, 
5.19.16-301.fc37.x86_64)
17:24:07.306 MESSAGE glChartCanvas.cpp:1060 OpenGL-> Version reported:  4.6 (Compatibility Profile) Mesa 22.2
.0
17:24:07.306 MESSAGE glChartCanvas.cpp:1067 OpenGL-> GLSL Version reported:  4.60
17:24:07.308 MESSAGE glChartCanvas.cpp:1150 OpenGL-> Texture rectangle format: de1
17:24:07.308 MESSAGE glChartCanvas.cpp:1165 OpenGL-> Using Vertexbuffer Objects
17:24:07.309 MESSAGE glChartCanvas.cpp:1022 OpenGL-> Framebuffer OK, size = 1918
17:24:07.309 MESSAGE glChartCanvas.cpp:1230 OpenGL-> Using Framebuffer Objects
17:24:07.309 MESSAGE glChartCanvas.cpp:1233 OpenGL-> Using FBO Stencil buffer
17:24:07.309 MESSAGE glChartCanvas.cpp:1240 OpenGL-> Using Stencil buffer clipping
17:24:07.309 MESSAGE glChartCanvas.cpp:1245 OpenGL-> Using Scissor Clipping
17:24:07.309 MESSAGE glChartCanvas.cpp:1352 OpenGL-> Using s3tc dxt1 compression
17:24:07.309 MESSAGE glChartCanvas.cpp:1381 OpenGL-> Compressed tile size: 128kb (8:1)
17:24:07.309 MESSAGE glChartCanvas.cpp:1256 OpenGL-> Minimum cartographic line width:  1.0
17:24:07.309 MESSAGE glChartCanvas.cpp:1259 OpenGL-> Minimum symbol line width:  1.0
17:24:07.341 MESSAGE glChartCanvas.cpp:548 BuildFBO 3
17:24:07.341 MESSAGE glChartCanvas.cpp:1022 OpenGL-> Framebuffer OK, size = 1918

```

bc38, 2022-11-05:
On Windows, I also noticed artifacts with stripes, including on BUAARE. And also noticed the loss of vertices on some polygons of restricted area, anchorages area. Examples CPN, 7cs ENC Viewer
[image]
[image]

In my opinion, the display of the line of the edge of the obstruction is not rendered correctly. It was correct before. I give examples.
[image]
[image]

https://nauticalcharts.noaa.gov/publications/docs/us-chart-1/ChartNo1.pdf view from page 52

bc38, 2022-11-05:
example with location coordinates on CM93
[image]

bdbcat, 2022-11-05:
Nohal...
Sorry, I cannot immediately reproduce this.
Can you give me a repeatable recipe?
Dave

nohal, 2022-11-06:
I can reproduce it only on the Fedora 37 system with Wayland (On both the APU integrated GPU and Radeon 6700). Not reproducible at all on my other Linux machines.

bdbcat, 2022-11-06:
And how do you reproduce it on F37/Wayland?  It may give me some ideas...

nohal, 2022-11-06:
Simply zooming in and out on FL ENCs, nothing special needed.

bdbcat, 2022-11-06:
This looks to me like it relates to a specific object class.  So I would do the following:
1.  Switch to "Base" category.  Still trourble?
2. Hack S52 library to NOP some render code, looking for the one object class the induces the trouble.
Example:
  Turn OFF simple line rendering.
  int s52plib::RenderGLLS(ObjRazRules *rzRules, Rules *rules) {
     return 0;
.
.
.
}

I think it may be this:  some GL shader transform is active, but there is some leftover direct-mode rendering happening somewhere.  So the per-object transform matrix is FUBAR.  Just a guess.
Dave

bdbcat, 2022-11-06:
And one more:
Switch to "plain" rather than "symbolized" area boundaries.

bc38, 2022-11-06:
In the "plain" mode, everything is fine, the vertices do not break
Win11 64x, nvidia GF1070 
[image]

bdbcat, 2022-11-06:
bc38...
Yes, I see that.  I can reproduce locally, as well, with cm93 at the indicated location.
I will take a closer look.  Thanks

What is the location of the previous image, with error on edge of obstruction?

bc38, 2022-11-06:
> Где находится предыдущее изображение с ошибкой на краю микрорайона?

[2V403022.zip](https://github.com/OpenCPN/OpenCPN/files/9945167/2V403022.zip)

nohal, 2022-11-14:
The artifacts seem to be related ro `UNSARE`, as soon as I disable Unsurveyed area objects, the problem is not reproducible anymore regardless of other settings. And still not reproducible at all on any of my other systems.

bdbcat, 2022-11-24:
Pavel...
I think I resolved this artifact.  Relates to different display adapters and/or drivers.  Some are more picky than others. I found a user with Windows10, exhibiting a similar bug, and Anydesk availability to debug.
Please verify from github master.

nohal, 2022-11-24:
Yes, the problem seems gone.

nohal, 2022-11-24:
Spoke too fast. The problem is still present when using the discrete GPU (Advanced Micro Devices, Inc. [AMD®/ATI] Navi 22 [Radeon RX 6700/6700 XT/6750 XT / 6800M]), just changed a little bit:
[image]

bdbcat, 2022-11-24:
One more try, in github now.
The root cause is that OpenGL drivers/cards seem to handle GL_POLYGON_SMOOTH in different ways.  What you are seeing is the NoDTA background of the chart as the driver tries to anti-alias some of the rendered triangles.  They should NOT be anti-aliased when drawing a large area of static chart geometry.  So I cleaned up the OpenGL state logic to ensure that.

nohal, 2022-11-24:
Yeah, this does the trick and also instantly fixes #2825 and #2826

closed 2022-11-24 by nohal