OpenCPN/OpenCPN #2813: RFE: Missing support for DSC NMEA alerts and messages

mgrouch, 2022-11-11:
Described here:

https://continuouswave.com/whaler/reference/DSC_Datagrams.html

More links:

See https://github.com/adrianmo/go-nmea/ (DSC, DSE)

https://www.cruisersforum.com/forums/f13/nmea-0183-sentences-dsc-dse-124887.html

https://continuouswave.com/whaler/reference/DSC_Datagrams.html

https://web.archive.org/web/20190303170916/http://continuouswave.com/whaler/reference/DSC_Datagrams.html

https://github.com/mariokonrad/marnav/blob/master/src/marnav/nmea/dsc.cpp

https://github.com/mariokonrad/marnav/blob/master/src/marnav/nmea/dse.cpp

Thanks

Hakansv, 2022-11-11:
There are support for DSC messages. Normally received from a VHF. We have though not heard a lot reports about the function. Troubled or not. I've seen a few on my test PC but they are rather rare. My VHF don't send these as NMEA0183.
Have you any reports to share?

mgrouch, 2022-11-11:
What is going to be displayed by OpenCPN when it receives DSC message?
Thanks

Hakansv, 2022-11-11:
It's a display as for a AIS CPA alert.
Something like this.

[image]

mgrouch, 2022-11-11:
Ok. Thanks.

I've tried these four.

```
$CDDSC,12,3380400790,12,06,00,1423108312,2019,,,S,E*6A
$CDDSC,12,3381581370,12,06,00,1423108312,0236,3381581370,,S,*20
$CDDSC,20,3381581370,00,21,26,1423108312,1902,,,B,E*7B
$CDDSE,1,1,A,3380400790,00,45894494*1B
```

First 2 worked, but I do not see nature of distress or flag in O 5.6.2. If I close OpenCPN distress location gets lost with all data about it. And sometimes OpenCPN crashes on shutting it down. Loses distress location data too.

Second 2: DSCone comes as position report and seems ok. No flag either.

DSE one (last one) has no effect at all. It's not reflected in O.

Thanks

Hakansv, 2022-11-12:
A DSE alert is normally received once and activate an Alert in OCPN. Nothing else. The info is not saved in any way so if you close O the info is naturally lost. Also is the alert shown until you acknowledge. 
The crash is hard to figure out. I've tested a lot DSE but not seen that. Would need more details.
About the messages:
This is a bit complex but let's see if I get it right;
There are more experts out here that may correct me.
Message 1 and 2 
What I can see both are kind of a mismatch.
Pos 1 = 12 says it's a distress but since Pos 3 = 12 it's a relayed report and then pos 1 would have been 16 (All ships)
Therefor the Nature of distress is not correct in our DSC Alert.
Message 3  is a position report (Pos 1 = 20) and not handled by OCPN
Message 4 is a DSE where we use the mmsi in Pos 3. and since Pos 4 = 00 we get its position from Pos 5. So yes it's used. 
But again since mes 1 and 2 says it's a relay the mmsi 3380400790 would refer to the base station so the position in the DSE seems useless?

See more [here](https://www.itu.int/dms_pubrec/itu-r/rec/m/R-REC-M.493-10-200005-S!!PDF-E.pdf) and this CF [post](https://www.cruisersforum.com/forums/f13/nmea-0183-sentences-dsc-dse-124887.html)
And OCPN [handling](https://github.com/OpenCPN/OpenCPN/blob/21607261303300dd6e912249244e57ea76ba7274/src/ais_decoder.cpp#L2268)

mgrouch, 2022-11-14:
I believe important ones needs to be saved upon receive. So they will not get lost on OpenCpn restarts/crashes.

Thanks

Hakansv, 2022-11-14:
I cant say I agree but that's me.
But if so: 
What's "important"? Specification please.
Saved? Saved where? An ECDIS Inmarsat system has a printer connected producing meters of paper. I suppose this is not your intention?

mgrouch, 2022-11-14:
Just try to think about it this way: You receive a distress message you close OpenCPN (or it crashes)
and you start OpenCPN again and distress location is gone.
I do not think it's a correct behavior.

Hakansv, 2022-11-14:
O shall not crash. Please try to reproductive again and describe actions leading to crash.

If you receive a DSC alert the position would be clear and alerted. Then you can judge if it's so close to you that it would lead to any actions from your side. If so you may listen carefully to the VHF where additional communication would appear. But why close OCPN meanwhile?
There could of course be a log file print for a received DSC alert if a common interest for that? But a special file or whatever else to save received DSC would not have much future interest?