OpenCPN/OpenCPN #2815: RFE: Missing Safety Messages NMEA 0183 ammendement support

mgrouch, 2022-11-11:
Safety Messages NMEA ammendement

SM1, SM2, SM3, SM4

http://www.nmea.org/Assets/20130816%20nmea%200183%20safetynet%20amendment.pdf

Examples:

```

/g:1-9-1234,s:egcterm1,n:213,c:1333636200*hh/$CSSM3,123456,005213,798,0,3,14,00,2012,04,05,14,30,3400,N,076,W,300*hh
/g:2-9-1234,n:214*hh/$CSSMB,008,001,0,123456,FROM:Maritime Rescue Coordination Centre xxx^0D^0ATO:*hh
/g:3-9-1234,n:215*hh/$CSSMB,008,002,0,123456, ALL SHIPS IN xxxxxxx^0D^0ASAR SITREP NO: 02^0D^0AFIS*hh
/g:4-9-1234,n:216*hh/$CSSMB,008,003,0,123456,HING BOAT 'xxx' WITH THREE PERSONS ON BOARD DEPARRTED*hh
/g:5-9-1234,n:217*hh/$CSSMB,008,004,0,123456, FROM xxx ISLAND ON^0D^0Axxx AT NOONTIME AND SINCE TH*hh
/g:6-9-1234,n:218*hh/$CSSMB,008,005,0,123456,EN NO INFORMATION ABOUT HER. PARTICULARS ...^0D^0ASHI*hh
/g:7-9-1234,n:219*hh/$CSSMB,008,006,0,123456,PS SAILING IN VICINITY ARE KINDLY REQUESTED TO KEEP A*hh
/g:8-9-1234,n:220*hh/$CSSMB,008,007,0,123456, SHARP LOOK OUT^0D^0AINFORMING MRCC^0D^0AREGARDS^0D*hh
/g:9-9-1234,n:221*hh/$CSSMB,008,008,0,123456,^0A DUTY OFFICER*hh

```

Thanks

mgrouch, 2022-11-11:
[SMV Amendment to NMEA 0183 Version 4.10 20180615](https://www.nmea.org/Assets/20180615_SMV_Sentence%20Amendment.pdf)

Hakansv, 2022-11-11:
So what pleasure craft would have any need for these massages? If I understand it correct it's satellite based so one would need a receiver for that connected to OCPN. Or use that receiver to handle such a message?
For me it's good enough to listen to VHF channel 16 for concerns in the neighborhood applicable for a small boat?

mgrouch, 2022-11-11:
> So what pleasure craft would have any need for these massages? If I understand it correct it's satellite based so one would need a receiver for that connected to OCPN. Or use that receiver to handle [...]

Is OpenCPN for pleasure boats only? Even on smaller boat one can install SDR and decode Inmarsat-C from it with a program.

Hakansv, 2022-11-11:
This could of course be discussed as a feature request for a new function. 
But I don't see it as an issue or bug. These are reports about failures in existing code.

rgleason, 2022-11-13:
Hakan, he has titled this RFE   (Request for Enhancement)

Hakansv, 2022-11-13:
Rick..
Ohh. that's an opportunity I may have missed.
Sorry Mikhail