OpenCPN/OpenCPN #2816: RFE: NMEA 0183 Man Overboard Notification (MOB) Sentence Support

mgrouch, 2022-11-11:
From NMEA Amendements:

https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwjWr4naw6b7AhWfGFkFHZA7AqAQFnoECBcQAQ&url=https%3A%2F%2Fwww.nmea.org%2FAssets%2F21030814%2520nmea%25200183_man%2520overboard%2520notification_mob_sentence%2520amendment.pdf&usg=AOvVaw3LOeiXPWW0nQSyDm3IKUp8

Thanks

mgrouch, 2022-11-11:
I've tried sending these:
```

$INMOB,ABCDE,V,123456.987,0,311219,132456.876,12.34,N,123.45,E,12.34,23.45,269104520,0*53
$CSSMV,2,1,5,12,123456789,,1234.5678.N,12345.6789,W,2018,01,23,12,34,D*35
$CSSMV,2,2,5,12,123456789,MAXIMUM LENGTH FOR VESSEL NAME,,,,,,,,,D*25

```

None of them had any reflection in OpenCPN.

Thanks

Hakansv, 2022-11-12:
That's correct. The NMEA0183 sentences "xxMOB" is not handled by OCPN. I'm not sure who may send them.

A AIS-SART message reporting a MOB is though handled.