OpenCPN/OpenCPN #2820: RFE: Need to distinguish when waypoint/route/track is in a layer

rgleason, 2022-11-19:
**Is your feature request related to a problem? Please describe.**  T
by Tony (Antipole)   https://opencpn.org/flyspray/index.php?do=details&task_id=2864

Need to distinguish when waypoint/route/track is in a layer.
This came up again with a user request/comment: [Re: Wanted: Smooth workaround to add wayoint to a layer](https://www.cruisersforum.com/forums/f134/wanted-smooth-workaround-to-add-wayoint-to-a-layer-270469.html#post3706452)

**Describe the solution you'd like**

1. The getGUIDs apis returns all the relevant GUIDs - waypoints, routes or tracks as appropriate.  This includes objects in layers, regardless of the layer visibility.
2. There is no way of determining whether an object is in a layer or not so it is not possible to determin the appropriate actions.
3. It seems possible to change layer objects but presummably these chanages are not saved on OpenCPN exit.
4. Actual issue: the HouseKeeper script finds objects with name issues or co-located with other points and will fixed them up - but the issues reappear on next run of OCPN
5. 

**Possible approaches:**

1. The getGUIDs apis need to be able to get either non-layer or layer objects or both (in which case we need to beable to indicate which they are)
2. OR the returned objects need a flag to indicate they are in a layer.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

bdbcat, 2022-11-19:
Tony:
How would this work for you?

```
typedef enum _OBJECT_LAYER_REQ {
  OBJECTS_ALL = 0,
  OBJECTS_NO_LAYERS,
  OBJECTS_ONLY_LAYERS
} OBJECT_LAYER_REQ;

extern DECL_EXP wxArrayString GetRouteGUIDArray(OBJECT_LAYER_REQ req);
extern DECL_EXP wxArrayString GetTrackGUIDArray(OBJECT_LAYER_REQ req);
extern DECL_EXP wxArrayString GetWaypointGUIDArray(OBJECT_LAYER_REQ req);

```
Dave

antipole2, 2022-11-20:
This would be great!

The present APIs of this name have no argument.  So could the argument be optional and if omitted default to OBJECTS_ALL for compatibility?
Actually, I doubt any user of these APIs has wanted OBJECTS_ALL, so I would probably make the JavaScript APIs default to  OBJECTS_NO_LAYERS for convenience.  But that can be a separate decision from the default in C++.

Using OBJECTS_ALL, there would be no way of distinguishing which of the returned objects is in a layer.  To do this, a flag would be needed in the objects.  However, if someone wants to be able to distinguish, they could retrieve both OBJECTS_NO_LAYERS and OBJECTS_ONLY_LAYERS and the distinction would be obvious that way.  So I don't think it worth the complication of adding a flag to the objects.

bdbcat, 2022-11-20:
"The present APIs of this name have no argument. So could the argument be optional and if omitted default to OBJECTS_ALL for compatibility?"
That we get automatically, since the previous definition of e.g. GetRouteGUIDArray() with no argument remains in the API.  c++ is nice, sometimes...

antipole2, 2023-02-06:
> "The present APIs of this name have no argument. So could the argument be optional and if omitted default to OBJECTS_ALL for compatibility?"
> That we get automatically, since the previous definition of e.g. GetRouteGUIDArray() with no argument remains in the API. c++ is nice, sometimes...

Myself, I don't care.  Have already prepped changes to JavaScript to use it and the simplest is to always use the parameter.  I am making the JavaScript default OBJECTS_NO_LAYERS (see above) but that is separate to how it defaults in C++.

antipole2, 2023-02-28:
Not yet implemented in OCPN 5.7.1 beta 4

bdbcat, 2023-03-01:
Tony...
The implementation of this API should now be available in current build:
https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/OpenCPN_5.7.1.pkg

Let me know how it goes....
Thanks

bdbcat, 2023-03-03:
Tony...
Any test results here?

antipole2, 2023-03-03:
Working on it right now.  Will advise here.

rgleason, 2023-03-03:
Any chance to get this done in the API?
"allow plugins access to the Show Text button for Squiddio"
https://github.com/OpenCPN/OpenCPN/issues/2614

rgleason, 2023-03-03:
Dave, It's here
 RFE: Plugin API - allow plugins access to the Show Text button for Squiddio #2614 [https://github.com/OpenCPN/OpenCPN/issues/2614]() 

Just hoping to catch the wave.

rgleason, 2023-03-04:
Isn't this completed?

bdbcat, 2023-03-04:
No, I am awaiting feedback from Tony.  He says:
"Working on it right now. Will advise here."
I guess your RFE is open, or not, but not related to this issue.  This is what happens with cross-posting of issues.  Please avoid in future.

rgleason, 2023-03-04:
You might note that, I am the one who posted this for Tony's issue as It was discussed on the forum and I thought it important. Sorry..

bdbcat, 2023-03-04:
Sorry, missed that point.  It is important, and Tony and I have been in discussion here....

antipole2, 2023-03-04:
Pleased to report this looks all OK.  Test JavaScript reports in my case:
```quote
Waypoint counts
All: 2464	Normal: 2421	Layer: 43
Sum check is true

Route counts
All: 307	Normal: 307	Layer: 0
Sum check is true

Track counts
All: 312	Normal: 109	Layer: 203
Sum check is true
```
I have checked the Layer counts against my Layers xml files
I have checked the All counts by running the same script on OCPN5.6.2
∴ The Normal counts must be correct

For interest, I attach the JavaScript. It needs my JavaScript v2.1.0 or later, not yet published.
[GetGuidTests.js.zip](https://github.com/OpenCPN/OpenCPN/files/10887908/GetGuidTests.js.zip)

rgleason, 2023-03-04:
Closing. Hope this ok

closed 2023-03-04 by rgleason

bdbcat, 2023-03-04:
Thanks, yes, complete.