OpenCPN/OpenCPN #2827: Artifacts stay on screen when panning down

nohal, 2022-11-24:
**Describe the bug**
When panning the chart in the downward direction, line artifacts stay on screen. Especially apparent with the GSHHG basemap, but also less prominently visible on ENCs

**To Reproduce**
Pan the basemap (slowly makes the issue more apparent) towards the bottom of the screen. Also reproducible by holding the UP arrow key. 

**Expected behavior**
No artifacts

**Screenshots**
[image]
[image]

**Desktop (please complete the following information if applicable):**
- Fedora 37, Wayland, OpenGL
- Advanced Micro Devices, Inc. [AMD®/ATI] Cezanne
- Advanced Micro Devices, Inc. [AMD®/ATI] Navi 22 [Radeon RX 6700/6700 XT/6750 XT / 6800M]

bdbcat, 2022-11-27:
Try github master now, please.

nohal, 2022-11-27:
Subjectively it is a bit better, but still not completely resolved. I can reproduce it on raster charts and wherever the basemap is on screen.
[image]
[image]

bdbcat, 2022-11-27:
OK, it was a bit of an experiment, anyway.  Don't understand it yet....

bdbcat, 2022-11-28:
Another try in master now.
Root cause of trouble: OpenGL does not like odd numbers for frame buffer height.  Who knows why...

nohal, 2022-11-28:
Yes, does the trick.