OpenCPN/OpenCPN #2828: Charts misplaced on Retina display with OpenGL

nohal, 2022-11-24:
**Describe the bug**
With OpenGL enabled, the Charts are misplaced on screen and do not agree at all with the lat/lon of the geographic features.

**To Reproduce**
On a Retina Mac, use OpenGL.

**Expected behavior**
Charts rendered in accordance with the geographic coordinates

**Screenshots**
OpenGL off, display OK
[image]
OpenGL on, chart shifted to wrong location
[image]

**Desktop (please complete the following information if applicable):**
- macOS 12.6.1
- Intel Retina MacBook Pro

bdbcat, 2022-12-07:
Pavel...
Sorry, cannot reproduce.
MacBook Intel/Retina  MacOS 11.6.1
Looks like the chart objects are a bunch of waypoints?  How about some actual chart rendering?  Is this related to GSHHS, for instance.
More color, please.

nohal, 2022-12-07:
Not related to GSHHG, all charts are wrong (In case of ENC not completely, texts are correct), everything else is correct, including the chart outlines.
The system is my usual 13 inch early-2015 macbook, just running a newer OS version.

This screenshot is taken from the Alpha3 bundle you posted after a `mv ~/Library/Preferences/opencpn ~/Library/Preferences/opencpn.bak`, so the environment I started with was as pristine as I can probably get it:

[image]

nohal, 2022-12-07:
The only place where the chart image seems to agree with everything else is the lower left corner of the screen.

bdbcat, 2022-12-07:
I see.  I was about to comment that there is a scale problem, as well as offset.  Evident in the prior pix, looking at outlines of NOAA IENC.  Anyway, not clearly reproducible here, so needs some thought.
[image]

bdbcat, 2022-12-07:
The fact that the correct rendering is in the lower left quarter of the screen makes me think of Retina 2x scaling....

nohal, 2022-12-07:
Yes, the DIP->physical pixel ratio handled differently between chart image production and everything else is the prime suspect, but I know pretty much nothing about the code involved.

bdbcat, 2022-12-07:
Will you verify glChartCanvas:434

`  m_displayScale = GetContentScaleFactor();
`
Should be 2.0.

nohal, 2022-12-07:
Yes, it is 2.

bdbcat, 2022-12-07:
OK, thanks, I'll sleep on it.

bdbcat, 2022-12-07:
Pavel...
Random experiment:
glChartCanvas.cpp:3925
Add this line:
    b_newview = true;

nohal, 2022-12-07:
No visible difference.

bdbcat, 2022-12-07:
I guess I'll try to load MacOS 12.6.  We should then have identical configs.

nohal, 2022-12-08:
Not entirely impossible, but I have to prepare it a bit.

bdbcat, 2022-12-08:
If it is not a true PIA, then that would be helpful.

bdbcat, 2022-12-12:
Some progress here.  wx changed the way they handle Retina display.
You may try github branch "monterey" to have a look at current state.  There are issues, but going in the right direction.

bdbcat, 2022-12-14:
More progress.  Getting pretty good, about ready to merge.
Problem exists on BigSur, as well, so not confined to Monterey.  Must be wx32 change.
The new solution is actually better than the previous method of post-scaling.  Now we get full pixel resolution of Retina display, instead of pixel post scaling by 2X.  The cost is slightly reduced performance.

bdbcat, 2022-12-14:
Merged to master now.

nohal, 2022-12-14:
Looks good. The full-res rendering is much better. We will probably have to adjust the default sizes of ENC texts and ships. I'm not completely blind yet, but had to increase both on the 13 inch screen to be comfortable. Anyway, that's the last cosmetic detail I can see.

closed 2023-01-31 by nohal