OpenCPN/OpenCPN #2830: Route line blured with OpenGL

nohal, 2022-11-24:
**Describe the bug**
The route line is blured

**To Reproduce**
Set route line width to eg. 10px

**Expected behavior**
The line should be solid, perhaps drawn with antialiasing, but not completely blured.

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
- Fedora 37, Wayland, OpenGL
- Advanced Micro Devices, Inc. [AMD®/ATI] Cezanne
- Advanced Micro Devices, Inc. [AMD®/ATI] Navi 22 [Radeon RX 6700/6700 XT/6750 XT / 6800M]

nohal, 2022-11-24:
Disabling https://github.com/OpenCPN/OpenCPN/blob/master/src/ocpndc.cpp#L594 leads to the following results:
[image]
[image]

bdbcat, 2022-11-25:
Query:
in ocpndc.cpp:602
```
        if (pen_width > parms[1])
          b_draw_thick = true;
        else
          glLineWidth(pen_width);

```
What is the result of this test on your graphics card/driver?  Is "b_draw_thick" true?  For wide lines only, or always, or never?

nohal, 2022-11-25:
Checked that yesterday, always false, the range is 0-2048 on both the cards I have.

bdbcat, 2022-11-25:
OK, thanks.

bdbcat, 2022-11-25:
Improved in master now.
What do you think?

nohal, 2022-11-25:
Better, still needs some more:

## Line smoothing disabled in OpenGL options

### Width 10px

[image]

### Width 1px

[image]

## Line smoothing enabled in OpenGL options

### Width 10px

[image]

### Width 2px

[image]

### Width 1px

Line completely invisible

nohal, 2022-11-25:
The screen is a 15.4" FullHD and the physical size is incorrectly detected as 508 mm, but setting it to the correct 340 makes no apparent difference.

bdbcat, 2022-11-26:
One the wider lines, I guess we may want to revert  to the O562 style, definitely not antialiased.  Personally, I find those jagged-edge line very disturbing on near-vertical lines. But that maybe just me...
I am sure not getting the one-pixel line problem here.
More to come.

bdbcat, 2022-11-28:
I reverted line drawing of routes to simpler code.  Just to verify that I am looking at the right code.  Should be same as O562 now.
Is there any sign of anti-aliasing on this hardware with O562?

nohal, 2022-11-28:
### Line smoothing *disabled* in OpenGL options: 
[image]

### Line smoothing *enabled* in OpenGL options:
[image]
Not antialiased, but transparent

bdbcat, 2022-11-28:
Those pix on master, or O562?

nohal, 2022-11-28:
With 5.6.2 OpenGL options make no difference, both cases look like
[image]

Hakansv, 2022-11-28:
Just for info when this "storm" is settled and possible connected? COG and heading predictor lines have become filled lines instead of long-short dashed lines. Both on Win and Rpi

bdbcat, 2022-11-28:
Ooops, my bad.
Fixed in master.  Not related, just some leftover debug stuff.

Hakansv, 2022-11-28:
👍

closed 2023-02-02 by nohal