OpenCPN/OpenCPN #2833: Own ship semi transparent

nohal, 2022-11-25:
**Describe the bug**
Own ship icon and COG/HDG predictors are drawn transparent

**To Reproduce**
Connect data source

**Expected behavior**
Should be drawn solid

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
- Fedora 37, Wayland, OpenGL
- Advanced Micro Devices, Inc. [AMD®/ATI] Cezanne
- Advanced Micro Devices, Inc. [AMD®/ATI] Navi 22 [Radeon RX 6700/6700 XT/6750 XT / 6800M]

bdbcat, 2022-11-26:
Completely guessing here.  Never seen this effect on any other system.
In glChartCanvas.cpp:2431
Add this:
 ```
     glEnable(GL_BLEND);
      glBlendFunc(GL_SRC_ALPHA, GL_ONE_MINUS_SRC_ALPHA);

```
-d

bdbcat, 2022-11-26:
Also, please compare to ENC object, like a buoy.  These are drawn with substantially the same OpenGL code.

nohal, 2022-11-26:
ENC objects are OK regardless of the settings

Adding does not seem to have any visually noticeable effect
[image]

Disabling smoothing for both lines and polygons in OpenGL options gets rid of the transparency, but there is no antialiasing
[image]

closed 2023-02-02 by nohal