OpenCPN/OpenCPN #2837: persistently crash on _cairo_arc_in_direction: Assertion `angle_max >= angle_min' failed

dhuyvetter, 2022-11-30:
After downloading a bunch of charts OpenCPN keeps crashing. This happens when I am using my mouse to move trough charts and persists even when running in safe mode without any plugins. I tried running in command line with `-unit_test_1:0` and `-unit_test_2` and when I start from command line `opencpn -l=error` in all these cases I get this error:
```
opencpn: ../../../../src/cairo-arc.c:189: _cairo_arc_in_direction: Assertion `angle_max >= angle_min' failed.
Aborted
```
I am running OpenCPN 5.6.2 on a Raspberry Pi 4 Model B Rev 1.5 running Raspbian GNU/Linux 11 (bullseye).

Since I do not know which chart is causing the error and it is crashing even when plugins are deactivated my OpenCPN has become completely unusable.

leamas, 2022-11-30:
Solving this is definitely above my paygrade. That said, could you please post also opencpn.log after starting opencpn with _--loglevel debug_ instead of _-l=error_

nohal, 2022-12-14:
@dhuyvetter at least please tell us what charts are you using when the problem happens, I don't see this assert with anything I have on my machine.

nohal, 2023-02-27:
@dhuyvetter Still have never seen this issue happening, either we need some aditional info to reproduce or we can't do anything about it...

nohal, 2023-04-15:
No reaction and no reports of similar issues from anybody else, closing.

closed 2023-04-15 by nohal