OpenCPN/OpenCPN #2845: Plugin Manager icons too large

rgleason, 2022-12-04:
OpenCPN 5.7.1+f6a1838
Small Plugin Icon problem. Icons to large in the Plugin Manager
1. Go to  Options > Plugins

**Expected behavior**
Smaller Icons

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
 - OS: [Windows
 - Version OpenCPN 5.7.1+f6a1838

**Additional context**
Tried adjusting the various User Interface settings, but none of the those are intended to control these icons.
Perhaps these Icon sizes should be attached to one of the user interface controls, such as User Interface Control Factor?

bdbcat, 2022-12-04:
Details:
Are you using display scaling > 100% ?

bdbcat, 2022-12-04:
Also, check settings->Advanced->Physical Screen width.  If set to "Auto", does the number match the screen size?

rgleason, 2022-12-05:
OpenCPN   Options > Display > Advanced    Physical Screen Width   "Auto"

Windows 11   Setttings > System > Display
Scale   250% (Recommended)
Display 3840x4200 (Recommended)

I had also just turned on  Options Display > Use Acclerated Graphics (OpenGL) before I shut down.
Now having restarted OpenCPN,

No real change in the chart text legibility (still problematic) or the Options > Plugins > Plugin Manager Icons.

[image]
[image]
[image]
[image]

rgleason, 2022-12-08:
Earlier this morning I thought I had updated my local repository, but I had not. Around 9am something didn't seem right and I acually git fetched successfully and updated my local repository. With this version running in debug mode the hamburger rollover boxes are working well, the route rollover box is also working well, and the Option > PI icons are better (the checkbox and other indicators are still a little large in my opinion, but it is better).

**Note: For High Resolution Screens 3840x 2400 with scale set to 250% in Windows10/11  Options > UI > Fonts >  ChartText and RouteLegRollover  have to be set to font size 18-20.**

Dave, should I close this now?

bdbcat, 2022-12-08:
Leave it open, for the (the checkbox and other indicators are still a little large in my opinion, but it is better).

rgleason, 2022-12-11:
I believe this is fixed now. Closing.  Will make new issue for another UI Plugin Manager small bug.
[image]

closed 2022-12-11 by rgleason