OpenCPN/OpenCPN #2846: Hoover over Hamburger  - Popup area is a little large (2 lines?)

rgleason, 2022-12-04:
**Describe the bug**
Hoover over Main Hamburger Icons - The popup identifier is a little large, perhaps two lines?

**Expected behavior**
Background Box for text should be better defined.

**Screenshots**

[image]

**Desktop (please complete the following information if applicable):**
 - OS: Windows
 - Version 7.7.1 +f6a1838

rgleason, 2022-12-08:
Still this way in master, but Dave has fixed it.

bdbcat, 2022-12-08:
Yes, fixed in master.  NOT fixed in Alpha-3

rgleason, 2022-12-08:
Earlier this morning I thought I had updated my local repository, but I had not. Around 9am something didn't seem right and I acually git fetched successfully and updated my local repository. With this version running in debug mode the hamburger rollover boxes are working well, the route rollover box is also working well, and the Option > PI icons are better (the checkbox and other indicators are still a little large in my opinion, but it is better).  

Should I close this now?

closed 2022-12-08 by rgleason