OpenCPN/OpenCPN #2849: Charts: Zoom out jittery

rgleason, 2022-12-04:
**Describe the bug**
Minor bug: When zooming out, with mouse wheel, once the full zoom out is reached, and when the user then tries further zoom out, the charts go into "jittery" mode until they settle back to the full zoom out.

I don't know if there is a good stable solution for this but it would be nice.

**To Reproduce**
As described.

**Expected behavior**
If zoom out has been reached and the user tries to zoom out again, just leave the charts stable.

**Screenshots**
I don't know if a screen shot does this effect justice, it really should be a video, but here it is.
[image]

**Desktop (please complete the following information if applicable):**
 - OS: Windows
 - Version 5.7.1+f6a1838

rgleason, 2022-12-04:
Further Comment after more use.
Sometimes zooming out settles down, and works as expected and is more stable.
But then if I try to left click drag the charts sideways or up or down when zoomed out, the jittery effect starts again.  And when I just try zooming out with the mouse wheel, it will start getting jittery. 
I haven't figured out quite what causes this.

bc38, 2022-12-04:
the sectors of the lighthouses are clearly twice as thick as the standard!

rgleason, 2022-12-04:
bc38  "the sectors of the lighthouses are clearly twice as thick as the standard!"

I don't know why that would occur.

bdbcat, 2022-12-05:
Rick...
I cannot reproduce.  Any possibility of a movie?

rgleason, 2022-12-06:
This is minor compared to the other issues.

rgleason, 2022-12-06:
I have some videos but I don't think they show much.

closed 2022-12-06 by rgleason