OpenCPN/OpenCPN #2850: Charts:  USCG Region 01 Chart Text is too small.

rgleason, 2022-12-04:
**Describe the bug**
Chart text is too small and font size does not appear to be controled by anything I know of.
Should we have to modify the  Fonts dropdown list?  If so, what selection has to be modified to increase the size.
Adjusting User Interface scale or Chart Object Scale does not affect small chart text.

**To Reproduce**
Steps to reproduce the behavior:
Just use a USCG 01 Enc chart.  This is also a problem in CMP93 charts. 
Small text is unreadable.

**Expected behavior**
A clear and concise size of text that can be read.

**Screenshots**
Here is a  USCG 01 ENC example, CMP93 has same problem.
[image]

**Desktop (please complete the following information if applicable):**
 - OS: Windows 11
 - Version [Opencpn 5.7.1 + f6a1838

bc38, 2022-12-04:
What is the screen size and resolution?

transmitterdan, 2022-12-04:
Look in the log file for the reported screen size? Does it match the actual screen size?

If you have to make big changes to the text size sliders it may be due to a wrongly identified screen size.

PS: This could be the cause of some of the other reports you made today.

rgleason, 2022-12-05:
OpenCPN Options > Display > Advanced Physical Screen Width "Auto"

Windows 11 Setttings > System > Display
Scale 250% (Recommended)
Display 3840x4200 (Recommended)

I had also just turned on Options Display > Use Acclerated Graphics (OpenGL) before I shut down.
Now having restarted OpenCPN,

No real change in the chart text legibility (still problematic) or the Options > Plugins > Plugin Manager Icons.

rgleason, 2022-12-05:
Here is a screenshot of text in 5.6.2
[image]

bdbcat, 2022-12-05:
OpenCPN Options > Display > Advanced Physical Screen Width "Auto"
What is the number reported there? (mm)

rgleason, 2022-12-05:
In Opencpn 5.7.1 
What I see is  Manual  366mm

After a number of crashes, I renamed my old opencpn.ini to  .bkup and restarted. 
It was still crashing on startup, then I got the charts back onto only Gshhg and it is more stable.
I have created new Chart paths to Cm93 and USCG Region 01 ENC  and have initiated a Scan Charts and Update Database, but it appears that OpenCPN is crashing due to chart problems. This is with OpenGL on.
I've gone back to safe mode now, with no OpenGL and am Rebuilding the entire database.

Are SENC files different between v5.6.2 and v5.7.1?
Will I be able to use v5.6.2 again without rescanning the entire database again?

rgleason, 2022-12-05:
17'" diagonal × 25.4 = 431.8 millimeters, but I am using "Auto" so the manual setting should not affect it.
 I wonder where the 366 came from?  

 In OpenCPN log I see that this dimension is "horizontal"  so 14.3" x 25.4 = 363 which is about 366.  Have set this to 366 manually again.

Still getting immediate crashes, removed opencpn.ini again and it has started working,  but I think Cm93 is the issue.

rgleason, 2022-12-05:
No, that is not the problem. 
It appears that I cannot define a unique separate path to Charts.  This setup fails.

```
[Settings/ChartDnldr]
ChartSources=All GSHHG basemaps|https://raw.githubusercontent.com/chartcatalogs/catalogs/master/GSHHG_Catalog.xml|C:\\Data-Hesper\\Chart\\GSHHG|
Source=-1
BaseChartDir=C:\\Data-Hesper\\Chart
PreselectNew=1
PreselectUpdated=1
AllowBulkUpdate=0
```

```
[Directories]
S57DataLocation=
InitChartDir=C:\\Data-Hesper\\Chart\\
GPXIODir=
TCDataDir=
BasemapDir=C:\\Data-Hesper\\Chart\\GSHHG\\
pluginInstallDir=
```

```
[ChartDirectories]
ChartDir1=C:\Data-Hesper\Chart\GSHHG
ChartDir2=C:\Data-Hesper\Chart\CM93ed2_2010May31
```
[opencpn - Copy.ini.txt](https://github.com/OpenCPN/OpenCPN/files/10158661/opencpn.-.Copy.ini.txt)

[opencpn - Copy.log.txt](https://github.com/OpenCPN/OpenCPN/files/10158668/opencpn.-.Copy.log.txt)

rgleason, 2022-12-05:
`Chart Text Size`
Increasing  Options > User Interface   Fonts Dropdown selected Charttext and found Font Size "18" worked much better than "9"   probably due to the high resolution of the screen.

[image]

bdbcat, 2022-12-06:
I am a little lost here.  Nothing about chert paths was changed in O57.  You should not (ever) need to manually edit the config file.  Charts may be anywhere you want them, as long as they are not "managed" by chart downloader.
Please try your problem description again, **in a new issue.**
We may close this issue now.  Chart text size is manually adjustable as you have described.  Probably only necessary for extremely down-scaled Windows displays.

rgleason, 2022-12-06:
Additionally, I had to increase the Options > User Interface Chart Object Size > quite significantly from 0 to something like 3 or 4  in order to see Depth Text.  

After doing that and zooming out the  Hour glass cursor has been showing for 10 minutes. 
There are some new bugs now.

closed 2022-12-06 by rgleason