OpenCPN/OpenCPN #2851: Memory leak in latest git master

transmitterdan, 2022-12-04:
**Describe the bug**
Heap usage increases without bounds until O crashes after several days.

**To Reproduce**
Steps to reproduce the behavior:
1. Start OpenCPN and wait a few days.
2. O will consume all memory and crash.

**Expected behavior**
Memory usage should be stable.

**Desktop (please complete the following information if applicable):**
 - OS: Windows
 - Version 11

I am looking for this bug but so far cannot find.  I do not believe it is in the AIS data streaming system as I have looked thoroughly there but maybe I missed something.  Whatever it is, building a debug version with memory checking does not reveal the offending source line.  The memory is probably allocated by a library built without memory debug checking and not OpenCPN source code else it would be reported in the memory dump.

bdbcat, 2022-12-04:
Probably going to need to git bisect to find this, assuming it is a recently introduced bug.

transmitterdan, 2022-12-04:
No, it's been around for a while.  Bisect will be hard given all the changes to the build system.  I am still looking for it.  I may have a few hints today.

bdbcat, 2022-12-04:
I have a Win10 running now, watching the memory load.  We'll see.

transmitterdan, 2022-12-04:
Hint: It happens when ENC charts (I'm using NOAA) are displayed.  Using RNC does not grow memory.

nohal, 2022-12-04:
Try with https://github.com/OpenCPN/OpenCPN/pull/2807/files

transmitterdan, 2022-12-04:
Testing...

transmitterdan, 2022-12-04:
> Try with https://github.com/OpenCPN/OpenCPN/pull/2807/files

This is a good fix for the leak.  Running for over an hour with no heap growth.

bdbcat, 2022-12-04:
Ooops.  My bad.
PR merged

bdbcat, 2022-12-06:
Close?

transmitterdan, 2022-12-06:
Tested fix and no change in memory usage over 4 hour span.

closed 2022-12-06 by transmitterdan