OpenCPN/OpenCPN #2854: Unable to change Waypoint Text Colour (Color) to Black

leighss, 2022-12-05:
**Describe the bug**
I am unable to change the waypoint text colour (color - for future searches) from white to black

**To Reproduce**
1. Install OpenCPN 5.6.2 from AUR https://aur.archlinux.org/packages/opencpn
2. Open OCPN
3. OpenCPN displays waypoint names in white
4. All other text is black
5. Ensure text colour in Options>User Interface under General Options: Fonts is black (#FFFFFF)
6. Use lxappearance to change from my usual dark to light widgets and icons
7. Waypoint text is still in white, despite above

**Expected behavior**
waypoint test in black to make them readable on light chart background

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
Linux archlabs 6.0.11-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 02 Dec 2022 17:25:31 +0000 x86_64 GNU/Linux

Thanks for excellent work!

Hakansv, 2022-12-11:
On Win.
Similar is valid for own track. It's not possible to change from default yellow to anything else.
Also not able to change a present track.

Hakansv, 2022-12-11:
The above comment about changing track color was false. Or better; based on a understandable user confusion.
When the track color was changed it was not visible since the, still yellow, track highlight was covering the track line and thus make it invisible. Zooming close showed also the color changed track line.
We may consider change the text for the highlight color option from "Track color" to "Highlight color". 
See pict.
[image]

nohal, 2023-02-04:
@leighss Can you retest with current beta versions, if not available from AUR, flatpak should work just fine on Arch.

leamas, 2023-02-04:
@leighss : To run the flatpak beta version:

    $ flatpak remote-add --if-not-exists flathub-beta https://flathub.org/beta-repo/flathub-beta.flatpakrepo
    $ flatpak install flathub-beta org.opencpn.OpenCPN
    $ flatpak run org.opencpn.OpenCPN//beta

leighss, 2023-02-04:
Will do. But only can do next week

rgleason, 2023-02-14:
@leighss   ?? Can you close this now, if it works?

leighss, 2023-02-14:
Hi. Sorry for delay. Laptop died, Setting up new laptop, and as a non flatpak user I had to work out how to get access to my charts with:
`flatpak override --filesystem=/run/media/Documents org.opencpn.OpenCPN `
It looks like it works!
MUCH easier to read now :smiley: 
Thanks!
[image]
[image]
[image]