OpenCPN/OpenCPN #2855: Charts & OpenGL: O fails if OpenGL is not on.

rgleason, 2022-12-05:
**Describe the bug**
When OpenGL is turned off, OpenCPN fails immediately;.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Options > Display and turn off opengl.
2. Try to load USCG 01 ENC charts by right click on sea and hover over Charts, to select your chart group for 01ENC
4. OpenCPN crashes.

**Expected behavior**
Show the chart properly.

**Screenshots**
I've uploaded some of these earlier, not much to see.

**Desktop (please complete the following information if applicable):**
 - OS: Windows 11
 - Version  5.7.1 + f6a1838

**Additional context**
When I turn on Display > Advanced > OpenGL   (and I cleared the texture cache) 

- All charts
- Gshhg
- ENC USCG 01 Region 

Chart Groups all seem to work.

bdbcat, 2022-12-06:
Not reported elsewhere, works fine for me and Hakan. Probably having trouble with super hi-def display on your device.
What is pixel size of your laptop display, again?

rgleason, 2022-12-06:
This Alpha version of Opencpn is not very successful on my machine and is unstable and fails often in a number of ways.. I believe that others will find the same thing.  I think I will wait for others to try it out.

My resolution is https://github.com/OpenCPN/OpenCPN/issues/2850#issuecomment-1336569728

Now I cannot seem to activate OpenGL and the hourglass does not go away.
There are too many issues to determine what the problem is.

Hakansv, 2022-12-06:
Dave..
The scenario pure S57 and OpenGl = off I never tried.
So I loaded a ENC (DK) instead of the oeusenc (o-charts) and turned OpenGL off.
While zoom/drag the ENC chart it crash somewhat similar to our o-charts trouble before.
Debug stops without a good trace but sometimes this:
```
wxmsw32u_core_vc_custom.dll!65b02690()	Unknown
 	[Frames below may be incorrect and/or missing, no symbols loaded for wxmsw32u_core_vc_custom.dll]	
 	[External Code]	
>	opencpn.exe!WinMain(HINSTANCE__ * hInstance, HINSTANCE__ * hPrevInstance, char * lpCmdLine, int nCmdShow) Line 953	C++
```

With OpenGl = ON I couldn't reproduce.
And when I again switched OpenGl = OFF a crash with this call stack occurred:

```
 	wxmsw32u_core_vc_custom.dll!65b02690()	Unknown
 	[Frames below may be incorrect and/or missing, no symbols loaded for wxmsw32u_core_vc_custom.dll]	
 	[External Code]	
>	opencpn.exe!MyFrame::DoOptionsDialog() Line 4470	C++
 	opencpn.exe!MyFrame::DoSettings() Line 3281	C++
 	opencpn.exe!MyFrame::OnToolLeftClick(wxCommandEvent & event) Line 3145	C++
 	[External Code]	
 	opencpn.exe!WinMain(HINSTANCE__ * hInstance, HINSTANCE__ * hPrevInstance, char * lpCmdLine, int nCmdShow) Line 953	C++
```

rgleason, 2022-12-06:
This version seems to be behaving better today, but the later version with memory leaks fixed seems to run nicely under debug.

However I am finding that if I turn off OpenGL in the debugging version, I can't turn OpenGL back on, but that is not this bug.

rgleason, 2022-12-06:
Sorry Hakan has a report.

bdbcat, 2022-12-11:
Close?

Hakansv, 2022-12-11:
Yes close. The OG-on/off is solved. Also what initially seemed connected but was a distant route problem is also solved by 06ec3b7d5.

closed 2022-12-11 by rgleason