OpenCPN/OpenCPN #2860: Charts: USCG Region 01 Deer Isle area missing charts

rgleason, 2022-12-08:
**Describe the bug**
A number of charts are missing in the Deer Isle area. Is this normal?
Is this a NOAA Chart problem or related to OpenCPN?

**To Reproduce**
Using USCG Region 01 ENC charts.
Zoom into the Deer Island ME area.:

**Expected behavior**
More complete mapping.
I expected the same with USCG Region 04 ENC.

**Screenshots**

[image]

**Desktop (please complete the following information if applicable):**
 - OS: Windows 11
 - Version OpenCPN 5..7.1  recent master build

rgleason, 2022-12-08:
This is turning into a puzzle. I have two different directories for USCG Region 01 which have both been updated by Chartdownloader.  Both chart directories have Maine  "Fox Island Thoroughfare" chart which is updated.

1. When I Right click > Charts > E01 which has a path of C:\Users\fcgle\OneDrive\Documents\Charts\ENC\US_CGD01  automatically created by Chartdownloader, I get more detail in Vinalhaven around Penobscott Island.
[image]

[image]

[image]
[image]

2.  When I right click > Charts > CG01 which has a path of C:\Data-Hesper\Chart\USenc\CG-01 which is also updated with 631 charts, I get much less detail and the "Fox Island Thoroughfare" chart is not shown but does exist with the correct updated date.

[image]

[image]

[image]

rgleason, 2022-12-08:
The same thing is happening when I run the updated master branch (shows as 5.7.1 in the program)  in debug mode.

Earlier this morning I thought I had updated my local repository, but I had not. Around 9am something didn't seem right and I acually git fetched successfully and updated my local repository.  With this version running in debug mode the hamburger rollover boxes are working well, the route rollover box is also working well, and the Option > PI  icons are better  (the checkbox and other indicators are still a little large in my opinion, but it is better).

rgleason, 2022-12-08:
The same thing is happening when I run Opencpn 5.6.2  the

Right click > Charts > E01 which has a path of C:\Users\fcgle\OneDrive\Documents\Charts\ENC\US_CGD01  charts show more detail around Vinalhaven  Penobscot Island than

right click > Charts > CG01 which has a path of C:\Data-Hesper\Chart\USenc\CG-01 which is also updated with 631 charts, I get much less detail and the "Fox Island Thoroughfare" chart is not shown but does exist with the correct updated date.

rgleason, 2022-12-11:
Looking at the big picture  (using the same set of charts downloaded and updated by Chartdownloader) but located in different directories  (as indicated above)   the  E01 selection of charts appears very different from the CGe01 selection.

[image]
[image]

nohal, 2023-02-04:
Not reproducible here. Please try to retest and if possible create a scenario that is usable to reliably reproduce.

closed 2023-02-04 by rgleason

rgleason, 2023-02-04:
The issues seem to be better now.