OpenCPN/OpenCPN #2861: Breaks: Are these bugs or not?

rgleason, 2022-12-08:
**Describe the bug**
Using recent Opencpn 5.8.1 with updated master branch, built.

Steps to reproduce the behavior:
Pan and Scroll with USCG Region 01 ENC charts
Several breaks after tried to continue.

**Expected behavior**
I thought hitting debug continue would skip and allow program to continue.

**Screenshots**
[image]
[image]
[image]

**Desktop (please complete the following information if applicable):**
 - OS: Windows 11
 - Version O 5.7.1

Let me know if I should close this please.

bdbcat, 2022-12-08:
These are bugs.
If you can reproduce reliably, that would be very helpful.

rgleason, 2022-12-08:
Will try later.

closed 2023-02-04 by nohal