OpenCPN/OpenCPN #2868: UI: Options  Icons and Text (selection menu at the top) issues

rgleason, 2022-12-11:
**Describe the bug**
Options menu,  with icons and text at the top. There are size and consistency issues.
"Connections" is abbreviated to   "Connect..." sometimes.
"Connections" is put into two lines sometimes  Connecti   and then on.
"User Interface" is turned into 2 lines because of the space.
When "User Interface" is turned into 2 lines it messes up the vertical size of this header menu so that it become unstable and will scroll.
When  "Connection" is turned into 2 lines it messes up the vertical Size of the header menu.

**To Reproduce**
Steps to reproduce the behavior:
Just use the Option interface.

**Expected behavior**
This main header menu should be very stable and not jump around.

**Screenshots**
[image]
[image]
[image]

**Desktop (please complete the following information if applicable):**
 - OS: Win 11
 - Version  5.7.1 built from master

**Additional context**
While I ignored this all through 5.6.2,  I think this needs to be fixed so the interface appears to be more stable.

Suggestions:
Just  add  the   ".."  if necessary but don't split into two lines.
Perhaps for User Interface add User-Interface,  or just use UI  <--Won't everybody get it?
Check this with each of the Options selected and unselected.

rgleason, 2023-01-30:
The Options Menu Bar

    Vertical slider is unnecessary.
    "User Interface" causes the slider.
    Ideas: Abbreviate User Interface similar to "Conne..."
    Ideas: Abbreviate User Interface to "UI" or just "User" or just "Interface"

[image]

[image]

[image]

transmitterdan, 2023-01-31:
Rick,

I think this is going to be futile. If UI text size is set by the global display settings (as in 150% for hi-res displays) then wxWidgets does this to try to display all the text. If the global UI text size is reduced then this problem goes away. In other words, it will always do this for some users no matter how we program the tab sizes.

rgleason, 2023-01-31:
There are a number of alternatives to fix this and lengthening the vertical constraint proportionally in cases with High Resolution is just one of them.
The other alternatives are to Change "User Interface" in some way by shortening it or cutting it off. IE change to "U.I."   or "Interface" or  "User Inter..." with two dots just like Connections does "Connect..." 
This will then eliminate the second line of text and the toolbar should then be less jumpy and more stable.

I don't think this problem should be ignored as it is just a part of a DPI aware software and should not be dismissed in this way.

rgleason, 2023-01-31:
Fixed in recent master. User Interface is now "User"  
Menus are much more stable and there is no longer a scrollbar!
Good improvment. Thanks.

closed 2023-01-31 by rgleason