OpenCPN/OpenCPN #2869: OCPN crash while deleting a zero length track

Hakansv, 2022-12-12:
**Describe the bug**
Deleting a "undefined" track with zero length and no legs causes a O crash.
Code from today's master.
Two scenarios:
- Delete all tracks -> O does crash but can be restarted.
- Delete only the zero length track. -> O does crash but can't be restarted until "navobj.xml.changes" is emptied or deleted.

**To Reproduce**
Steps to reproduce the behavior:
1. Make sure no connection to active positions is activated.
2. Enable tracking. 
3. Open route and mark manager, tab "Tracks".
4. _First scenario:_ "Delete all"  -> "Yes" to delete all. -> Crash.
5. Restart OCPN and redo task 2 to 3.
6. _Second scenario:_ Select the new route with zero length. -> Delete -> "Yes" to delete.  -> Crash
7. Restart OCPN -> Crash
8. Delete or empty "navobj.xml.changes".
9. Restart OCPN.

**Expected behavior**
Even a zero length route should be able to delete or being included in "Delete all". (And thus stop tracking)

**Desktop (please complete the following information if applicable):**
 - Win 10
 - RPi400 -64 - Debian 11

**Additional context**
Call stack snippet from scenario one.
```
>	ntdll.dll!770b5ee1()	Unknown
 	[Frames below may be incorrect and/or missing, no symbols loaded for ntdll.dll]	
 	[External Code]	
 	opencpn.exe!wxJSONWriter::WriteSeparator(wxOutputStream & os) Line 1232	C++
 	opencpn.exe!wxJSONWriter::DoWrite(wxOutputStream & os, const wxJSONValue & value, const wxString * key, bool comma) Line 500	C++
 	opencpn.exe!wxJSONWriter::Write(const wxJSONValue & value, wxOutputStream & os) Line 301	C++
 	opencpn.exe!wxJSONWriter::Write(const wxJSONValue & value, wxString & str) Line 280	C++
 	opencpn.exe!PlugInManager::SendJSONMessageToAllPlugins(const wxString & message_id, wxJSONValue v) Line 2308	C++
 	opencpn.exe!MyFrame::TrackOff(bool do_add_point) Line 3487	C++
 	opencpn.exe!RoutemanGui::DeleteTrack(Track * pTrack) Line 249	C++
 	opencpn.exe!MyFrame::TrackOff(bool do_add_point) Line 3491	C++
 	opencpn.exe!RoutemanGui::DeleteTrack(Track * pTrack) Line 249	C++
```

bdbcat, 2022-12-12:
How does one generate a "zero length" track?

bdbcat, 2022-12-12:
OK, I see that.  I skipped while reading....
I'll take a look. Thanks.

Hakansv, 2022-12-18:
Solved by a few commits. Thanks

closed 2022-12-18 by Hakansv