OpenCPN/OpenCPN #2870: CI: macos build on 14.0 fails on final linkage

leamas, 2022-12-12:
**Describe the bug**
CI build on CircleCIO labeled Macos Latest fails

**To Reproduce**
Steps to reproduce the behavior:
1. Do any commit
2. Push
3. Check output on circleci dashboard: 

```
[ 83%] Linking CXX executable OpenCPN.app/Contents/MacOS/OpenCPN
/Applications/Xcode-14.1.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -O2 -stdlib=libc++ -arch x86_64 -g -O3 -stdlib=libc++ -arch x86_64 -isysroot /Applications/Xcode-14.1.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.0.sdk -mmacosx-version-min=10.10 -Wl,-search_paths_first -Wl,-headerpad_max_install_names -O2 -stdlib=libc++ -arch x86_64  -framework CoreGraphics CMakeFiles/OpenCPN.dir/src/about.cpp.o CMakeFiles/OpenCPN.dir/src/AboutFrame.cpp.o CMakeFiles/OpenCPN.dir/src/AboutFrameImpl.cpp.o CMakeFiles/OpenCPN.dir/src/ais.cpp.o CMakeFiles/OpenCPN.dir/src/AISTargetAlertDialog.cpp.o CMakeFiles/OpenCPN.dir/src/AISTargetListDialog.cpp.o CMakeFiles/OpenCPN.dir/src/AISTargetQueryDialog.cpp.o CMakeFiles/OpenCPN.dir/src/ais_info_gui.cpp.o CMakeFiles/OpenCPN.dir/src/CanvasConfig.cpp.o CMakeFiles/OpenCPN.dir/src/canvasMenu.cpp.o CMakeFiles/OpenCPN.dir/src/CanvasOptions.cpp.o CMakeFiles/OpenCPN.dir/src/catalog_mgr.cpp.o CMakeFiles/OpenCPN.dir/src/cat_settings.cpp.o CMakeFiles/OpenCPN.dir/src/certificates.cpp.o CMakeFiles/OpenCPN.dir/src/chartdb.cpp.o CMakeFiles/OpenCPN.dir/src/chartdbs.cpp.o CMakeFiles/OpenCPN.dir/src/chartimg.cpp.o CMakeFiles/OpenCPN.dir/src/chcanv.cpp.o CMakeFiles/OpenCPN.dir/src/ChInfoWin.cpp.o CMakeFiles/OpenCPN.dir/src/color_handler.cpp.o CMakeFiles/OpenCPN.dir/src/compass.cpp.o CMakeFiles/OpenCPN.dir/src/concanv.cpp.o CMakeFiles/OpenCPN.dir/src/ConfigMgr.cpp.o CMakeFiles/OpenCPN.dir/src/conn_params_panel.cpp.o CMakeFiles/OpenCPN.dir/src/connections_dialog.cpp.o CMakeFiles/OpenCPN.dir/src/DetailSlider.cpp.o CMakeFiles/OpenCPN.dir/src/download_mgr.cpp.o CMakeFiles/OpenCPN.dir/src/FlexHash.cpp.o CMakeFiles/OpenCPN.dir/src/FontDesc.cpp.o CMakeFiles/OpenCPN.dir/src/FontMgr.cpp.o CMakeFiles/OpenCPN.dir/src/GoToPositionDialog.cpp.o CMakeFiles/OpenCPN.dir/src/gshhs.cpp.o CMakeFiles/OpenCPN.dir/src/gui_lib.cpp.o CMakeFiles/OpenCPN.dir/src/IDX_entry.cpp.o CMakeFiles/OpenCPN.dir/src/iENCToolbar.cpp.o CMakeFiles/OpenCPN.dir/src/kml.cpp.o CMakeFiles/OpenCPN.dir/src/Layer.cpp.o CMakeFiles/OpenCPN.dir/src/LinkPropDlg.cpp.o CMakeFiles/OpenCPN.dir/src/logger.cpp.o CMakeFiles/OpenCPN.dir/src/MarkInfo.cpp.o CMakeFiles/OpenCPN.dir/src/mbtiles.cpp.o CMakeFiles/OpenCPN.dir/src/mDNS_query.cpp.o CMakeFiles/OpenCPN.dir/src/mDNS_service.cpp.o CMakeFiles/OpenCPN.dir/src/MUIBar.cpp.o CMakeFiles/OpenCPN.dir/src/multiplexer.cpp.o CMakeFiles/OpenCPN.dir/src/navutil.cpp.o CMakeFiles/OpenCPN.dir/src/NMEALogWindow.cpp.o CMakeFiles/OpenCPN.dir/src/ocpCursor.cpp.o CMakeFiles/OpenCPN.dir/src/ocpn_app.cpp.o CMakeFiles/OpenCPN.dir/src/OCPN_AUIManager.cpp.o CMakeFiles/OpenCPN.dir/src/ocpndc.cpp.o CMakeFiles/OpenCPN.dir/src/ocpn_frame.cpp.o CMakeFiles/OpenCPN.dir/src/OCPNListCtrl.cpp.o CMakeFiles/OpenCPN.dir/src/ocpn_pixel.cpp.o CMakeFiles/OpenCPN.dir/src/OCPNPlatform.cpp.o CMakeFiles/OpenCPN.dir/src/ocpn_plugin.cpp.o CMakeFiles/OpenCPN.dir/src/ocpn_print.cpp.o CMakeFiles/OpenCPN.dir/src/OCPNRegion.cpp.o CMakeFiles/OpenCPN.dir/src/options.cpp.o CMakeFiles/OpenCPN.dir/src/peer_client.cpp.o CMakeFiles/OpenCPN.dir/src/piano.cpp.o CMakeFiles/OpenCPN.dir/src/pluginmanager.cpp.o CMakeFiles/OpenCPN.dir/src/plugin_paths.cpp.o CMakeFiles/OpenCPN.dir/src/printtable.cpp.o CMakeFiles/OpenCPN.dir/src/priority_gui.cpp.o CMakeFiles/OpenCPN.dir/src/Quilt.cpp.o CMakeFiles/OpenCPN.dir/src/REST_server_gui.cpp.o CMakeFiles/OpenCPN.dir/src/RolloverWin.cpp.o CMakeFiles/OpenCPN.dir/src/routemanagerdialog.cpp.o CMakeFiles/OpenCPN.dir/src/routeman_gui.cpp.o CMakeFiles/OpenCPN.dir/src/route_gui.cpp.o CMakeFiles/OpenCPN.dir/src/route_point_gui.cpp.o CMakeFiles/OpenCPN.dir/src/routeprintout.cpp.o CMakeFiles/OpenCPN.dir/src/RoutePropDlg.cpp.o CMakeFiles/OpenCPN.dir/src/RoutePropDlgImpl.cpp.o CMakeFiles/OpenCPN.dir/src/S57QueryDialog.cpp.o CMakeFiles/OpenCPN.dir/src/safe_mode.cpp.o CMakeFiles/OpenCPN.dir/src/SendToGpsDlg.cpp.o CMakeFiles/OpenCPN.dir/src/SendToPeerDlg.cpp.o CMakeFiles/OpenCPN.dir/src/Station_Data.cpp.o CMakeFiles/OpenCPN.dir/src/styles.cpp.o CMakeFiles/OpenCPN.dir/src/svg_utils.cpp.o CMakeFiles/OpenCPN.dir/src/TCDataFactory.cpp.o CMakeFiles/OpenCPN.dir/src/TCDataSource.cpp.o CMakeFiles/OpenCPN.dir/src/TCDS_Ascii_Harmonic.cpp.o CMakeFiles/OpenCPN.dir/src/TCDS_Binary_Harmonic.cpp.o CMakeFiles/OpenCPN.dir/src/tcmgr.cpp.o CMakeFiles/OpenCPN.dir/src/TCWin.cpp.o CMakeFiles/OpenCPN.dir/src/thumbwin.cpp.o CMakeFiles/OpenCPN.dir/src/toolbar.cpp.o CMakeFiles/OpenCPN.dir/src/track_gui.cpp.o CMakeFiles/OpenCPN.dir/src/trackprintout.cpp.o CMakeFiles/OpenCPN.dir/src/TrackPropDlg.cpp.o CMakeFiles/OpenCPN.dir/src/TTYScroll.cpp.o CMakeFiles/OpenCPN.dir/src/TTYWindow.cpp.o CMakeFiles/OpenCPN.dir/src/undo.cpp.o CMakeFiles/OpenCPN.dir/src/update_mgr.cpp.o CMakeFiles/OpenCPN.dir/src/waypointman_gui.cpp.o CMakeFiles/OpenCPN.dir/src/viewport.cpp.o CMakeFiles/OpenCPN.dir/src/shaders.cpp.o CMakeFiles/OpenCPN.dir/src/ais_bitstring.cpp.o CMakeFiles/OpenCPN.dir/src/ais_decoder.cpp.o CMakeFiles/OpenCPN.dir/src/ais_target_data.cpp.o CMakeFiles/OpenCPN.dir/src/base_platform.cpp.o CMakeFiles/OpenCPN.dir/src/catalog_handler.cpp.o CMakeFiles/OpenCPN.dir/src/catalog_parser.cpp.o CMakeFiles/OpenCPN.dir/src/chartdata_input_stream.cpp.o CMakeFiles/OpenCPN.dir/src/comm_ais.cpp.o CMakeFiles/OpenCPN.dir/src/comm_navmsg.cpp.o CMakeFiles/OpenCPN.dir/src/comm_navmsg_bus.cpp.o CMakeFiles/OpenCPN.dir/src/comm_appmsg.cpp.o CMakeFiles/OpenCPN.dir/src/comm_appmsg_bus.cpp.o CMakeFiles/OpenCPN.dir/src/comm_bridge.cpp.o CMakeFiles/OpenCPN.dir/src/comm_decoder.cpp.o CMakeFiles/OpenCPN.dir/src/comm_drv_file.cpp.o CMakeFiles/OpenCPN.dir/src/comm_drv_factory.cpp.o CMakeFiles/OpenCPN.dir/src/comm_drv_n0183.cpp.o CMakeFiles/OpenCPN.dir/src/comm_drv_n0183_net.cpp.o CMakeFiles/OpenCPN.dir/src/comm_drv_n0183_serial.cpp.o CMakeFiles/OpenCPN.dir/src/comm_drv_n2k.cpp.o CMakeFiles/OpenCPN.dir/src/comm_drv_n2k_serial.cpp.o CMakeFiles/OpenCPN.dir/src/comm_drv_registry.cpp.o CMakeFiles/OpenCPN.dir/src/comm_drv_signalk.cpp.o CMakeFiles/OpenCPN.dir/src/comm_drv_signalk_net.cpp.o CMakeFiles/OpenCPN.dir/src/comm_n0183_output.cpp.o CMakeFiles/OpenCPN.dir/src/comm_util.cpp.o CMakeFiles/OpenCPN.dir/src/comm_plugin_api.cpp.o CMakeFiles/OpenCPN.dir/src/conn_params.cpp.o CMakeFiles/OpenCPN.dir/src/cutil.cpp.o CMakeFiles/OpenCPN.dir/src/downloader.cpp.o CMakeFiles/OpenCPN.dir/src/garmin_protocol_mgr.cpp.o CMakeFiles/OpenCPN.dir/src/geodesic.cpp.o CMakeFiles/OpenCPN.dir/src/georef.cpp.o CMakeFiles/OpenCPN.dir/src/hyperlink.cpp.o CMakeFiles/OpenCPN.dir/src/nav_object_database.cpp.o CMakeFiles/OpenCPN.dir/src/navutil_base.cpp.o CMakeFiles/OpenCPN.dir/src/ocpn_utils.cpp.o CMakeFiles/OpenCPN.dir/src/plugin_blacklist.cpp.o CMakeFiles/OpenCPN.dir/src/plugin_cache.cpp.o CMakeFiles/OpenCPN.dir/src/plugin_handler.cpp.o CMakeFiles/OpenCPN.dir/src/plugin_loader.cpp.o CMakeFiles/OpenCPN.dir/src/position_parser.cpp.o CMakeFiles/OpenCPN.dir/src/REST_server.cpp.o CMakeFiles/OpenCPN.dir/src/route.cpp.o CMakeFiles/OpenCPN.dir/src/route_point.cpp.o CMakeFiles/OpenCPN.dir/src/routeman.cpp.o CMakeFiles/OpenCPN.dir/src/select.cpp.o CMakeFiles/OpenCPN.dir/src/select_item.cpp.o CMakeFiles/OpenCPN.dir/src/semantic_vers.cpp.o CMakeFiles/OpenCPN.dir/src/ser_ports.cpp.o CMakeFiles/OpenCPN.dir/src/track.cpp.o CMakeFiles/OpenCPN.dir/src/garmin_wrapper.cpp.o CMakeFiles/OpenCPN.dir/src/macutils.c.o CMakeFiles/OpenCPN.dir/src/glTextureDescriptor.cpp.o CMakeFiles/OpenCPN.dir/src/glTexCache.cpp.o CMakeFiles/OpenCPN.dir/src/glChartCanvas.cpp.o CMakeFiles/OpenCPN.dir/src/glTextureManager.cpp.o -o OpenCPN.app/Contents/MacOS/OpenCPN   -L/tmp/wx321_opencpn50_macos1010/lib  -Wl,-rpath,/tmp/wx321_opencpn50_macos1010/lib libs/observable/lib_OBSERVABLE.a libs/ssl_sha1/libSHA1.a libs/mongoose/libMONGOOSE.a libs/mDNS/libMDNS.a libs/SQLiteCpp/libSQLITE_CPP.a /Applications/Xcode-14.1.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.0.sdk/usr/lib/libbz2.tbd /Applications/Xcode-14.1.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.0.sdk/usr/lib/libz.tbd libs/wxcurl/libWXCURL.a /Applications/Xcode-14.1.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.0.sdk/usr/lib/libcurl.tbd libs/serial/libSERIAL.a libs/libtess2/libTESS2.a libs/sound/ [... cut ...]
ld: in libs/garmin/libGARMINHOST.a(garmin_gps.h.o), archive member 'garmin_gps.h.o' with length 1752920 is not mach-o or llvm bitcode file 'libs/garmin/libGARMINHOST.a' for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```
Link: https://app.circleci.com/pipelines/github/leamas/OpenCPN/2528/workflows/5399911d-5608-4e13-857f-acf9105252d9/jobs/4679
**Expected behavior**
Working build

See also https://github.com/OpenCPN/OpenCPN/pull/2858 which I think is targeting the same issue.

@nohal^

nohal, 2022-12-12:
I have to check the build for Apple Silicon, that is where the complaint about jeeps headers appeared first already with some older compiler versions.

leamas, 2022-12-12:
I think I have a better patch  in #2873. In particular, see [this commit](https://github.com/OpenCPN/OpenCPN/pull/2873/commits/28c609190d6b)

PR #2873 (2022-12-13): Garmin 2870
This should close #2870, but we need to check the ci build for this PR first.

EDIT: 

Closes: #2870
closed 2022-12-13 by bdbcat