OpenCPN/OpenCPN #2877: RFE: Improve anchor alarm graphics

mgrouch, 2022-12-13:
When anchor alarm is set it would be nice to see on a zoomed chart where anchor is, where boat is, what is boat heading, 
and some representation of history of a boat movement around anchor. How many circles it did around anchor should be displayed.

Thanks

Hakansv, 2022-12-13:
You know how it works, do you?
Set a mark at anchor drop. Name it to the distance you need for the alert. On right click menu "Set anchor watch" (This menu item only when enough close to the ship)

Hakansv, 2022-12-13:
Could be but it's not a bug. It is working as intended since years. Suggestions how to improve details are of course welcome. Personally I think the function is handy and enough usable. "Not user friendly" is a very general and personal note not able to design from. 
There's an anchor alarm in plugin Watchdog if you like that better? Although is that plugin not yet available in Win 5.7.1

nohal, 2024-02-14:
I miss any actionable definition of improvements that are requested. Everything mentioned in the original description seems to be there, unless the requested improvements are specified, this issue will be closed.

rgleason, 2024-02-14:
See discussion and suggestions for WD anchor alarm
https://github.com/rgleason/watchdog_pi/issues/71

bdbcat, 2024-04-25:
No action taken.

closed 2024-04-25 by bdbcat