OpenCPN/OpenCPN #2880: Poor performance with many IENC cells

nohal, 2022-12-14:
**Describe the bug**
With many (~70-90) IENC cells in the quilt, we hit the roof on CPU and appear stuck (on the mac seemingly completely, on the much faster Ryzen machine still somewhat usable as the spike ends much sooner). The terminal window shots are `perf top` during the problem showing the hotspots in memory allocations. The problem is the cells kick in much too early.

**To Reproduce**
Default settings
German IENC cells
Zoom in and out to a cell until many cells are to be quilted.
Gets worse if the cells are not yet processed to SENC and the background threads are started for them.

**Expected behavior**
We should probably throw in some more protection against this in case the hotspots themselves can't be addressed - For example skipping a cell with display size < than XxX pixels.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information if applicable):**
 - macOS on i7 Intel
 - Linux on very fast Ryzen

bdbcat, 2022-12-14:
Thanks for the detail.  This is much on my mind.  NOAA is moving to a tiled ENC model, already present on CGRegion01.  Similar situation, evident in Long Island sound.  Also some o-charts ENC providers.
I have some ideas for quilt composition.
Also need to load it up, and do some valgrind profiling.
On my list.

bdbcat, 2022-12-15:
Pavel..
Cannot quite get your display on my linux box.
What is Settings->Display->Advanced->Zoom/Scale weight for ENC?
On zoom in from far away, at what reported chart scale (lower right hamburger) do the charts begin to be rendered?

bdbcat, 2022-12-15:
Another simple thing to look at.  We have an arbitrary limit on the number of chart cells cached in the core.  The default is 20, probably exceeded on your image.  Serious thrashing at small scale.

Can be set in config file:
Read(_T ( "NCacheLimit" ), &g_nCacheLimit);
Just a thought, for experimentation.

nohal, 2022-12-15:
Weighting set to 0 for all chart types.
It's not totally consistent, but a pretty good "trick" seems to be to use zoom to cursor, point the cursor at an IENC cell somewhere in the middle of Germany and use the mouse wheel to zoom in. Once some quilt is composed, usually around 1:2M, use the wheel to zoom out. At around 1:6M the problem kicks hard.
Other way often leading to this is at some 1:15-20M hold the left mouse button somewhere over Germany and start panning the view around.
Once the quilt with many charts kicks in, it is pretty hard to actually get rid of it.

nohal, 2022-12-15:
My usual routine:
[Screencast from 2022-12-15 00-49-31.webm](https://user-images.githubusercontent.com/249856/207768416-dda88648-dcbd-4b13-bb35-e010bf907550.webm)

nohal, 2022-12-15:
The situation improves with `NCacheLimit` set higher than the actual size of the quilt - at `100` there still is visible busy moment every time stuff is recalculated when there are 40+ cells in the quilt, but on this very fast machine it is pretty bearable.

nohal, 2022-12-15:
With `NCacheLimit` set to `1000` and charts already remembered, I would consider the performance completely acceptable even at the moment I managed to produce this with weighting set to 5:
[image]

bdbcat, 2022-12-15:
Wow...
There is still something odd here. Using your first method above:
" a pretty good "trick" seems to be to use zoom to cursor, point the cursor at an IENC cell somewhere in the middle of Germany and use the mouse wheel to zoom in. Once some quilt is composed, usually around 1:2M,..."

On my system, decent display, I do not see charts until around 1:500,000.   This of course removes a lot of pressure from nCharts and CPU. This is supposed to be controlled by some logic/math related to screen size and resolution.  Essentially the same logic that calculates the scale reported in hamburger box.   But it is not working the same on different platforms. I need to trace this a bit more before I propose anything.

bdbcat, 2022-12-15:
If you have the time, could you:
1.  Break on chcanv.cpp:2384.
2. We are interested in "m_canvas_scale_factor".
3. The value should be interpreted as  "pixels per meter" of your physical display device.
4. On my system, 20" monitor, 1920 pix horizontal, I get about 3780 pixels per meter.  Make sense.
5. How on yours?

Thanks

nohal, 2022-12-15:
The scale at which the quilt gets composed is not consistent, at around 1:500k it works pretty much anywhere, at 1:1M+ it happens only on some places, usually junctions of rivers with larger contiguous area covered by charts.
The problematic moment is actually the second step, when at zooming out more and more cells are added.
The last shot is pretty random, I'm not able to reproduce it consistently - Simply sometimes while panning around it happens.
Too late here, wil get back to it in the morning.

bdbcat, 2022-12-15:
OK, yes.  Not consistent.  Some logic corner case, I think.

bcn2, 2022-12-15:
I would suggest to run the test with the Finnish ENCs/o-charts set. That one is known as the most challenging one. Triggers as well a huge number of cells with a lot of details and objects. Critical on low performance systems.

nohal, 2022-12-15:
> * On my system, 20" monitor, 1920 pix horizontal, I get about 3780 pixels per meter.  Make sense.
> * How on yours?

5580, seems OK for a 15.4" Full HD (344mm width/1920=~0.18mm/px)

nohal, 2022-12-17:
Have just the Mac with me now. Got actually much worse, already at 1:16M+ the charts kick in, the app stalls and gets almost unusable.
[image]

bdbcat, 2022-12-17:
Hmm..
Did not test on Retina.

bdbcat, 2022-12-17:
With a slight tweak, here is what I see:

https://user-images.githubusercontent.com/1015682/208225785-c5f03fa7-77c1-4aaa-aa70-d6a7017dddee.mp4

bdbcat, 2022-12-17:
OK, try this:
https://www.dropbox.com/s/727nvtzcbzpqllb/Screencast%202022-12-16%2023%3A50%3A36.mp4?dl=0

nohal, 2022-12-17:
Still no joy, the Linux machine behaves pretty much the same, just is many times faster so even with this behavior it is perfectly usable
[Screencast from 2022-12-17 07-14-22.webm](https://user-images.githubusercontent.com/249856/208237049-3b015fb5-5fcd-422a-95e4-11e136e6057b.webm)

nohal, 2022-12-17:
BTW the problem in all it's gravity seems to be caused just by the southern region visible in https://github.com/OpenCPN/OpenCPN/issues/2880#issuecomment-1355993627 - as soon as I zoom to anything else up north, the behavior is similar to your video.

bdbcat, 2022-12-17:
Will you post the list of Inland charts you have loaded?

bcn2, 2022-12-17:
Dave,

what I see from the screenshot this is German InlandENCs, the complete portfolio.

nohal, 2022-12-17:
On the Linux machine only German IENCs, freshly obtained using the chart downloader today. On the mac some older release of the same charts and also the CZ set, but it does not seem to play any role.

bdbcat, 2022-12-18:
Investigated. Result:
Many of the cells in DE_Inland set seem to be coded with a non-sensible scale value. Most of the cells have scale of 1:2000, which seems reasonable.  However, some cells, concentrated mainly in the south, show a scale 1:1MM.  This is not a reasonable value for a cell that covers only a few km of a river.
This non-sense scale value for cell that covers a tiny part of the screen completely overwhelms the quilting algorithm, today.
I think that these are simple ENC coding errors. But the published charts are what they are. The question becomes, what to do?
I suppose we could code a sanity check, maybe comparing the longest linear dimension on the cell (meters) to the scale multiplied by a very large but not impossible onscreen display size (say 2 m).  Or something like that.  We would need to do that for every ENC cell loaded.  I am not thrilled by this idea, obviously.
Any other ideas?

nohal, 2022-12-18:
Are we able to get the on-screen size of a bounding box for the cell in pixels? If we can, simply skipping cells covering less than let's say 20 (or 50 or whatever) pixels in X or Y direction, which obviously can't result in anything usefull I can imagine, would probably do the trick.

bdbcat, 2022-12-18:
I had code for that in an earlier version.  Dropped it for I-cant-remember-why.
We can get cell pixel size at render time.  But I was hoping to detect this case upstream, at cell add time, and skip the cell entirely.
Here is the reason:  We compose the quilt based initially on the values stored in the chart database, before actually loading any cells.  Faster that way.  So having an unusable cell in the database is an annoyance.
But I'll think about some non-intrusive compose-time logic.

nohal, 2022-12-18:
Other wild idea - if the extension of the whole quilt is 50x 50 degrees, a cell covering 5x5 minutes probably can be skipped.

bdbcat, 2022-12-18:
Not a bad idea, resurrected.  In github now, give it a try.  If cell area < 1% screen area, drop it.
Still feels a bit hacky, since we really have a broken quilt that we are detecting and fixing dynamically.  But, whatever...

Hakansv, 2022-12-18:
Sorry to intrude in this important task. A very good attempt to get rid of intrusive big scaled details. Of course it can also give the opposite as all compromises. My comments are observations for consideration only. 
Now comparing a DK ENC. With today's master I've to increase weight zoom scale from 0 to 2 to get a reasonable amount of details at zoom scale 1:100 000.  56°42 N 011°24 at earlier provided charts.

nohal, 2022-12-18:
For the IENCs it seems to work really well so far, even the CZ "noodles" display at a very reasonable scale.

nohal, 2022-12-18:
I've downloaded and somewhat checked all the European IENCs currently available. Overall the new logic works beautifully.

- There are some cells in Romania reporting scale 1:1000, these appear late for my taste (I vaguely remember seeing something similar with the US IENCs reporting scale 1:5000)
- I have seen a cell reporting scale 1:10, that of course is not working at all
- The Dutch switched to en uneven tiled model of cells, seems pretty good for testing that scenario

[image]

closed 2023-02-02 by nohal