OpenCPN/OpenCPN #2881: Track Icon not correct after cycling throught dawn/night mode

bcn2, 2022-12-15:
**Describe the bug**
With "Track active" the icon on the tool bar will show "not enabled" after cycling through Night Mode. 

**To Reproduce**
1. Enable "Track"
2. Go through Dawn/Night/Dusk --> Day
3. Icon shows Track not enabled - however it is still. Provokes to click on the Track icon again and thus first stopping and next enabling the track again. What breaks the track into a new one.

**Desktop (please complete the following information if applicable):**
 - OS: observed under Windows and Debian, macOS is ok.
 - Version OCPN 5.6.2

LeaseOnLife, 2022-12-19:
This may be related: Since MUI in 5.0, I sometimes click on the track icon, and the status of the icon doesn't change, but the function has changed, which can be seen in the "mouse-over" track field.
It may have to do with moving the mouse while clicking..

rgleason, 2023-01-04:
Lets see some screen shots?

closed 2023-02-02 by nohal