OpenCPN/OpenCPN #2882: Lights description with Mo (X) Color not correct in Object Query

bcn2, 2022-12-15:
**Describe the bug**
Lights using a Morse Code as timing characteristics are not described correctly in the Object query.

Seen at drilling platforms on Norwegian charts.

On the canvas the lights are depicted correctly, however at an object query the result differs.
One example on the screen shot. 
OCPN 5.6.2, o-chars PI 1.0.29

[image]

bdbcat, 2023-03-11:
I guess we want the query to say exactly as the chart shows, e.g.
Mo(U) R 15s 3 Nm

??

bcn2, 2023-03-11:
Correct.

bdbcat, 2023-03-15:
Corrected now in o-charts v1.0.32.0, and v2.0.0.42.
Both in master catalog now.
Also correcting in O571 Core, in github master and CI now.

bcn2, 2023-03-15:
Checked and ok...

closed 2023-03-15 by bcn2