OpenCPN/OpenCPN #2889: Segfault on complex shapes

nohal, 2022-12-21:
I'm able to reliably reproduce a segfault on shapes with thousands of vertices as found on `DK1GSOUT.000` (South tip of Greenland from the DK S57 set) on the Linux AMD machine with current master.

```
Thread 1 "opencpn" received signal SIGSEGV, Segmentation fault.
(gdb) bt
#0  0x00007ffff5495b4a in __memmove_avx_unaligned_erms () at /lib64/libc.so.6
#1  0x00007fffc69ce601 in _mesa_glthread_upload () at /usr/lib64/dri/radeonsi_dri.so
#2  0x00007fffc69d289c in _mesa_marshal_DrawArrays () at /usr/lib64/dri/radeonsi_dri.so
#3  0x0000000000d3cb51 in s52plib::RenderToGLAC_GLSL(_ObjRazRules*, _Rules*) (this=0x6fef190, rzRules=0xb3926e0, rules=0xa99a700)
    at /home/nohal/source/OpenCPN/libs/s52plib/src/s52plib.cpp:8347
#4  0x0000000000d3b8f1 in s52plib::RenderToGLAC(_ObjRazRules*, _Rules*) (this=0x6fef190, rzRules=0xb3926e0, rules=0xa99a700) at /home/nohal/source/OpenCPN/libs/s52plib/src/s52plib.cpp:8022
#5  0x0000000000d3e112 in s52plib::RenderAreaToGL(wxGLContext const&, _ObjRazRules*) (this=0x6fef190, glcc=..., rzRules=0xb3926e0)
    at /home/nohal/source/OpenCPN/libs/s52plib/src/s52plib.cpp:8761
#6  0x0000000000dbceb9 in s57chart::DoRenderOnGL(wxGLContext const&, ViewPort const&) (this=0xa1966f0, glc=..., VPoint=...) at /home/nohal/source/OpenCPN/src/s57chart.cpp:1630
#7  0x0000000000dbcd00 in s57chart::DoRenderRegionViewOnGL(wxGLContext const&, ViewPort const&, OCPNRegion const&, LLRegion const&, bool)
    (this=0xa1966f0, glc=..., VPoint=..., RectRegion=..., Region=..., b_overlay=false) at /home/nohal/source/OpenCPN/src/s57chart.cpp:1595
#8  0x0000000000dbc8c3 in s57chart::RenderRegionViewOnGLNoText(wxGLContext const&, ViewPort const&, OCPNRegion const&, LLRegion const&)
    (this=0xa1966f0, glc=..., VPoint=..., RectRegion=..., Region=...) at /home/nohal/source/OpenCPN/src/s57chart.cpp:1471
#9  0x0000000000cb5223 in glChartCanvas::RenderQuiltViewGL(ViewPort&, OCPNRegion const&) (this=0x17d9f10, vp=..., rect_region=...) at /home/nohal/source/OpenCPN/src/glChartCanvas.cpp:3216
#10 0x0000000000cb6505 in glChartCanvas::RenderCharts(ocpnDC&, OCPNRegion const&) (this=0x17d9f10, dc=..., rect_region=...) at /home/nohal/source/OpenCPN/src/glChartCanvas.cpp:3476
#11 0x0000000000cb8f49 in glChartCanvas::Render() (this=0x17d9f10) at /home/nohal/source/OpenCPN/src/glChartCanvas.cpp:4188
#12 0x0000000000cad1bd in glChartCanvas::OnPaint(wxPaintEvent&) (this=0x17d9f10, event=...) at /home/nohal/source/OpenCPN/src/glChartCanvas.cpp:1429
```

Hakansv, 2022-12-21:
Just for info..
Nothing similar seen on Win nor Rpi400-arm64 Rasbian-Debian 11 for the same chart.

nohal, 2022-12-22:
Yes, it looks like a Mesa/driver issue which we just trigger by feeding the pipeline with complex shape.

nohal, 2023-01-29:
Seems fixed in Mesa after latest updates.

nohal, 2023-09-07:
It's back with current packages on Fedora. And likely also Ubuntu as https://www.cruisersforum.com/forums/f134/help-faq-30943-128.html#post3818584 sounds pretty much describing the same using similar HW config.

nohal, 2023-09-07:
The problem is not dependent on AVX, persists with it disabled when SSE is used instead

```
#0  0x00007ffff539868d in __memmove_ssse3 () at /lib64/libc.so.6
#1  0x00007fffc21268cc in _mesa_glthread_upload ()
    at /usr/lib64/dri/radeonsi_dri.so
#2  0x00007fffc23706d4 in _mesa_marshal_DrawArrays ()
    at /usr/lib64/dri/radeonsi_dri.so
#3  0x0000000000d50489 in s52plib::RenderToGLAC_GLSL(_ObjRazRules*, _Rules*)
    (this=0x44cbf90, rzRules=0xadfecb0, rules=0x165c130)
    at /home/nohal/source/OpenCPN/libs/s52plib/src/s52plib.cpp:8268
#4  0x0000000000d4f077 in s52plib::RenderToGLAC(_ObjRazRules*, _Rules*)
    (this=0x44cbf90, rzRules=0xadfecb0, rules=0x165c130)
    at /home/nohal/source/OpenCPN/libs/s52plib/src/s52plib.cpp:7932
```

nohal, 2024-01-04:
With address sanitizer a possibly related problem is reported on the same system

```
./opencpn 
=================================================================
==559713==ERROR: AddressSanitizer: stack-use-after-return on address 0x7f89c9aafdc0 at pc 0x7f89da66eb32 bp 0x7fff84826be0 sp 0x7fff848263a0
READ of size 16 at 0x7f89c9aafdc0 thread T0
    #0 0x7f89da66eb31 in __interceptor_memcpy (/lib64/libasan.so.8+0x6eb31) (BuildId: 7fcb7759bc17ef47f9682414b6d99732d6a6ab0c)
    #1 0x7f89b098203b in _mesa_glthread_upload (/usr/lib64/dri/radeonsi_dri.so+0x18203b) (BuildId: 949da8ba60d5fddadf24e632642a5d4f18e1c1ad)
    #2 0x7f89b0bd4823 in _mesa_marshal_DrawArrays (/usr/lib64/dri/radeonsi_dri.so+0x3d4823) (BuildId: 949da8ba60d5fddadf24e632642a5d4f18e1c1ad)
    #3 0xd87eeb in ocpnDC::DrawLine(int, int, int, int, bool) /home/nohal/source/OpenCPN/gui/src/ocpndc.cpp:661
    #4 0x151ab14 in glChartCanvas::GridDraw() /home/nohal/source/OpenCPN/gui/src/glChartCanvas.cpp:1981
    #5 0x1554ee7 in glChartCanvas::DrawFloatingOverlayObjects(ocpnDC&) /home/nohal/source/OpenCPN/gui/src/glChartCanvas.cpp:2659
    #6 0x1559e7c in glChartCanvas::Render() /home/nohal/source/OpenCPN/gui/src/glChartCanvas.cpp:4385
    #7 0x155f4db in glChartCanvas::OnPaint(wxPaintEvent&) /home/nohal/source/OpenCPN/gui/src/glChartCanvas.cpp:1433
    #8 0x7f89d95c0aa1 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) (/lib64/libwx_baseu-3.2.so.0+0x1c0aa1) (BuildId: 989c209d44037d18092f742b47617361331adcb2)
    #9 0x7f89d95c2052 in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) (/lib64/libwx_baseu-3.2.so.0+0x1c2052) (BuildId: 989c209d44037d18092f742b47617361331adcb2)
    #10 0x7f89d95c31f5 in wxEvtHandler::TryHereOnly(wxEvent&) (/lib64/libwx_baseu-3.2.so.0+0x1c31f5) (BuildId: 989c209d44037d18092f742b47617361331adcb2)
    #11 0x7f89d95c327e in wxEvtHandler::ProcessEventLocally(wxEvent&) (/lib64/libwx_baseu-3.2.so.0+0x1c327e) (BuildId: 989c209d44037d18092f742b47617361331adcb2)
    #12 0x7f89d95c3391 in wxEvtHandler::ProcessEvent(wxEvent&) (/lib64/libwx_baseu-3.2.so.0+0x1c3391) (BuildId: 989c209d44037d18092f742b47617361331adcb2)
    #13 0x7f89d95c4182 in wxEvtHandler::SafelyProcessEvent(wxEvent&) (/lib64/libwx_baseu-3.2.so.0+0x1c4182) (BuildId: 989c209d44037d18092f742b47617361331adcb2)
    #14 0x7f89d9bb9250 in wxWindow::GTKSendPaintEvents(_cairo*) (/lib64/libwx_gtk3u_core-3.2.so.0+0x3b9250) (BuildId: 967faae8f3e6d6fc17b831ba495deb0e5bd55136)
    #15 0x7f89da2f662d in draw (/lib64/libwx_gtk3u_gl-3.2.so.0+0xf62d) (BuildId: 34a2dd1f419003a4b7093bfcab56e4bba23f7aa8)
    #16 0x7f89d8ca1633 in _gtk_marshal_BOOLEAN__STRING.part.0 (/lib64/libgtk-3.so.0+0xa1633) (BuildId: 14536d239f8c70710e10f28df80d180d24d93604)
    #17 0x7f89d8f55675 in gtk_widget_draw_marshaller (/lib64/libgtk-3.so.0+0x355675) (BuildId: 14536d239f8c70710e10f28df80d180d24d93604)
    #18 0x7f89d8554529 in g_closure_invoke (/lib64/libgobject-2.0.so.0+0x14529) (BuildId: 0378350ec67b6c4cb51aad5c4d581cce802a077b)
    #19 0x7f89d8582feb in signal_emit_unlocked_R.isra.0 (/lib64/libgobject-2.0.so.0+0x42feb) (BuildId: 0378350ec67b6c4cb51aad5c4d581cce802a077b)
    #20 0x7f89d8573679 in signal_emit_valist_unlocked (/lib64/libgobject-2.0.so.0+0x33679) (BuildId: 0378350ec67b6c4cb51aad5c4d581cce802a077b)
    #21 0x7f89d8573f90 in g_signal_emit_valist (/lib64/libgobject-2.0.so.0+0x33f90) (BuildId: 0378350ec67b6c4cb51aad5c4d581cce802a077b)
    #22 0x7f89d8574052 in g_signal_emit (/lib64/libgobject-2.0.so.0+0x34052) (BuildId: 0378350ec67b6c4cb51aad5c4d581cce802a077b)
    #23 0x7f89d8f66eb0 in gtk_widget_draw_internal (/lib64/libgtk-3.so.0+0x366eb0) (BuildId: 14536d239f8c70710e10f28df80d180d24d93604)
    #24 0x7f89d8f6c340 in gtk_widget_render (/lib64/libgtk-3.so.0+0x36c340) (BuildId: 14536d239f8c70710e10f28df80d180d24d93604)
    #25 0x7f89d8e097ff in gtk_main_do_event (/lib64/libgtk-3.so.0+0x2097ff) (BuildId: 14536d239f8c70710e10f28df80d180d24d93604)
    #26 0x7f89d8b3f416 in _gdk_event_emit (/lib64/libgdk-3.so.0+0x3a416) (BuildId: 7948520532691f84c2a378c0501355da7f6ccf48)
    #27 0x7f89d8b517b8 in _gdk_window_process_updates_recurse_helper.lto_priv.0 (/lib64/libgdk-3.so.0+0x4c7b8) (BuildId: 7948520532691f84c2a378c0501355da7f6ccf48)
    #28 0x7f89d8b55e84 in gdk_window_process_updates_internal (/lib64/libgdk-3.so.0+0x50e84) (BuildId: 7948520532691f84c2a378c0501355da7f6ccf48)
    #29 0x7f89d8b56090 in gdk_window_process_updates_with_mode.lto_priv.0 (/lib64/libgdk-3.so.0+0x51090) (BuildId: 7948520532691f84c2a378c0501355da7f6ccf48)
    #30 0x7f89d9bb65e7 in wxWindow::Update() (/lib64/libwx_gtk3u_core-3.2.so.0+0x3b65e7) (BuildId: 967faae8f3e6d6fc17b831ba495deb0e5bd55136)
    #31 0xa11610 in ChartCanvas::OnPaint(wxPaintEvent&) /home/nohal/source/OpenCPN/gui/src/chcanv.cpp:10730
    #32 0x7f89d95c0aa1 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) (/lib64/libwx_baseu-3.2.so.0+0x1c0aa1) (BuildId: 989c209d44037d18092f742b47617361331adcb2)
    #33 0x7f89d95c2052 in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) (/lib64/libwx_baseu-3.2.so.0+0x1c2052) (BuildId: 989c209d44037d18092f742b47617361331adcb2)
    #34 0x7f89d95c31f5 in wxEvtHandler::TryHereOnly(wxEvent&) (/lib64/libwx_baseu-3.2.so.0+0x1c31f5) (BuildId: 989c209d44037d18092f742b47617361331adcb2)
[... cut ...]
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==559713==ABORTING
```

leamas, 2024-01-04:
On my system:
``` 
  $ addr2line -e build/opencpn 0x15351ff
  /home/mk/src/OpenCPN/gui/src/glChartCanvas.cpp:1040 (discriminator 22)
```
Now this is on my system, yours might differ.  Still, 1040 is some string UTF8 mangling which at a glance looks suspicious.

nohal, 2024-01-04:
Points to https://github.com/OpenCPN/OpenCPN/blob/master/gui/src/glChartCanvas.cpp#L5392 for me.
Are you actually able to reproduce any of this? Either the crash on big shapes or the sanitizer report? For me the common denominator is all these problems happen in `_mesa_marshal_DrawArrays` in the `radeonsi` driver rather than being related to specific OpenCPN code (I can equally well make it happen in the basemap rendering code for example)

leamas, 2024-01-04:
No, just wanted to contribute *addr2line*.

leamas, 2024-01-04:
... and I'm on the nvidia driver (not nouveau). 

> Are you actually able to reproduce any of this?

Might make a try if you can point me to the chart you use.

nohal, 2024-01-04:
I'm pretty convinced this is actually a Mesa bug as it went away with some versions and then returned. But still would be fine to understand and work around if possible.
I can reproduce it with pretty much any NOAA ENC chartset, happens as soon as we draw something that has more than ~1000 nodes.
The sanitizer triggers immediately during startup even with no charts at all. (On the AMD GPU Linux system, not reproducible on macOS)

bdbcat, 2024-01-04:
The fault location points to the instruction:
      glBindBuffer(GL_ARRAY_BUFFER, rzRules->obj->auxParm0);

The instruction directs the driver to instruct the GPU to use a memory buffer that has already been uploaded to GPU space, prior to activating the associated shader code resident in the GPU.  If the buffer has been discarded by the GPU due to space constraints in GPU space, the buffer is refreshed for core space.  The buffer is of course larger, for complex shapes.

If the error is reliably reproducible, one test would be to disable VBO objects in this method.  This will require the GPU to upload the buffer whenever it needs it do render.

We have logic to use, or not use, VBO on GLAC (Area Color) methods, depending on the hardware vendor found.  See the comment at 617:

 // No chart type in current use requires VBO for GLAC rendering
  // Experimentation has shown that VBO is slower for GLAC rendering,
  //  since the per-object state change of glBindBuffer() is slow
  //  on most hardware, especially RPi.
  // However, we have found that NVidea GPUs
  //  perform much better with VBO on GLAC operations, so set that up.

  if ((renderer.Upper().Contains("NVIDIA")) ||
      (renderer.Upper().Contains("QUADRO")) ||
      (renderer.Upper().Contains("GEFORCE")))
    m_GLAC_VBO = true;

So, the simple experiment is to cause m_GLAC_VBO to be false for this particular hardware.

nohal, 2024-01-04:
Nope, crashes exactly the same in the `radeonsi` driver regardless of the `m_GLAC_VBO` value.

bdbcat, 2024-01-04:
Line number of crash?

nohal, 2024-01-04:
Hard to say right now as it takes even GDB with it currently
```
(gdb) bt
#0  0x00007ffff517c8d6 in __memmove_avx_unaligned_erms () at /lib64/libc.so.6
#1  0x00007fffd138203c in _mesa_glthread_upload () at /usr/lib64/dri/radeonsi_dri.so
#2  0x00007fffd15d4824 in _mesa_marshal_DrawArrays () at /usr/lib64/dri/radeonsi_dri.so

Fatal signal: Segmentation fault  <- this is GDB crashing
```

bdbcat, 2024-01-04:
As you mentioned, seems to point to an out-of-bounds memory move operation in mesa driver.

nohal, 2024-01-04:
These are the calls that trigger the issue with ENCs for me, disabling the two lines of course breaks the rendering, but no crashes are observed anywhere else in the code.
```diff
diff --git a/libs/s52plib/src/s52plib.cpp b/libs/s52plib/src/s52plib.cpp
index 5afa699c2..810e50ea5 100755
--- a/libs/s52plib/src/s52plib.cpp
+++ b/libs/s52plib/src/s52plib.cpp
@@ -3813,7 +3813,7 @@ int s52plib::RenderGLLS(ObjRazRules *rzRules, Rules *rules) {
 
       if (b_drawit) {
         // render the segment
-        glDrawArrays(GL_LINE_STRIP, seg_vbo_offset / (2 * sizeof(float)) , point_count);
+        //glDrawArrays(GL_LINE_STRIP, seg_vbo_offset / (2 * sizeof(float)) , point_count);
       }
     }
     ls_list = ls_list->next;
@@ -8283,7 +8283,7 @@ int s52plib::RenderToGLAC_GLSL(ObjRazRules *rzRules, Rules *rules) {
         box = p_tp->tri_box;
 
       if (!BBView.IntersectOut(box)) {
-        glDrawArrays(p_tp->type, VBO_offset_index, p_tp->nVert);
+        //glDrawArrays(p_tp->type, VBO_offset_index, p_tp->nVert);
       }
 
       VBO_offset_index += p_tp->nVert;
```

brian-mccarter, 2024-01-04:
I use radeonsi on Gentoo Linux and have seen this error since roughly July 2023 when mesa-23.1.3 was stabilized.  

Loading just about any ENC from the NOAA chart set will crash OpenCPN with a stack trace similar to [this](https://github.com/OpenCPN/OpenCPN/issues/2889#issuecomment-1710230238).  I can use mesa-22.0.4 without problems, but OpenCPN crashes starting with mesa-23.1.0.  The most resent version I've tested is mesa-23.3.2, which also crashes.

So a temporary workaround is to revert to <= mesa-22.0.4 until the problem is fixed, whether it's mesa or how OpenCPN is using mesa.

nohal, 2024-01-20:
Interestingly I can actually run the ASAN enabled build successfully without the segfault if I tell it not to stop on errors and ignore the issue it detects mentioned in https://github.com/OpenCPN/OpenCPN/issues/2889#issuecomment-1877144523

```
cmake -DENABLE_SANITIZER=address ..
make
ASAN_OPTIONS=detect_leaks=0,halt_on_error=0 ./opencpn 
```

Puzzling...

nohal, 2024-01-22:
I have tried some of the examples in https://github.com/MicBosi/VisualizationLibrary/ which also call to `glDrawArrays` on the same system and they work fine, so we probably might have a chance to avoid the problem somehow.

nohal, 2024-02-25:
I found it. The problem is GL threading was enables for the `radeonsi` driver by default in Mesa 22.3 (https://www.phoronix.com/news/Mesa-22.3-RadeonSI-glthread-On)

The fix is to simply

```diff
diff --git a/gui/src/ocpn_app.cpp b/gui/src/ocpn_app.cpp
index 48248eba9..4fe7a7833 100644
--- a/gui/src/ocpn_app.cpp
+++ b/gui/src/ocpn_app.cpp
@@ -953,6 +953,7 @@ MyApp::MyApp()
   // Handle e. g., wayland default display -- see #1166.
   if (wxGetEnv( "WAYLAND_DISPLAY", NULL))
     setenv("GDK_BACKEND", "x11", 1);
+  setenv("mesa_glthread", "false", 1);
 #endif   // __linux__
 }
```

bdbcat, 2024-02-25:
Wow.
Deep in the weeds there...

nohal, 2024-02-25:
I have pushed this workaround to `master`.

`mesa_glthread` is actually supported by many drivers, including Intel, just disabled by default, so if you want to have some fun, change the value to `true` and see if your machines explode...

commit d40a5645 (2024-02-25): Disable mesa_glthread on Linux. Closes #2889

- We know mesa_glthread is completely broken on AMD Radeon for us
- We suspect #3437 might be the same issue
closed 2024-02-25 by nohal