OpenCPN/OpenCPN #2891: PIM: Bug  Uncheck a plugin checkbox, don't apply, uncheck a second checkbox, crash.

rgleason, 2022-12-26:
**Describe the bug**
 Uncheck a plugin checkbox, don't apply, uncheck a second checkbox, crash.

**To Reproduce**
Steps to reproduce the behavior:
1. It was sloppy of me. I unchecked two plugin boxes without applying.
2. Uncheck one plugin box, then with the plugin box still open
3. Try unchecking another plugin.
4. Crash.

**Expected behavior**
Better behavior.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information if applicable):**
 - OS: Win11 
 - Version 5.7.1 Beta #1

**Additional context**
Add any other context about the problem here. Screenshots are usually a good way to ilustrate problems, attach them if applicable.

rgleason, 2023-01-29:
I tried this and it does not seem to be crashing. Can I close this now?

bdbcat, 2023-01-29:
Might have been for a specific set of two plugins.  Not a general problem.
Close for now, it may come back.

closed 2023-01-29 by rgleason