OpenCPN/OpenCPN #2893: PI:Dashboard   High res screen resizing has a vertical constraint.

rgleason, 2022-12-26:
**Describe the bug**
Resize Dashboard on a high res screen.
Vertical has a constraint.
Dashboard appears to be oversized somewhat.

**To Reproduce**
Steps to reproduce the behavior:
Not much to write here.

**Expected behavior**
Ability to resize the Dashboard more completely.

**Screenshots**
This appears to be the minimum vertical height for the normal instruments vertical arrangement and for the historic data screen.
Is this what is intended?  I thought we could restrict the size more in previous versions.
[image]

Also, I notice that there are fewer instruments (at least I think so).  Were derived instruments removed?

**Desktop (please complete the following information if applicable):**
 - OS: [e.g. macOS]
 - Version [e.g. 22]

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: [e.g. Galaxy S20]
 - Andorid version: [e.g. 10]

**Additional context**
Add any other context about the problem here. Screenshots are usually a good way to ilustrate problems, attach them if applicable.

rgleason, 2023-01-02:
Some further examples of the problem.

Restriction does not work ok.
[image]

Expansion seems to work ok
[image]

rgleason, 2023-01-02:
I am going to get more picky about Dashboard.. Sorry but I think it should work properly and now is the time to fix it.
Prerferences is unsatisfactory the way it is.
[image]

Hakansv, 2023-01-02:
Rick..
I'm not the author of the Dashboard as such but I've tried to adjust a little and scaled Win is tricky.
Discussion of your second picture with all "Pull up stops here":
Basically all min size is adapted to chosen text font size. Data with a sized font has to have space between lines and borders. A resize of a panel does not change the font size and I'm not the clever enough to change that. But here are more folks?
For the first panel from left it's one of the instruments controlling min size. And some is already to narrow for certain fields. Data is covering each others.
For the second I think GNSS Status graph is the "master" of min size. We need space for 12 satellites in a row without text or staples climbing upon each others.
The third panel from left can't be smaller given the selected font size.

The preferences windows is not nice on you scaled Windows. Agree. No work is done to that dialog for years. Can possible be better if anyone has the time. On the other hand it's not used frequently.

rgleason, 2023-01-02:
Are you saying that you don't have these problems on Windows?

rgleason, 2023-01-02:
Regarding the Preferences Menu, this should be made resizeable in a number of ways. It is a pain even if you have a standard screeen res, because it can't be adjusted.  Also the adjustments made  for size and proportions should be remembered the next time OpenCPN is opened.  IE: they should be stored in the ini or config file.

Hakansv, 2023-01-03:
Rick..
I'm not defending or answering. It was meant as a discussion point.
The scaled WIn also in combination with wxWindows 3.2.1 for Win and some other is rather new.
I do agree this needs some future tweaks. Not less since the "good old" Dashboard does face environment not anticipated.

rgleason, 2023-01-03:
No problem. I am going to try to fix it (big trouble) since this affects two other plugins that have to be fixed too. Wish me luck

Hakansv, 2023-01-03:
You asked if my Windows is good.
Of the Win around four is very fine also after the scaling adjustments done. These are not scaled, that's 100%.
The fifth where scaling to 175% is used is also fine now. There the font sizes also needed some tweaks to 12 for data, 10 for label and the rest sized to 8. A one time occurrence.
Your preferences example is weird. So there your Win scale can use some kind of scale dependent adjustment.

rgleason, 2023-01-05:
Would be interested in some screen shots.  Here are Tactics screenshots.  The preferences screen is much better because it is resizeable.  I believe that Dashboard has devolved significantly and should be fixed.
Additionally, both Dashboard and Tactics have a bad anomyly with regard to setting of text sizes. Note that when text is set to 12 the text is way too small and when set on 14 the text jumps up in size significantly, and much more than it should.

[image]

[image]

[image]

rgleason, 2023-01-06:
In **TACTICS_pi**
I am finding a similar discontinuity in the sizing of fonts. There is a huge jump in size between Font size 12 and 14. Why?  It makes it difficult to find a good size.  For example:
             Normal     Way too small
Title       12                 --- <--Scales ok but the background bar does not change thickness.
Data      34                  14    <---Big discontinuity in size
Label     14                  12    <---Big discontinuity in size
Small     14                  12    <---Big discontinuity in size

[image]

[image]

Hakansv, 2023-01-15:
Rick..
Would it be worth to consider this "adaptation" for high DPI Windows as a kind of solution?
See the link on this [CF post](https://www.cruisersforum.com/forums/f134/windows-11-small-text-271961.html#post3728986)
I've not tested. My display is not showing the same degree of bad sizing as yours.

rgleason, 2023-01-15:
I just learned that the Windows High Res DPI settings for specific programs that Hakan had suggested were working!, because I undid those changes and OpenCPN looks a bit more like the earlier one.  These settings are

1. Using the search,  Find the OpenCPN.exe being used in C:\ProgramData\Microsoft\Windows\Start Menu\Programs\OpenCPN 

2 Right click on it and pick properties.

3. Click on the "Compatibility" Tab

4. Turn on High DPI Settings for this program.

5. Under  "Program DPI"  check the box.  "Use this settings to fix scalling problems for the program..."

6. Under "High DPI scaling override,  check the box  "Override high DPI scaling behavior.

7. Under the same area in the dropdown, select  "Application"

8. Pick OK.

9. Then close OpenCPN and reboot.

Then when that is done, together with Dave's improvements and changes, everything works much much better.

Thank you.   Will post screenshots in this thread.

rgleason, 2023-01-16:
After some testing, using Dave's recent changes, built with VS2020 and the exe copied to the most current Beta2 directory,  with the Windows High Res DPI settting as above, but # 7
Under the same area in the dropdown, select "System (enhanced)"  seems to better.

[image]
Sliders in UI are all at zero except the Toolbar which I made -1 smaller.

rgleason, 2023-01-16:
One of the problems with this setup is that some of the plugins do not appear when opened. I've tried most everything I can think of, and I don't think it is offscreen somewhere lost in space, as I can't retrieve it using normal methods.

rgleason, 2023-01-16:
Next thing I will do is turn off High Res DPI under Properties and see what changes Dave has done.

rgleason, 2023-01-16:
Leaving the exact same settings as above for the UI slider adjustments.
Icons are a little larger, so the slider will go down another notch to -2 probably.
Dashboard settings are good.

I like the way the Plugin Menus are showing, they are denser and about right compare to the plugin menus above which are more spread out and occupy more real estate.
The rollover text is way too small and in way too big a yellow field, this is from Jon's ocpn_draw boundary.

The enc depth is way too small and the chart Text for places is way too small.

As you can see the Statusbar plugin text has completely broken down and is not legible really.
Why is this happening?

Using this configuration, it appears that we will have difficulty with tactics_pi and dashboard_tactics  because they have text that is way too small, but they are using similar settings to Dashboard (actually Dashboard is using Arial 8 for everything).  Some change is going to be necessary with these plugins.
[image]

I found that the two plugins that are now showing with the High Res DPI set are Watchdog and Plots which both have empty windows to start with, before they become populated. That might have some bearing with the problem.

rgleason, 2023-01-16:
BTW my Windows resize is the recommended 250% !
I have not changed it during any of these reports. It has been that way for a very long time.

rgleason, 2023-01-16:
Also the right click on the sea menu has changed.
The text is too small for me but the shortcuts are not overwritting the commands now.

Also the ENC Depth is too small and the Text labels for places is too small!!

[image]

rgleason, 2023-01-16:
Hakan wrote earlier:
```
Rick..
Would it be worth to consider this "adaptation" for high DPI Windows as a kind of solution?
See the link on this [CF post](https://www.cruisersforum.com/forums/f134/windows-11-small-text-271961.html#post3728986)
I've not tested. My display is not showing the same degree of bad sizing as yours.
```
I think it is going to be part of the solution, but not all have this alternative available, so lets hope they have less high res screens.

rgleason, 2023-01-16:
Also without using Windows High Res DPI settings the console window has become way too small.  Also see the poor Autopilot_raymarine plugin next to the console?
[image]

rgleason, 2023-01-16:
Dave I am going through the core issues now. Will post a new Issue.

Hakansv, 2023-01-22:
Rick..
If possible please try present Dashboard master update. Not tested on your high DPI screen.

rgleason, 2023-01-23:
Hakan,  I git pulled and built the current version of master.
Then tried Dashboard. It is better.  The fonts are good!

[image]

The instruments are acting well and I can resize the heights up to a certain point (as shown).
For me, this is acceptable.

Thank you!

I do have some problems with the way Preferences work. It takes up too much real estate and the icons are oversized.  It should be resizable.  

Also I have one particular peeve about Adding Instruments. I like to see more than just 3 instruments, I am always hunting and pecking for one that is not there because I can only see a very very short list.
I would like this Add Instrument menu to be resizeable!

Screenshots follow.
[image]
[image]
[image]

rgleason, 2023-01-29:
I think this has been largely addressed and should be closed now.
Additional more specific items can become Issues.
Closing

closed 2023-01-29 by rgleason